project-chip/connectedhomeip #9185: SDK forces user code to use certain folder structrues.

eve-cxrp, 2021-08-23:
#### Problem
PR #9124 moved the gen folder into zap-generated folder. Now e.g. `src/app/util/config.h` uses `#include <zap-generated/gen_config.h>` as an include. So the sdk user would also need to change their folder structure (rename). Is this really necessary to force that on the sdk user?

andy31415, 2022-02-07:
I expect that any generated files should eventually be fully separated out from 'regular' source files. I expect build rules to set required include paths for accessing files.

closed 2022-02-07 by andy31415

eve-cxrp, 2022-02-08:
> I expect that any generated files should eventually be fully separated out from 'regular' source files. I expect build rules to set required include paths for accessing files.

But is it the case for now?