project-chip/connectedhomeip #9214: Resource leak in Linux/BlueZ

samadDotDev, 2021-08-24:
#### Problem

In `src/platform/Linux/bluez/MainLoop.cpp`,
`chip::Device::Internal::MainLoop::EnsureStarted()` spawns a thread which is never joined or cleaned up after.
```
int pthreadErr = pthread_create(&mThread, nullptr, &MainLoop::Thread, reinterpret_cast<void *>(this));
```

 #### Proposed Solution

`pthread_join()` for `mThread` needs to be called at an appropriate place to wait for the thread to terminate. Since `pthread_create` is called with null `attr`, the thread is in joinable state.

samadDotDev, 2021-08-24:
@msandstedt @z-michel

holbrookt, 2022-02-01:
@samadDotDev it looks like this is still an issue - can you confirm? Do you know who to assign it to?

samadDotDev, 2022-02-02:
@holbrookt Yes, the latest master still has `mThread` in joinable state but is never joined.

woody-apple, 2022-02-07:
Given this is a linux platform example issue, can be resolved post v1.0

msandstedt, 2022-10-10:
Still an issue.  But as @woody-apple, can be post-1.0.

msandstedt, 2022-11-08:
Fixed by https://github.com/project-chip/connectedhomeip/pull/23320.

closed 2022-11-08 by msandstedt