project-chip/connectedhomeip #9221: ScanNetworks callback array type support for Android (and in general)

fengli, 2021-08-24:
#### Problem

It seems for array types in commands callback, the encoder and decoder aren't finalized yet. One example you can see is for [ScanNetworks command](https://github.com/project-chip/connectedhomeip/blob/b683e6aaaa139624bce72ba797aa797dfa73616c/src/controller/java/zap-generated/ChipClusters.java#L3504):

```
    public interface ScanNetworksResponseCallback {
      void onSuccess(int errorCode, String debugText
          // wifiScanResults: /* TYPE WARNING: array array defaults to */ uint8_t *
          // Conversion from this type to Java is not properly implemented yet
          // threadScanResults: /* TYPE WARNING: array array defaults to */ uint8_t *
          // Conversion from this type to Java is not properly implemented yet
          );

      void onError(Exception error);
    }
```

wifiScanResults is missing in JAVA implementation, similar for iOS.

 #### Proposed Solution
Fix the issue.

fengli, 2021-08-24:
seems linked to #6620

woody-apple, 2022-01-25:
Given this is an older issue, is this complete at this point? If not, please re-open.

closed 2022-01-25 by woody-apple