project-chip/connectedhomeip #9320: Native Thread connection fails

pramodhg123, 2021-08-30:
resolve 5544332211 1 fails 
OTBR and node up date passes but the device is not detected and the next zclread or other commands fail saying not connected

Setup the OTBR using the github otbr code and setup for etherent
Build the nrfconnect light example
Build the python chip-device-ctrl and then run the same
The following is the error observed

NRF device side log shows no error

: 882399 [DL]Sending notification for CHIPoBLE TX (ConnId 00, len 3)      	 
D: 882475 [DL]Notification for CHIPoBLE TX done (ConnId 0x00)              	 
D: 884962 [DL]Write request received for CHIPoBLE RX (ConnId 0x00)         	 
D: 887468 [DL]Sending notification for CHIPoBLE TX (ConnId 00, len 3)      	 
D: 887545 [DL]Notification for CHIPoBLE TX done (ConnId 0x00)              	 
D: 890032 [DL]Write request received for CHIPoBLE RX (ConnId 0x00)         	 
D: 892538 [DL]Sending notification for CHIPoBLE TX (ConnId 00, len 3)      	 
D: 892615 [DL]Notification for CHIPoBLE TX done (ConnId 0x00)              	 
D: 895102 [DL]Write request received for CHIPoBLE RX (ConnId 0x00)         	 
D: 897608 [DL]Sending notification for CHIPoBLE TX (ConnId 00, len 3)      	 
D: 897685 [DL]Notification for CHIPoBLE TX done (ConnId 0x00)              	 
D: 900172 [DL]Write request received for CHIPoBLE RX (ConnId 0x00)         	 
D: 902678 [DL]Sending notification for CHIPoBLE TX (ConnId 00, len 3)      	 
D: 902755 [DL]Notification for CHIPoBLE TX done (ConnId 0x00)              	 
D: 905242 [DL]Write request received for CHIPoBLE RX (ConnId 0x00)         	 
D: 907747 [DL]Sending notification for CHIPoBLE TX (ConnId 00, len 3)      	 
D: 907825 [DL]Notification for CHIPoBLE TX done (ConnId 0x00)              	 
D: 910312 [DL]Write request received for CHIPoBLE RX (ConnId 0x00)         	 
D: 912817 [DL]Sending notification for CHIPoBLE TX (ConnId 00, len 3)      	 
D: 912895 [DL]Notification for CHIPoBLE TX done (ConnId 0x00)              	 
D: 915381 [DL]Write request received for CHIPoBLE RX (ConnId 0x00)

on the python device controller side when the resolve command is run

[1630317400.796478][150244:150252] CHIP:CTL: OperationalDiscoveryComplete for device ID 1
[1630317401.867509][150244:150252] CHIP:DIS: Node ID resolved for 0x0000000000000001 to [2002:c0a8:40d:10:e65f:1ff:fe0f:27d8]:5540
Node address has been updated
[1630317401.867853][150244:150252] CHIP:CTL: OperationalDiscoveryComplete for device ID 1
[1630317401.868171][150244:150252] CHIP:DIS: Node ID resolved for 0x0000000000000001 to [2002:c0a8:40d:10:e65f:1ff:fe0f:27d8]:5540
Node address has been updated
[1630317401.868377][150244:150252] CHIP:CTL: OperationalDiscoveryComplete for device ID 1
[1630317401.868758][150244:150252] CHIP:DIS: Node ID resolved for 0x0000000000000001 to [2002:c0a8:40d:10:e65f:1ff:fe0f:27d8]:5540
Node address has been updated
[1630317401.868953][150244:150252] CHIP:CTL: OperationalDiscoveryComplete for device ID 1
[1630317401.869214][150244:150252] CHIP:DIS: Node ID resolved for 0x0000000000000001 to [2002:c0a8:40d:10:e65f:1ff:fe0f:27d8]:5540
Node address has been updated
[1630317401.869402][150244:150252] CHIP:CTL: OperationalDiscoveryComplete for device ID 1
[1630317401.869658][150244:150252] CHIP:DIS: Node ID resolved for 0x0000000000000001 to [2002:c0a8:40d:10:e65f:1ff:fe0f:27d8]:5540
Node address has been updated
**[1630317401.869841][150244:150252] CHIP:CTL: OperationalDiscoveryComplete for device ID 1
[1630317403.227857][150244:150252] CHIP:SC: CASESession timed out while waiting for a response from the peer. Expected message type was 49**
[1630317403.227967][150244:150252] CHIP:CTL: Device connection failed. Error ../../src/protocols/secure_channel/CASESession.cpp:269: CHIP Error 0x00000032: Timeout
Failed to commission: 50
[1630317408.372701][150244:150252] CHIP:DIS: Node ID resolved for 0x0000000000000001 to [2002:c0a8:40d:10:e65f:1ff:fe0f:27d8]:5540
Node address has been updated

Damian-Nordic, 2021-09-01:
Please describe your setup better. It looks like the controller cannot reach the device using the 2002:c0a8:40d:10:e65f:1ff:fe0f:27d8 address, so you must be missing some routing configuration.

pramodhg123, 2021-09-02:
Raspberry 1 - running OTBR with ethernet connected and built for ethernet
Raspberry 2 - running Python Device Controller
Nordic nrf devkit acting as Thread Accessory
Raspberry 1 and Raspberry 2 connected via ethernet through a TPLink router

pramodhg123, 2021-09-07:
Attaching the latest test ping log and also picture of the thread accessory ip address
[image]

ubuntu@ubuntuservercontroller:~/connectedhomeip/out/python_env/bin$ 
PING fd97:c850:630f:6a66:98eb:2212:f137:ddc5(fd97:c850:630f:6a66:98eb:2212:f137:ddc5) 56 data bytes
^C
--- fd97:c850:630f:6a66:98eb:2212:f137:ddc5 ping statistics ---
8 packets transmitted, 0 received, 100% packet loss, time 7149ms

ubuntu@ubuntuservercontroller:~/connectedhomeip/out/python_env/bin$ ping -6 fd1f:47a7:e70d:2295:0:ff:fe00:4c07
PING fd1f:47a7:e70d:2295:0:ff:fe00:4c07(fd1f:47a7:e70d:2295:0:ff:fe00:4c07) 56 data bytes
^C
--- fd1f:47a7:e70d:2295:0:ff:fe00:4c07 ping statistics ---
8 packets transmitted, 0 received, 100% packet loss, time 7161ms

ubuntu@ubuntuservercontroller:~/connectedhomeip/out/python_env/bin$ 
PING fd1f:47a7:e70d:2295:90d2:8e80:a8e6:d2ca(fd1f:47a7:e70d:2295:90d2:8e80:a8e6:d2ca) 56 data bytes
^C
--- fd1f:47a7:e70d:2295:90d2:8e80:a8e6:d2ca ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 2031ms

ubuntu@ubuntuservercontroller:~/connectedhomeip/out/python_env/bin$ ping -6 fe80:0:0:0:4841:d0e9:d6ee:92f8
PING fe80:0:0:0:4841:d0e9:d6ee:92f8(fe80::4841:d0e9:d6ee:92f8) 56 data bytes
ping: sendmsg: Invalid argument
ping: sendmsg: Invalid argument
ping: sendmsg: Invalid argument
ping: sendmsg: Invalid argument
ping: sendmsg: Invalid argument
ping: sendmsg: Invalid argument
ping: sendmsg: Invalid argument
^C
--- fe80:0:0:0:4841:d0e9:d6ee:92f8 ping statistics ---
7 packets transmitted, 0 received, 100% packet loss, time 6138ms

woody-apple, 2022-01-25:
Is this still happening on recent builds?

closed 2022-01-25 by woody-apple