project-chip/connectedhomeip #9326: ESP32 Bridge app causing the device to reboot

pramodhg123, 2021-08-30:
ESP32 bridge app build on test_event_5
The device reboots showing the following error
V (1296) chip[DIS]: Init fabric pairing table with server storage
I (1306) chip[SVR]: Retrieved value from server storage.
I (1306) chip[SVR]: Retrieved value from server storage.
I (1316) chip[SVR]: Retrieved value from server storage.
I (1326) chip[SVR]: Retrieved value from server storage.
I (1326) chip[SVR]: Retrieved value from server storage.
I (1336) chip[SVR]: Retrieved value from server storage.
I (1336) chip[SVR]: Retrieved value from server storage.
I (1346) chip[SVR]: Retrieved value from server storage.
I (1356) chip[SVR]: Retrieved value from server storage.
I (1356) chip[SVR]: Retrieved value from server storage.
I (1366) chip[SVR]: Retrieved value from server storage.
I (1366) chip[SVR]: Retrieved value from server storage.
I (1376) chip[SVR]: Retrieved value from server storage.
I (1376) chip[SVR]: Retrieved value from server storage.
I (1386) chip[SVR]: Retrieved value from server storage.
I (1396) chip[SVR]: Retrieved value from server storage.
V (1396) chip[IN]: UDP::Init bind&listen port=5540
V (1406) chip[IN]: UDP::Init bind&listen port=5540
V (1406) chip[IN]: TransportMgr initialized
I (1426) chip[DL]: NVS set: chip-counters/GlobalMCTR = 509000 (0x7C448)
I (1426) chip[SVR]: Retrieved value from server storage.
I (1426) chip[SVR]: Delete value in server storage
V (1436) chip[DIS]: Fabric 1 is not yet configured
I (1436) chip[SVR]: Retrieved value from server storage.
I (1446) chip[SVR]: Delete value in server storage
V (1456) chip[DIS]: Fabric 2 is not yet configured
I (1456) chip[SVR]: Retrieved value from server storage.
I (1456) chip[SVR]: Delete value in server storage
E (1466) chip[DL]: Long dispatch time: 620 ms, for event type 2
V (1466) chip[DIS]: Fabric 3 is not yet configured
I (1466) chip[DL]: Starting ESP WiFi layer
I (1476) chip[SVR]: Retrieved value from server storage.
I (1486) wifi:mode : sta (84:cc:a8:5f:45:c8)
I (1486) wifi:enable tsf
W (1496) wifi:Haven't to connect to a suitable AP now!
I (1496) chip[DL]: Done driving station state, nothing else to do...
I (1496) chip[SVR]: Delete value in server storage
W (1506) wifi:Haven't to connect to a suitable AP now!
I (1516) chip[DL]: Done driving station state, nothing else to do...
V (1516) chip[DIS]: Fabric 4 is not yet configured
I (1526) chip[DL]: Changing ESP WiFi mode: STA -> STA+AP
I (1526) chip[SVR]: Retrieved value from server storage.
I (1536) wifi:mode : sta (84:cc:a8:5f:45:c8) + softAP (b0:b0:b5:c7:b4:1a)
I (1546) wifi:Total power save buffer number: 16
I (1546) wifi:Init max length of beacon: 752/752
I (1556) wifi:Init max length of beacon: 752/752
I (1566) chip[DL]: Configuring WiFi AP: SSID MATTER-F00-235A-FEFF, channel 1
I (1566) chip[SVR]: Delete value in server storage
I (1566) chip[DL]: WiFi AP state change: NotActive -> Activating
V (1576) chip[DIS]: Fabric 5 is not yet configured
I (1576) chip[DL]: WIFI_EVENT_STA_START
I (1586) chip[SVR]: Retrieved value from server storage.
W (1586) wifi:Haven't to connect to a suitable AP now!
I (1596) chip[DL]: Done driving station state, nothing else to do...
I (1596) chip[SVR]: Delete value in server storage
I (1606) bridge-devicecallbacks: Current free heap: 102084

V (1616) chip[DIS]: Fabric 6 is not yet configured
I (1616) chip[DL]: WIFI_EVENT_AP_START
I (1626) chip[SVR]: Retrieved value from server storage.
I (1626) chip[DL]: WiFi AP state change: Activating -> Active
I (1636) chip[SVR]: Delete value in server storage
I (1636) bridge-devicecallbacks: Current free heap: 101760

V (1646) chip[DIS]: Fabric 7 is not yet configured
I (1656) chip[SVR]: Retrieved value from server storage.
I (1666) chip[SVR]: Delete value in server storage
V (1666) chip[DIS]: Fabric 8 is not yet configured
I (1676) chip[SVR]: Retrieved value from server storage.
I (1676) chip[SVR]: Delete value in server storage
V (1686) chip[DIS]: Fabric 9 is not yet configured
I (1686) chip[SVR]: Retrieved value from server storage.
I (1696) chip[SVR]: Delete value in server storage
V (1706) chip[DIS]: Fabric 10 is not yet configured
I (1706) chip[SVR]: Retrieved value from server storage.
I (1706) chip[SVR]: Delete value in server storage
V (1716) chip[DIS]: Fabric 11 is not yet configured
I (1716) chip[SVR]: Retrieved value from server storage.
I (1726) chip[SVR]: Delete value in server storage
V (1736) chip[DIS]: Fabric 12 is not yet configured
I (1736) chip[SVR]: Retrieved value from server storage.
I (1746) chip[SVR]: Delete value in server storage
V (1756) chip[DIS]: Fabric 13 is not yet configured
I (1756) chip[SVR]: Retrieved value from server storage.
I (1756) chip[SVR]: Delete value in server storage
V (1766) chip[DIS]: Fabric 14 is not yet configured
I (1766) chip[SVR]: Retrieved value from server storage.
I (1776) chip[SVR]: Delete value in server storage
V (1786) chip[DIS]: Fabric 15 is not yet configured
I (1786) chip[SVR]: Retrieved value from server storage.
I (1796) chip[SVR]: Delete value in server storage
V (1806) chip[DIS]: Fabric 16 is not yet configured
I (1806) chip[SVR]: Retrieved value from server storage.
I (1806) chip[SVR]: Retrieved value from server storage.
I (1816) chip[SVR]: Retrieved value from server storage.
I (1826) chip[SVR]: Retrieved value from server storage.
I (1826) chip[SVR]: Retrieved value from server storage.
I (1836) chip[SVR]: Retrieved value from server storage.
I (1836) chip[SVR]: Retrieved value from server storage.
I (1846) chip[SVR]: Retrieved value from server storage.
I (1856) chip[SVR]: Retrieved value from server storage.
I (1856) chip[SVR]: Retrieved value from server storage.
I (1866) chip[SVR]: Retrieved value from server storage.
I (1866) chip[SVR]: Retrieved value from server storage.
I (1876) chip[SVR]: Retrieved value from server storage.
I (1886) chip[SVR]: Retrieved value from server storage.
I (1886) chip[SVR]: Retrieved value from server storage.
I (1896) chip[SVR]: Retrieved value from server storage.
I (1896) chip[DIS]: WaitForPairing(): HasAdvertisementDelegate 1
I (1906) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable, device name MATTER-3840)
I (1916) chip[DIS]: CHIP minimal mDNS started advertising.
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=2 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40I (1926) chip[DIS]: Replying to DNS-SD service listing request

I (1936) chip[DIS]: Replying to DNS-SD service listing request
I (1946) chip[DIS]: Replying to DNS-SD service listing request
I (1956) chip[DIS]: Replying to DNS-SD service listing request
I (1956) chip[DIS]: Replying to DNS-SD service listing request
I (1966) chip[DL]: CHIPoBLE advertising started
I (1966) chip[DIS]: Replying to DNS-SD service listing request
I (1976) bridge-devicecallbacks: Current free heap: 101784

I (1976) chip[DIS]: Replying to DNS-SD service listing request
I (1996) chip[DIS]: Failed to find a valid admin pairing. Node ID unknown
I (1996) chip[DIS]: Start dns-sd server - no current nodeId
V (2006) chip[DIS]: Using wifi MAC for hostname
V (2006) chip[DL]: rotatingDeviceId: 00007C5F6E176CD40F68685D100A1CF8A98B
I (2016) chip[DIS]: Advertise commission parameter vendorID=9050 productID=65279 discriminator=3840/00
I (2026) chip[DIS]: CHIP minimal mDNS configured a
***ERROR*** A stack overflow in task main has been detected.

Backtrace:0x4008f6c2:0x3ffbb3d0 0x4008fdf1:0x3ffbb3f0 0x4009311e:0x3ffbb410 0x400919bc:0x3ffbb490 0x4008fee8:0x3ffbb4b0 0x4008fe9a:0x00000000 |<-CORRUPTED
0x4008f6c2: panic_abort at /home/zigbee/tools/esp-idf/components/esp_system/panic.c:367

0x4008fdf1: esp_system_abort at /home/zigbee/tools/esp-idf/components/esp_system/system_api.c:112

0x4009311e: vApplicationStackOverflowHook at /home/zigbee/tools/esp-idf/components/freertos/port/xtensa/port.c:489

0x400919bc: vTaskSwitchContext at /home/zigbee/tools/esp-idf/components/freertos/tasks.c:3289

0x4008fee8: _frxt_dispatch at /home/zigbee/tools/esp-idf/components/freertos/port/xtensa/portasm.S:432

0x4008fe9a: _frxt_int_exit at /home/zigbee/tools/esp-idf/components/freertos/port/xtensa/portasm.S:231

pramodhg123, 2021-08-30:
The device keeps rebooting and not able to connect

dhrishi, 2021-08-30:
@pramodhg123 Can you please increase the ESP32 main task stack size? In`examples/bridge-app/esp32/sdkconfig.defaults` add:
 CONFIG_ESP_MAIN_TASK_STACK_SIZE=4096

And then clean and compile the application.

This is wrt the discussion here https://github.com/project-chip/connectedhomeip/pull/9263#issuecomment-907156606

pramodhg123, 2021-08-30:
Working fine after changing the same. Issue sometimes observed in other apps also.

pramodhg123, 2021-08-30:
Other issue observed with Bridge app after this fix is 
udp endpoint pool FULL
and resolve is not working

dhrishi, 2021-08-30:
With the latest set of changes in the CHIP repo, the fabric id is not 0. You will have to fetch it first and then use the same in your resolve command. Steps:

$ chip-device-ctrl> getfabricid
<- Output ->

$ chip-device-ctrl> resolve <fabric_id_in_decimal> <node_id>

Let me know if this helps. We are modifying the README files of the examples to update this.

pramodhg123, 2021-08-30:
This is test_event_5 branch and the issue is only observed with bridge app

pramodhg123, 2021-08-30:
will test with the solution you have provided and update

jonsmirl, 2021-08-30:
If you use the python app it is going to generate a new compressed fabric ID every time you restart it. The chip-tool app persists the value to a file so that it is stable.