project-chip/connectedhomeip #9395: ESP: EnableNetwork command fails with python controller.

sweetymhaiske, 2021-09-01:
#### Problem
* With recent commit head on master branch 2ccdcc1f276f8155578f3610fa56e9008b4041a9 On python controller, zcl EnableNetwork command fails. 
* **Controller Log**
```
chip-device-ctrl > zcl NetworkCommissioning EnableNetwork 12344321 0 0 networkID=str:Wce***** breadcrumb=0 timeoutMs=1000
[1630481699.771274][179176:179184] CHIP:IN: Encrypted message 0x7fd7fd1a4740 to 0x0000000000BC5C01 of type 8 and protocolId 5 on exchange 39022.
[1630481699.771285][179176:179184] CHIP:IN: Sending msg 0x7fd7fd1a4740 to 0x0000000000BC5C01 at utc time: 10813111 msec
[1630481699.771289][179176:179184] CHIP:IN: Sending secure msg on generic transport
[1630481699.771307][179176:179184] CHIP:IN: Secure msg send status ../../src/transport/raw/BLE.cpp:104: Success
[1630481711.774169][179176:179184] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 39022
[1630481711.774217][179176:179184] CHIP:ZCL: DefaultResponse:
[1630481711.774221][179176:179184] CHIP:ZCL:   Transaction: 0x7fd80089d138
[1630481711.774223][179176:179184] CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
Failed to receive command response: None
chip-device-ctrl > [1630481717.341926][179176:179184] CHIP:BLE: ack recv timeout, closing ep 0x7fd8008a01a0
[1630481717.343301][179176:179184] CHIP:BLE: Auto-closing end point's BLE connection.
[1630481717.343312][179176:179184] CHIP:DL: Closing BLE GATT connection (con 0x7fd7f402e030)
[1630481719.683800][179176:179180] CHIP:DL: Bluez notify CHIPoBluez connection disconnected
```
* **Device side log**
```
I (259257) chip[DL]: WIFI_EVENT_AP_STOP
I (259257) chip[DL]: WiFi AP state change: Deactivating -> NotActive
I (259267) app-devicecallbacks: Current free heap: 83652

I (261627) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
V (261627) chip[IN]: Clearing BLE pending packets.
I (261627) chip[BLE]: Releasing end point's BLE connection back to application.
E (261637) chip[BLE]: HandleChipConnectionReceived failed, err = d
E (261647) chip[BLE]: failed handle new chip BLE connection, status = d
I (262297) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,0>, prof:1
```

bzbarsky-apple, 2021-09-03:
@sweetymhaiske Is this fixed now that https://github.com/project-chip/connectedhomeip/pull/9407 merged?

sweetymhaiske, 2021-09-03:
Yes this issue is resolved.

closed 2021-09-03 by sweetymhaiske