project-chip/connectedhomeip #9406: ReliableMessageProtocolConfig.h can give you a 0 sized retransmit table

drempelg, 2021-09-01:
#### Problem
#ifndef CHIP_CONFIG_RMP_RETRANS_TABLE_SIZE
#ifdef PBUF_POOL_SIZE
#define CHIP_CONFIG_RMP_RETRANS_TABLE_SIZE std::min(PBUF_POOL_SIZE, CHIP_CONFIG_MAX_EXCHANGE_CONTEXTS)
#elif CHIP_SYSTEM_CONFIG_PACKETBUFFER_POOL_SIZE != 0
#define CHIP_CONFIG_RMP_RETRANS_TABLE_SIZE std::min(CHIP_SYSTEM_CONFIG_PACKETBUFFER_POOL_SIZE, CHIP_CONFIG_MAX_EXCHANGE_CONTEXTS)
#else
#define CHIP_CONFIG_RMP_RETRANS_TABLE_SIZE CHIP_CONFIG_MAX_EXCHANGE_CONTEXTS
#endif // PBUF_POOL_SIZE
#endif // CHIP_CONFIG_RMP_RETRANS_TABLE_SIZE

This block of code will result in a 0 sized retransmit table if you have PBUF_POOL_SIZE set to 0.  Which is what you are instructed to do in the configuration header for LWIP if you set LWIP_PBUF_FROM_CUSTOM_POOLS to 1.

This resulted in us not being able to use RMP because our product is using custom pbuf pools.  We crashed pretty quickly.

 #### Proposed Solution
we worked around it by changing 
#ifdef PBUF_POOL_SIZE 

to 

#if PBUF_POOL_SIZE != 0

Without being super familiar with the code though I don't know if that's a good general solution or not.

EOD though you shouldn't get a 0 sized table with a valid LWIP configuration.

yunhanw-google, 2021-09-01:
@yufengwangca  please take a look, thanks

yufengwangca, 2021-09-03:
https://github.com/project-chip/connectedhomeip/pull/9465

closed 2022-02-07 by woody-apple