project-chip/connectedhomeip #9417: Is there a guide document how to test report functionality?

cabin15, 2021-09-02:
#### Problem
<what's wrong or missing, please include any applicable:
This is not a bug report, but rather a question.
I wish if there is a guide to test report functionality.
My understanding is that,
1) cluster (i.e lighting-app) is launched.
2) controller (i.e chip-tool or python controller) provisions the controllee.
3) controller triggers report to cluster.
4) Any attribute change on the cluster is sent to controller.

Question is, how to test this?
It would be so kind someone show me where the guide document is.
If there is not document, it would be kind to let me know the test command on cluster and controller side.

Thank you in advance.
 >

learnSu, 2021-09-08:
Do you know how to test? Actually, the report function has been added in the project, but I did not see the report from the wireless message

woody-apple, 2022-01-25:
Please follow up with @kidip

closed 2022-01-25 by woody-apple