project-chip/connectedhomeip #9440: [ota-provider-app] Support qpg

holbrookt, 2021-09-02:
#### Problem
ota-provider-app does not have an implementation for qpg

 #### Proposed Solution
Support qpg

tima-q, 2021-11-18:
@holbrookt  Thanks for creating this issue. A question though, as a constrained platform I would expect we'll need support for the ota-requestor-app side ?
Running a provider would be useful for testing purposes, but can that be run as a 'non-controller' device ? We'll need to pull in images through a serial API or something alike.

holbrookt, 2021-11-23:
@tima-q You should definitely focus on support for the OTA Requestor first. Note that we are in active development of an API for this cluster, however some other platforms have submitted copy/paste example apps based off of the Linux implementation.

After a quick scan, I don't see anything in the spec that specifically prevents constrained devices from being OTA Providers. However based on the complexity of the Provider and recommendations on image storage, it may be difficult.

closed 2022-02-08 by selissia