project-chip/connectedhomeip #9441: Rotating device id representation in TLV Additional Data needs to be aligned with the spec

bzbarsky-apple, 2021-09-02:
#### Problem
Per spec, the rotating device id in the TLV Additional Data in the onboarding payload is just a byte blob.  The spec currently does not specify how that byte blob is encoded into TLV, but https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/3945 will clarify that it's an octet string.

In our implementation right now the rotating device id is instead encoded as a UTF-8 string holding a hex representation of the bytes.  This is correct for the mdns advertisement, but not for the bluetooth advertisement.

 #### Proposed Solution
Align implementation with spec by storing the bytes as bytes, not hex.

@chrisdecenzo @hnnajh

chrisdecenzo, 2021-09-02:
More feedback from Boris on implementation:

AdditionalDataPayloadGenerator::generateAdditionalDataPayload has a buffer of size RotatingDeviceId::kHexMaxLength which is 2*RotatingDeviceId::kMaxLength+1 , where RotatingDeviceId::kMaxLength is kLifetimeCounterSize + kHashSuffixLength , so 18, and hence the buffer in the generator is 37 bytes.

The generator then fills this buffer with two bytes of counter, then the hex string representation of the rotating device id (which is not what the spec says to do, afaict).   That's 36 bytes, plus the null terminator.  And stores it in the TLV as a UTF-8 string, not an octet string, which is nuts, because it's not valid UTF-8 for those first two bytes.

Now looking at the parser, before my changes, it set up a buffer of size SetupPayloadData::kRotatingDeviceIdLength and tried to copy the string into it.  kRotatingDeviceIdLength is 18.  So that call just always returned failure.

After my changes we at least manage to get data out, and as long as it fits in kHexMaxLength (which in this case it does) pass it out to the caller.  But obviously the payload parser was never exercised anywhere.....

But more to the point, the parser and generator don't agree on their buffer sizes.

Need unit tests

AdditionalDataPayloadGenerator::generateAdditionalDataPayload currently in the tree does:
    ReturnErrorOnFailure(writer.OpenContainer(AnonymousTag, kTLVType_Structure, innerWriter));
so it's already doing an anonymous tag, not what the spec used to say.

And the mdns version of rotating device id has a different format from the version in the additional data payload
In that the mdns version is hex-encoded and the additional data payload just has binary
Per spec, that is.  But the implementation doesn't do that right now.

The BLE version in GATT is only implemented on Linux bluez backend and not even properly wrapped in build conditions

hnnajh, 2021-09-02:
Thanks @bzbarsky-apple  for your feedback, i will send a PR soon for fixing those points

This feature was tested on linux and i created a CHIP-Tool command for parsing the generated rotating id and that was working back then. I will add some unit tests to avoid any regressions in the future.
examples/chip-tool/commands/payload/AdditionalDataParseCommand.cpp 

regarding the linux mdns implementation, the rotating id was guarded by build conditions in my PR
https://github.com/project-chip/connectedhomeip/pull/4210/files , not sure what happens afterwards, it seems some of the refactoring didn't take that into consideration.

holbrookt, 2022-02-01:
@hnnajh it looks like #9455 fixed this issue but you didn't close it. Please reopen if I'm incorrect

closed 2022-02-01 by holbrookt