project-chip/connectedhomeip #9479: KeyValueStoreManagerImpl::_Get does not seem to handle a large offset_bytes safely.

bzbarsky-apple, 2021-09-04:
#### Problem
The code looks like this:
```
    uint8_t buf[read_size];
    ReturnErrorOnFailure(mStorage.ReadValueBin(key, buf, read_size, read_size));

    size_t copy_size = std::min(value_size, read_size - offset_bytes);
...
    ::memcpy(value, buf + offset_bytes, copy_size);
```

What happens when `offset_bytes > read_size`?  Looks to me like we will happily start reading at past the end of the buffer and probably read `value_size` bytes from there.

 #### Proposed Solution
Probably need to return an error if `offset_bytes > read_size`.

@rgoliver @andy31415

andy31415, 2022-01-31:
looks critical: memory safety issue (also fix seems easy enough if we just make it error out)

andy31415, 2022-01-31:
Marking fixed from #9954

closed 2022-01-31 by andy31415