project-chip/connectedhomeip #9481: New CommissioningComplete code does not ensure there is a working CASE session

bzbarsky-apple, 2021-09-04:
#### Problem
The `DeviceCommissioner::CommissioningComplete` function added in #9041 uses `GetDevice`, not `GetConnectedDevice`, so does not ensure that there is an actual CASE session to the given node id.

 #### Proposed Solution
Fix that.  Though see #9480 about whether this function should exist in the first place.

And maybe rename `GetDevice` to `GetMaybeDisconnectedDevice` to make it clearer what it does?

@sweetymhaiske @pan-apple

sweetymhaiske, 2021-09-06:
@bzbarsky-apple `CommissioningComplete` function gets called after `OperationalDiscoveryComplete` function. In `OperationalDiscoveryComplete` function `GetConnectedDevice` API is called. Hence `CommissioningComplete` API would get called only after successful CASE session.

bzbarsky-apple, 2021-09-16:
> CommissioningComplete function gets called after OperationalDiscoveryComplete function

This requirement is not documented in the API documentation for CommissioningComplete, and is relying on implementation details of other functions that might change, no?  This is basically a bug waiting to happen.

Though again, this function should not exist.

closed 2021-09-28 by woody-apple