project-chip/connectedhomeip #9526: Add documentation about why this cast is required

woody-apple, 2021-09-07:
could you add some documentation on why this cast is ok? What guarantees that the avahi types and chip inet types are the same and why is a cast needed (if they are the same types I would expect just assignment to work)?

_Originally posted by @andy31415 in https://github.com/project-chip/connectedhomeip/pull/9484#r703509120_

msandstedt, 2021-09-08:
Avahi uses int and we use unsigned.  Avahi appears to be using int in order to support the `AVAHI_IF_UNSPEC` special value:

```
enum {
     AVAHI_IF_UNSPEC = -1       
 };
 ```

Furthermore, apparently Avahi promises to use no other negative values:

``` 
 #define AVAHI_IF_VALID(ifindex) (((ifindex) >= 0) || ((ifindex) == AVAHI_IF_UNSPEC))
```

We are therefore safe by checking for `AVAHI_IF_UNSPEC` before casting:
```
            if (interface != AVAHI_IF_UNSPEC)
            {
                service.mInterface = static_cast<chip::Inet::InterfaceId>(interface);
```

Obviously if Avahi changes and the previous promises of `AVAHI_IF_VALID` no longer hold, things could break.  But with current code, this appears safe.

andy31415, 2021-09-14:
A comment in the code at the static cast place should help code readers (and if that ever fails, we would be able to look back on what assumptions changed)