project-chip/connectedhomeip #9531: Is there a guide how to test report functionality?

learnSu, 2021-09-08:
Hi,All
 When I use the lighting project, I see that the reporting function has been added, but when I trigger it via the local button, there is no wireless message reported by the device. How should I verify this reporting function?

tewarid, 2021-09-17:
What's your hardware setup?

App log should show you reporting happen - for example (linux all-clusters-app)

```log
[1630583381.359413][974:974] CHIP:DMG: <RE> Sending report...
[1630583381.359476][974:974] CHIP:DMG: IM RH moving to [Reporting]
[1630583381.359638][974:974] CHIP:IN: Secure message was encrypted: Msg ID 1
[1630583381.359720][974:974] CHIP:IN: Encrypted message 0x5769b8 to 0x5F419ACFB36E2E1C of type 5 and protocolId 5 on exchange 44256.
[1630583381.359811][974:974] CHIP:IN: Sending msg 0x5769b8 to 0x5F419ACFB36E2E1C at utc time: 64765164 msec
[1630583381.359878][974:974] CHIP:IN: Sending secure msg on generic transport
[1630583381.365141][974:974] CHIP:IN: Secure msg send status ../../third_party/connectedhomeip/src/inet/IPEndPointBasis.cpp:917: Success
[1630583381.365327][974:974] CHIP:DMG: <RE> ReportsInFlight = 1 with readHandler 0, RE has no more messages
[1630583381.483279][974:974] CHIP:EM: Received message of type 0x40 with vendorId 0x0000 and protocolId 0x0000 on exchange 44256
[1630583381.483504][974:974] CHIP:DMG: in state Reporting, receive status report, protocol id is 5, protocol code is 0
[1630583381.483583][974:974] CHIP:DMG: <RE> OnReportConfirm: NumReports = 0
[1630583381.483645][974:974] CHIP:DMG: IM RH moving to [Uninitialized]
[1630583381.483791][974:974] CHIP:IN: Secure message was encrypted: Msg ID 2
[1630583381.483874][974:974] CHIP:IN: Encrypted message 0xbeb83860 to 0x5F419ACFB36E2E1C of type 16 and protocolId 0 on exchange 44256.
[1630583381.483965][974:974] CHIP:IN: Sending msg 0xbeb83860 to 0x5F419ACFB36E2E1C at utc time: 64765289 msec
[1630583381.484030][974:974] CHIP:IN: Sending secure msg on generic transport
[1630583381.484353][974:974] CHIP:IN: Secure msg send status ../../third_party/connectedhomeip/src/inet/IPEndPointBasis.cpp:917: Success
```

woody-apple, 2022-01-25:
Can you please follow up with @kidip ? Thanks!

closed 2022-01-25 by woody-apple