project-chip/connectedhomeip #9540: Stack overflow on M5

cecille, 2021-09-08:
#### Problem
There is a stack overflow on M5. Even after https://github.com/project-chip/connectedhomeip/pull/9263 went in, we're still seeing a stack overflow in our testing. This is related to the generation of the rotating device id in https://github.com/project-chip/connectedhomeip/pull/8888. 

 #### Proposed Solution
short term: Can we disable rotating device ID on ESP32 as a short term fix to unblock testing?
Long term: Is this an exceptionally stack heavy process? Is there a more appropriate place to do this work? Related - is there a proper way to determine the max stack size for espressif? Related - see https://github.com/project-chip/connectedhomeip/issues/6825

hnnajh, 2021-09-08:
@cecille , do you have any stack trace or logs? i tried it on my esp32 and i don't see any stack overflow

cecille, 2021-09-08:
Taking this back for now - I thought it was related to the previous crash, but this is a fun new crash. After some bisecting, it looks like it was introduced in
cb5d39e70 support different dhcp client in Linux (#9490)

cecille, 2021-09-09:
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x800e0001  PS      : 0x00060c30  A0      : 0x800f5dec  A1      : 0x3ffe6ee0  
A2      : 0x800e0001  A3      : 0x3ffd0d64  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x3ffbc710  A7      : 0x000000c0  A8      : 0x800f1fb8  A9      : 0x00000001  
A10     : 0x3ffd0d64  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3ffe6f00  
A14     : 0x3ffe6ee0  A15     : 0x3ffe6e50  SAR     : 0x00000007  EXCCAUSE: 0x00000014  
EXCVADDR: 0x800e0000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace:0x400dfffe:0x3ffe6ee0 0x400f5de9:0x3ffe6f00 0x400f5e3a:0x3ffe6f20 0x400f5ffe:0x3ffe6f40 0x400f6025:0x3ffe6fa0 0x40095235:0x3ffe6fc0
0x400dfffe: moveHue(unsigned char, unsigned short, unsigned char, unsigned char, bool) at /home/cecille/chip/connectedhomeip/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/clusters/color-control-server/color-control-server.cpp:458

0x400f5de9: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToSystemLayer(chip::DeviceLayer::ChipDeviceEvent const*) at /home/cecille/chip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.cpp:262

0x400f5e3a: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/cecille/chip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.cpp:222

0x400f5ffe: chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/cecille/chip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:364
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at /home/cecille/chip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:179

0x400f6025: chip::DeviceLayer::PlatformManager::RunEventLoop() at /home/cecille/chip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:296
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /home/cecille/chip/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:204

0x40095235: vPortTaskWrapper at /home/cecille/chip/esp-idf/components/freertos/port/xtensa/port.c:168

cecille, 2021-09-09:
After some bisecting, it looks like it was introduced with 3a505010973 Virtualize System::Layer interfaces (#9366)

kpschoedel, 2021-09-09:
From the `DispatchEventToSystemLayer` in the trace, this is a duplicate of #9563 fixed by #9569.

closed 2021-09-09 by kpschoedel