project-chip/connectedhomeip #9550: For on network provisioning chip-tool is not calling the PersistDeviceList

cjandhyala, 2021-09-08:
When chip tool is used with on network provisioning, its not calling the PersistDeviceList . So after pairing is complete, Device can not be controlled, since it fails retrieving the PariedDeviceList from the storage.

cjandhyala, 2021-09-08:
@pan-apple @cecille

cecille, 2021-09-16:
Can you provide more details about the commands you are running? I'm having trouble reproducing this.

cecille, 2021-09-16:
I'm attaching logs from my system. Controller side:
`cecille@cecille-z2:~/chip/connectedhomeip/out/debug/standalone$ ./chip-tool pairing onnetwork 0 20202021 3840 fc00::a 5540
[1631806823.040014][1364087:1364087] CHIP:CTL: Read local id 0x000000000001B669, remote id 0xA70B41F97BCC2335
[1631806823.043102][1364087:1364087] CHIP:CTL: Generating NOC
[1631806823.043386][1364087:1364087] CHIP:CTL: Generating ICAC
[1631806823.049351][1364087:1364087] CHIP:IN: UDP::Init bind&listen port=5541
[1631806823.049445][1364087:1364087] CHIP:IN: UDP::Init bound to port=5541
[1631806823.049466][1364087:1364087] CHIP:IN: UDP::Init bind&listen port=5541
[1631806823.049522][1364087:1364087] CHIP:IN: UDP::Init bound to port=5541
[1631806823.049542][1364087:1364087] CHIP:IN: TransportMgr initialized
[1631806823.049555][1364087:1364087] CHIP:DIS: Init fabric pairing table with server storage
[1631806823.051915][1364087:1364087] CHIP:DL: MDNS failed to join multicast group on heth0 for address type IPv4: ../../src/inet/IPEndPointBasis.cpp:259: Inet Error 0x00000110: Address not found
[1631806823.051992][1364087:1364087] CHIP:ZCL: Using ZAP configuration...
[1631806823.052037][1364087:1364087] CHIP:ZCL: deactivate report event
[1631806823.052464][1364087:1364087] CHIP:DIS: Verifying the received credentials
[1631806823.053565][1364087:1364087] CHIP:DIS: Added new fabric at index: 1, Initialized: 1
[1631806823.053588][1364087:1364087] CHIP:DIS: Assigned compressed fabric ID: 0x7F0FEE1620293FEA, node ID: 0x000000000001B669
[1631806823.053599][1364087:1364087] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x7F0FEE1620293FEA
[1631806823.053785][1364087:1364092] CHIP:DL: CHIP task running
[1631806823.053918][1364087:1364092] CHIP:CTL: Generated random node id: 0x989C34B92E9435E5
[1631806823.054368][1364087:1364092] CHIP:SC: Assigned local session key ID 5
[1631806823.054421][1364087:1364092] CHIP:IN: Build plaintext message 0x5623290f5d98 to 0x0000000000000000 of type 32 and protocolId 0 on exchange 19670 with MessageCounter 448395102.
[1631806823.054443][1364087:1364092] CHIP:IN: Sending plaintext msg 0x5623290f5d98 to 0x0000000000000000 at utc time: 2026543551 msec
[1631806823.054456][1364087:1364092] CHIP:IN: Sending msg on generic transport
[1631806823.054558][1364087:1364092] CHIP:SC: Sent PBKDF param request
[1631806823.061921][1364087:1364092] CHIP:EM: Received message of type 0x21 with vendorId 0x0000 and protocolId 0x0000 on exchange 19670
[1631806823.061948][1364087:1364092] CHIP:EM: Rxd Ack; Removing MessageCounter:1AB9F75E from Retrans Table
[1631806823.061957][1364087:1364092] CHIP:EM: Removed CHIP MessageCounter:1AB9F75E from RetransTable
[1631806823.061969][1364087:1364092] CHIP:SC: Received PBKDF param response
[1631806823.061985][1364087:1364092] CHIP:SC: Peer assigned session ID 0
[1631806823.062609][1364087:1364092] CHIP:EM: Piggybacking Ack for MessageCounter:D5383B17 with msg
[1631806823.062626][1364087:1364092] CHIP:IN: Build plaintext message 0x5623290f5d98 to 0x0000000000000000 of type 34 and protocolId 0 on exchange 19670 with MessageCounter 448395103.
[1631806823.062638][1364087:1364092] CHIP:IN: Sending plaintext msg 0x5623290f5d98 to 0x0000000000000000 at utc time: 2026543559 msec
[1631806823.062646][1364087:1364092] CHIP:IN: Sending msg on generic transport
[1631806823.062704][1364087:1364092] CHIP:SC: Sent spake2p msg1
[1631806823.063619][1364087:1364092] CHIP:EM: Received message of type 0x23 with vendorId 0x0000 and protocolId 0x0000 on exchange 19670
[1631806823.063646][1364087:1364092] CHIP:EM: Rxd Ack; Removing MessageCounter:1AB9F75F from Retrans Table
[1631806823.063657][1364087:1364092] CHIP:EM: Removed CHIP MessageCounter:1AB9F75F from RetransTable
[1631806823.063670][1364087:1364092] CHIP:SC: Received spake2p msg2
[1631806823.064181][1364087:1364092] CHIP:EM: Piggybacking Ack for MessageCounter:D5383B18 with msg
[1631806823.064201][1364087:1364092] CHIP:IN: Build plaintext message 0x5623290f5d98 to 0x0000000000000000 of type 36 and protocolId 0 on exchange 19670 with MessageCounter 448395104.
[1631806823.064215][1364087:1364092] CHIP:IN: Sending plaintext msg 0x5623290f5d98 to 0x0000000000000000 at utc time: 2026543561 msec
[1631806823.064225][1364087:1364092] CHIP:IN: Sending msg on generic transport
[1631806823.064280][1364087:1364092] CHIP:SC: Sent spake2p msg3
[1631806823.064592][1364087:1364092] CHIP:EM: Received message of type 0x40 with vendorId 0x0000 and protocolId 0x0000 on exchange 19670
[1631806823.064619][1364087:1364092] CHIP:EM: Rxd Ack; Removing MessageCounter:1AB9F760 from Retrans Table
[1631806823.064629][1364087:1364092] CHIP:EM: Removed CHIP MessageCounter:1AB9F760 from RetransTable
[1631806823.064645][1364087:1364092] CHIP:IN: New secure session created for device 0x989C34B92E9435E5, key 0!!
[1631806823.064674][1364087:1364092] CHIP:CTL: Remote device completed SPAKE2+ handshake
[1631806823.064684][1364087:1364092] CHIP:CTL: Sending OpCSR request to 0x7ffcae15cf80 device
[1631806823.064708][1364087:1364092] CHIP:DMG: ICR moving to [Initialize]
[1631806823.064721][1364087:1364092] CHIP:DMG: ICR moving to [AddCommand]
[1631806823.064765][1364087:1364092] CHIP:IN: Secure message was encrypted: Msg ID 0
[1631806823.064777][1364087:1364092] CHIP:IN: Build encrypted message 0x5623290f5d98 to 0x989C34B92E9435E5 of type 8 and protocolId 5 on exchange 19671 with MessageCounter 0.
[... cut ...]
[1631806890.152795][1361870:1361870] CHIP:IN: Sending msg on generic transport
[1631806890.152882][1361870:1361870] CHIP:DMG: ICR moving to [   Sending]
[1631806890.152903][1361870:1361870] CHIP:DMG: ICR moving to [Uninitiali]
[1631806890.154076][1361870:1361870] CHIP:EM: Received message of type 0x10 with vendorId 0x0000 and protocolId 0x0000 on exchange 2798
[1631806890.154132][1361870:1361870] CHIP:EM: Rxd Ack; Removing MessageCounter:00000001 from Retrans Table
[1631806890.154153][1361870:1361870] CHIP:EM: Removed CHIP MessageCounter:00000001 from RetransTable
`

cecille, 2021-09-17:
Is this just the open commissining window command that is failing?

If so, it's probably because chip tool generates a new random node id for all pairing commands. Can you confirm whether this is the command you're having problems running?

cecille, 2021-09-17:
https://github.com/project-chip/connectedhomeip/pull/9806 - Does this work for what you're trying to do?

cecille, 2021-09-21:
If you wipe the chip_* files in /tmp, does it work?

mossmgr, 2022-11-18:
This issue still persists, I am getting the same error. 

`2022-11-18 15:06:18 Beeta-67376B PersistentStorage[20157] WARNING Initializing persistent storage from file: /tmp/chip-device-ctrl-storage.json
2022-11-18 15:06:18 Beeta-67376B PersistentStorage[20157] WARNING Loading configuration from /tmp/chip-device-ctrl-storage.json...
2022-11-18 15:06:18 Beeta-67376B chip.TS[20157] INFO Last Known Good Time: 2022-11-18T12:09:17
2022-11-18 15:06:18 Beeta-67376B PersistentStorage[20157] INFO SetSdkKey: g/gcc = b'\xa0\x0f\x00\x00'
2022-11-18 15:06:18 Beeta-67376B PersistentStorage[20157] INFO Committing...
2022-11-18 15:06:18 Beeta-67376B PersistentStorage[20157] INFO SetSdkKey: g/gdc = b'\xa0\x0f\x00\x00'
2022-11-18 15:06:18 Beeta-67376B PersistentStorage[20157] INFO Committing...
2022-11-18 15:06:18 Beeta-67376B chip.ZCL[20157] INFO Using ZAP configuration...
2022-11-18 15:06:18 Beeta-67376B chip.DL[20157] ERROR MDNS failed to join multicast group on eth0 for address type IPv4: ../../src/inet/UDPEndPointImplSockets.cpp:761: Inet Error 0x00000110: Address not found
2022-11-18 15:06:18 Beeta-67376B chip.DL[20157] ERROR MDNS failed to join multicast group on wlan0 for address type IPv4: ../../src/inet/UDPEndPointImplSockets.cpp:761: Inet Error 0x00000110: Address not found
2022-11-18 15:06:18 Beeta-67376B chip.IN[20157] INFO CASE Server enabling CASE session setups
2022-11-18 15:06:18 Beeta-67376B chip.DIS[20157] INFO Updating services using commissioning mode 0
2022-11-18 15:06:18 Beeta-67376B chip.DL[20157] ERROR MDNS failed to join multicast group on eth0 for address type IPv4: ../../src/inet/UDPEndPointImplSockets.cpp:761: Inet Error 0x00000110: Address not found
2022-11-18 15:06:18 Beeta-67376B chip.DL[20157] ERROR MDNS failed to join multicast group on wlan0 for address type IPv4: ../../src/inet/UDPEndPointImplSockets.cpp:761: Inet Error 0x00000110: Address not found
2022-11-18 15:06:18 Beeta-67376B chip.DIS[20157] INFO CHIP minimal mDNS started advertising.
2022-11-18 15:06:18 Beeta-67376B chip.DIS[20157] INFO Advertise commission parameter vendorID=65521 productID=32769 discriminator=3840/15 cm=0
2022-11-18 15:06:18 Beeta-67376B chip.DIS[20157] INFO CHIP minimal mDNS configured as 'Commissioner device'; instance name: 684C32FF0F97EB99.
2022-11-18 15:06:18 Beeta-67376B chip.DIS[20157] INFO mDNS service published: _matterd._udp
Exception ignored in: <function FabricAdmin.__del__ at 0xb5a9c028>
Traceback (most recent call last):
  File "/root/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/FabricAdmin.py", line 125, in __del__
    self.Shutdown()
  File "/root/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/FabricAdmin.py", line 118, in Shutdown
    if (self._isActive):
AttributeError: 'FabricAdmin' object has no attribute '_isActive'
__init__() missing 1 required positional argument: 'vendorId'`

cecille, 2022-11-22:
Can you elaborate more on the problem you're having? I'm seeing a couple of errors here, but it doesn't appear to be related to persisting the device list.

Can you indicate:
1) what device are these logs from?
2) what command are you running that's failing - what do you expect to happen, what's the actual behaviour?
3) What interfaces are available on this device? Logs list eth0 and wlan0 - are there others?

mossmgr, 2022-11-22:
Yes of course. These logs are from a custom hardware running Linux debian armhf (bullseye).
The device has Z-Wave chip support and BLE support and has wlan and ethernet. wlan0 is the intended interface for running Matter over IP protocol. The device is connected to the local network via ethernet cable and wlan0 is active.

I am following the guide located at [https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/python_chip_controller_building.md](url)

when I run this command in my python_env with root privileges i get the above mentioned error.
`out/python_env/bin/chip-device-ctrl` 

What I expect is running `chip-device-ctrl > ble-scan` and get a list of available commissionee devices.

Thanks.

cecille, 2022-11-22:
I'm still not totally sure what the setup is here. The logs look like they're coming from the python code (chip-device-ctrl), which is the controller side. Is that what's running on the custom hardware? Or do you have a device application running there and these are the host side logs?

Either way, you've hit a documentation bug - chip-device-ctrl is deprecated. PR up here to add a documentation warning: https://github.com/project-chip/connectedhomeip/pull/23724. If you want to use a python-based controller tool, matter-repl is the one that's currently being maintained. The other option is chip-tool, which is what is mostly used by the certification group.

Can you specify:
1) what specifically is running on your custom hardware? is it an example application (which one), something custom, or is the custom hardware acting as the controller/commissioner?
2) what is the other side of the connection? Is it a separate piece of hardware or are you trying to run the device and the controller on the same hardware? If so, what's the setup there?

If you're a CSA member, I would also suggest getting onto the slack.

mossmgr, 2022-11-23:
Yes the logs are coming from the python code. I tried to follow the documentation and I noticed that chip-repl has replaced chip-device-ctrl but I needed the device to act as a commissioner and there were some features in chip-repl which as of now do not exist in chip-repl. In any case I had a problem with MDNS, as the logs show.

1. I am trying to run a commissioner on my Device to discover Matter devices(via BLE ). The device should act as a controller/commissioner.
2. The other side of the connection as of now is nothing, I either have to create a dummy device to simulate a matter compatible device (client) or buy a raspberry pie. But currently i have no end device ( just want to test if commissioner runs on my Edge computer) 

Thank you.

cecille, 2022-11-23:
Which features do you need that are not available in chip-repl? Commissioning is definitely supported.

When you start the repl, an object called devCtrl is created. You can call a number of different commissioning options there. The code offers commissioning helpers at several granularities. For the most part, if you're just looking to commission a device, you would just call either CommissionWifi or CommissionWithCode, depending on what you wanted to do. 

To do the equivalent using chip-tool (C++ tool):
build it using:
`./gn_build.sh`
then if you're sure you definitely want to set up with ble:
`./out/debug/standalone/chip-tool pairing ble-wifi <nodeid> <ssid> <password> <setup-pin-code> <discriminator>`
or if you prefer using the code (commissioner will look on the network as well):
`./out/debug/standalone/chip-tool pairing code-wifi <node-id> <ssid> <password> <payload>`
where payload is the QR or manual setup code.

If you're simulating a matter device, you're talking about the server side (mostly). The commissioner/controller is the client side. You won't be able to test any of the actual commissioning code without an end device. Both the repl and chip-tool will fail out saying they can't find the device, so you're not going to be testing much. If you're using linux, you can run a simulated device on the same system as the commissioner. Most people use either the all-cluster or lighting example for basic testing. You can build that either using the scripts/build/build_examples.py script, or just using ninja directly. If you run that in a separate terminal, you can commission between them. The caveat here is that you will NOT be using BLE commissioning - you will be using on-network commissioning. If you want to test BLE, you need a separate BLE-capable device.

mossmgr, 2022-11-24:
Thanks for the comprehensive answer. Now logically i have a clear idea about my test scenario. However the issue that i originally had persists. I followed the python guide on building python_env for testing, when i try to run `chip-device-ctrl > ble-scan` the result is not a logical error ( the chip controller can not find a matter accessory because it does not exist), the result is a programming error which is written below:

`Exception ignored in: <function FabricAdmin.__del__ at 0xb5a47028>
Traceback (most recent call last):
  File "/root/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/FabricAdmin.py", line 125, in __del__
    self.Shutdown()
  File "/root/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/FabricAdmin.py", line 118, in Shutdown
    if (self._isActive):
AttributeError: 'FabricAdmin' object has no attribute '_isActive'`

and If i run the `chip-device-ctrl` without any arguments I get this python error:
`__init__() missing 1 required positional argument: 'vendorId'
Failed to bringup CHIPDeviceController CLI`

Now according to your suggestions I can just use chip-repl instead of chip-device-ctrl ... i get that but it would be nice to either change the guide or fix the issues with the chip-device-ctrl code.

mossmgr, 2022-11-24:
Update: I build the 'chip-all-clusters-app' on my Macbook (intel) and i am running the executable. It is advertising on the local network. I also compiled the chip-tool for my arm device and try to commission over WIFI the chip-all-clusters-app . 

When i run `./chip-tool pairing onnetwork 333221 20202021` I expect that the all clusters app would be commissioned into the network. however this does not happen. instead i receive `Timeout waiting for mDNS resolution.`

cecille, 2022-11-24:
The guide has been updated to indicate that chip-device-ctrl is deprecated.

Can you post the logs from both sides, and indicate how you determined that all-clusters was advertising?

mossmgr, 2022-11-24:
Yes. After a system shutdown, I tried the chip-all-clusters project again and now I am able to see the listed device in the output of the `chip-tool discover commissionables` however the last output log of chip-tool is still `Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:513: CHIP Error 0x00000032: Timeout`

The output of chip-all-clusters-app running on my Macbook.

`[1669303867834] [6383:43091] CHIP: [DL] KVS will be written to: file:///tmp/chip_kvs
[1669303867855] [6383:43091] CHIP: [SPT] *** WARNING: Using temporary passcode 20202021 due to no neither --passcode or --spake2p-verifier-base64 given on command line. This is temporary and will disappear. Please update your scripts to explicitly configure onboarding credentials. ***
[1669303867855] [6383:43091] CHIP: [SPT] *** WARNING: Using temporary test discriminator 3840 due to --discriminator not given on command line. This is temporary and will disappear. Please update your scripts to explicitly configure discriminator. ***
[1669303867855] [6383:43091] CHIP: [SPT] PASE PBKDF iterations set to 1000
[1669303867855] [6383:43091] CHIP: [SPT] LinuxCommissionableDataProvider didn't get a PASE salt, generating one.
[1669303867859] [6383:43091] CHIP: [DL] Device Configuration:
[1669303867860] [6383:43091] CHIP: [DL]   Serial Number: TEST_SN
[1669303867860] [6383:43091] CHIP: [DL]   Vendor Id: 65521 (0xFFF1)
[1669303867861] [6383:43091] CHIP: [DL]   Product Id: 32769 (0x8001)
[1669303867861] [6383:43091] CHIP: [DL]   Hardware Version: 0
[1669303867861] [6383:43091] CHIP: [DL]   Setup Pin Code (0 for UNKNOWN/ERROR): 20202021
[1669303867861] [6383:43091] CHIP: [DL]   Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 3840 (0xF00)
[1669303867862] [6383:43091] CHIP: [DL]   Manufacturing Date: (not set)
[1669303867862] [6383:43091] CHIP: [DL]   Device Type: 65535 (0xFFFF)
[1669303867862] [6383:43091] CHIP: [-] ==== Onboarding payload for Standard Commissioning Flow ====
[1669303867863] [6383:43091] CHIP: [SVR] SetupQRCode: [MT:-24J0AFN00KA0648G00]
[1669303867863] [6383:43091] CHIP: [SVR] Copy/paste the below URL in a browser to see the QR Code:
[1669303867863] [6383:43091] CHIP: [SVR] https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3A-24J0AFN00KA0648G00
[1669303867863] [6383:43091] CHIP: [SVR] Manual pairing code: [34970112332]
[1669303867863] [6383:43091] CHIP: [SVR] Server initializing...
[1669303867863] [6383:43091] CHIP: [FP] Initializing FabricTable from persistent storage
[1669303867864] [6383:43091] CHIP: [TS] Last Known Good Time: 2022-11-24T11:51:28
[1669303867864] [6383:43091] CHIP: [DMG] AccessControl: initializing
[1669303867864] [6383:43091] CHIP: [DMG] Examples::AccessControlDelegate::Init
[1669303867864] [6383:43091] CHIP: [DMG] AccessControl: setting
[1669303867864] [6383:43091] CHIP: [DMG] DefaultAclStorage: initializing
[1669303867865] [6383:43091] CHIP: [DMG] DefaultAclStorage: 0 entries loaded
[1669303867865] [6383:43091] CHIP: [IN] UDP::Init bind&listen port=5540
[1669303867865] [6383:43091] CHIP: [IN] UDP::Init bound to port=5540
[1669303867865] [6383:43091] CHIP: [IN] UDP::Init bind&listen port=5540
[1669303867865] [6383:43091] CHIP: [IN] UDP::Init bound to port=5540
[1669303867865] [6383:43091] CHIP: [IN] TransportMgr initialized
[1669303867871] [6383:43091] CHIP: [ZCL] Using ZAP configuration...
[1669303867871] [6383:43091] CHIP: [DMG] Failed to read stored attribute (0, 0x0000_0028, 0x0000_0005: ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/platform/Darwin/KeyValueStoreManagerImpl.mm:222: CHIP Error 0x000000A0: Value not found in the persisted storage
[1669303867872] [6383:43091] CHIP: [DMG] Failed to read stored attribute (0, 0x0000_0028, 0x0000_0010: ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/platform/Darwin/KeyValueStoreManagerImpl.mm:222: CHIP Error 0x000000A0: Value not found in the persisted storage
[1669303867873] [6383:43091] CHIP: [DMG] Failed to read stored attribute (0, 0x0000_002C, 0x0000_0000: ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/platform/Darwin/KeyValueStoreManagerImpl.mm:222: CHIP Error 0x000000A0: Value not found in the persisted storage
[1669303867873] [6383:43091] CHIP: [DMG] Failed to read stored attribute (0, 0x0000_002C, 0x0000_0001: ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/platform/Darwin/KeyValueStoreManagerImpl.mm:222: CHIP Error 0x000000A0: Value not found in the persisted storage
[1669303867873] [6383:43091] CHIP: [DMG] Failed to read stored attribute (0, 0x0000_002D, 0x0000_0000: ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/platform/Darwin/KeyValueStoreManagerImpl.mm:222: CHIP Error 0x000000A0: Value not found in the persisted storage
[1669303867874] [6383:43091] CHIP: [DMG] Failed to read stored attribute (1, 0x0000_0006, 0x0000_0000: ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/platform/Darwin/KeyValueStoreManagerImpl.mm:222: CHIP Error 0x000000A0: Value not found in the persisted storage
[1669303867874] [6383:43091] CHIP: [DMG] Failed to read stored attribute (1, 0x0000_0006, 0x0000_4003: ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/platform/Darwin/KeyValueStoreManagerImpl.mm:222: CHIP Error 0x000000A0: Value not found in the persisted storage
[... cut ...]
[1669303914.470003][6781:6781] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1669303914.470391][6781:6781] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1669303914.470698][6781:6781] CHIP:DL: Inet Layer shutdown
[1669303914.470881][6781:6781] CHIP:DL: BLE shutdown
[1669303914.471042][6781:6781] CHIP:DL: System Layer shutdown
[1669303914.471740][6781:6781] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:513: CHIP Error 0x00000032: Timeout`

cecille, 2022-11-24:
Good to hear. If you check the logs, it looks like the discover command was able to find things. Not sure why it's giving a timeout at the end, but if you're able to commission then the commissioner and device are both fine.

tunm1228, 2023-10-03:
> Good to hear. If you check the logs, it looks like the discover command was able to find things. Not sure why it's giving a timeout at the end, but if you're able to commission then the commissioner and device are both fine.

Pair devices via IP. Specifically the knx device. It has a control address of "1/1/1". I don't know how to put it together. Please guide me