project-chip/connectedhomeip #9551: Create the dedicated pool for global dirty path list

yunhanw-google, 2021-09-08:
I also don't think it's a wise idea to have this global dirty rely on the same ClusterInfo resource pool that is used for allocating paths on incoming subscribe/read processing. This couples one onto the other - if there are a large number of subscribers who use up the majority of the pool, your dirty tracking system is not going to function well, if it all.

They have separate purposes, so having separate pools would be more sensible. Fine to do this as a follow-up issue.

_Originally posted by @mrjerryjohns in https://github.com/project-chip/connectedhomeip/pull/9510#discussion_r704793327_

mrjerryjohns, 2021-11-19:
This is pretty important to get a functioning subscription going.

tcarmelveilleux, 2022-01-31:
@mrjerryjohns Says it's done

closed 2022-01-31 by tcarmelveilleux

mrjerryjohns, 2022-01-31:
Implemented in #13472