project-chip/connectedhomeip #9563: ESP32&ESP32C3 crash on system layer

wqx6, 2021-09-09:
#### Problem
After PR #9366 merged, ESP32 & ESP32C3 fails when booting.
ESP32 error logs:
```
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.                                                                    
                                                                             
Core  1 register dump:                                                       
PC      : 0x00000001  PS      : 0x00060630  A0      : 0x800f2e40  A1      : 0x3ffe3040                                                                     
A2      : 0x00000001  A3      : 0x3ffcdd64  A4      : 0x00000001  A5      : 0x00000000                                                                     
A6      : 0x3ffbc7a0  A7      : 0x000000c0  A8      : 0x800ef078  A9      : 0x00000001                                                                     
A10     : 0x3ffcdd64  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3ffe3060                                                                     
A14     : 0x3ffe3040  A15     : 0x3ffe2fb0  SAR     : 0x00000007  EXCCAUSE: 0x00000014                                                                     
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000                                                                     
                                                                             
Backtrace:0xfffffffe:0x3ffe3040 0x400f2e3d:0x3ffe3060 0x400f2e8e:0x3ffe3080 0x400f3056:0x3ffe30a0 0x400f307d:0x3ffe3100 0x40093c15:0x3ffe3120              
0x400f2e3d: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToSystemLayer(chip::DeviceLayer::ChipDeviceEvent const*) at /home/wangqixiang/workspace/mygithub/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/
esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.cpp:262                                                         
                                                                             
0x400f2e8e: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/wangqixiang/workspace/mygithub/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_
party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.cpp:222                                                                     
                                                                             
0x400f3056: chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/wangqixiang/workspace/mygithub/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:3
64                                                                           
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at /home/wangqixiang/workspace/mygithub/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/inclu
de/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:179             

0x400f307d: chip::DeviceLayer::PlatformManager::RunEventLoop() at /home/wangqixiang/workspace/mygithub/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:296
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /home/wangqixiang/workspace/mygithub/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/
src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:204    

0x40093c15: vPortTaskWrapper at /home/wangqixiang/workspace/mygithub/esp-idf/components/freertos/port/xtensa/port.c:168                                    
```
ESP32C3 error logs:
```
Stack dump detected                  
Core  0 register dump:                                                                                                                                     
MEPC    : 0x3fcc0000  RA      : 0x420217ea  SP      : 0x3fcb61e0  GP      : 0x3fc8f31c                                                                     
0x420217ea: chip::System::LayerImplLwIP::HandleEvent(chip::System::Object&, int, unsigned int) at /home/wangqixiang/workspace/mygithub/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/system/SystemLayerImplLwIP.cpp:219
                                                                             
TP      : 0x3fc855a0  T0      : 0x00000000  T1      : 0x00000000  T2      : 0x00000003  
S0/FP   : 0x3fc9d594  S1      : 0x3fca5a88  A0      : 0x3fc9d62c  A1      : 0x00000001                                                                     
A2      : 0x00000000  A3      : 0x00000004  A4      : 0x600c2000  A5      : 0x3fcc0000  
A6      : 0xc8000000  A7      : 0x00000064  S2      : 0x3fc9d62c  S3      : 0x00000001  
S4      : 0x000000c0  S5      : 0x00000000  S6      : 0x3c0f0000  S7      : 0x00000000                                                                     
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000                                                                     
T3      : 0x00000001  T4      : 0x00000000  T5      : 0x3fc9d62c  T6      : 0x00000001  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000001  MTVAL   : 0x3fcc0000  
0x40380001: _vector_table at ??:?        
                                                                                                                                                           
MHARTID : 0x00000000                                          
                                                                             
Failed to run gdb_panic_server.py script: Command '['riscv32-esp-elf-gdb', '--batch', '-n', '/home/wangqixiang/workspace/mygithub/connectedhomeip/examples/all-clusters-app/esp32/build/chip-all-clusters-app.elf', '-ex', 'target remote | "/home/wangqixiang/.espressif/python_env/idf4.3_py3.8_env/bin/python" "/hom
e/wangqixiang/workspace/mygithub/esp-idf/tools/../tools/gdb_panic_server.py" --target esp32c3 "/tmp/tmpajv0hqij"', '-ex', 'bt']' returned non-zero exit status 127.
b'riscv32-esp-elf-gdb: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory\n'
                                                                                                                                                           
                                                                             
Core  0 register dump:                                                                                                                                     
MEPC    : 0x3fcc0000  RA      : 0x420217ea  SP      : 0x3fcb61e0  GP      : 0x3fc8f31c  
TP      : 0x3fc855a0  T0      : 0x00000000  T1      : 0x00000000  T2      : 0x00000003                                      
S0/FP   : 0x3fc9d594  S1      : 0x3fca5a88  A0      : 0x3fc9d62c  A1      : 0x00000001  
A2      : 0x00000000  A3      : 0x00000004  A4      : 0x600c2000  A5      : 0x3fcc0000  
A6      : 0xc8000000  A7      : 0x00000064  S2      : 0x3fc9d62c  S3      : 0x00000001                                                                     
S4      : 0x000000c0  S5      : 0x00000000  S6      : 0x3c0f0000  S7      : 0x00000000                                                                     
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000001  T4      : 0x00000000  T5      : 0x3fc9d62c  T6      : 0x00000001  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000001  MTVAL   : 0x3fcc0000  
MHARTID : 0x00000000  

Stack memory:
3fcb61e0: 0x00000000 0x3fc9d000 0x3fc9dad4 0x3fc9dadc 0x3fc9dacc 0x000e7ef0 0x3fcb6240 0x420265a4
[... 7 lines cut ...]
[... cut ...]
3fcb6520: 0x3fcb6550 0x3fcb6640 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4038229e
3fcb6540: 0x00000000 0x00000000 0x00000000 0x00000054 0x3fcb6550 0x3fcb6550 0x3fcb6550 0x3fcb6550
3fcb6560: 0x00000000 0x3fcb6568 0xffffffff 0x3fcb6568 0x3fcb6568 0x00000000 0x3fcb657c 0xffffffff
3fcb6580: 0x3fcb657c 0x3fcb657c 0x00000001 0x00000001 0x00000000 0x0000ffff 0x00000000 0x3fcb65a0
3fcb65a0: 0x3fcb6548 0x00000054 0x00000000 0x3fcb65a8 0x00000000 0x00000000 0x00000000 0x3fcb65c0
3fcb65c0: 0xffffffff 0x3fcb65c0 0x3fcb65c0 0x00000000 0x3fcb65d4 0xffffffff 0x3fcb65d4 0x3fcb65d4
```

ClockRight, 2021-09-09:
M5Stack also fails on booting. I am pretty sure it is the same problem.

```
...
I (1611) chip[DL]: CHIPoBLE advertising started
I (1621) app-devicecallbacks: Unhandled cluster ID: 768
E (1621) chip[DL]: Long dispatch time: 710 ms, for event type 2
uru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00000001  PS      : 0x00060230  A0      : 0x800f5dec  A1      : 0x3ffe6ee0  
A2      : 0x00000001  A3      : 0x3ffd0d9c  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x3ffbc710  A7      : 0x000000c0  A8      : 0x800f1fcc  A9      : 0x00000001  
A10     : 0x3ffd0d9c  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3ffe6f00  
A14     : 0x3ffe6ee0  A15     : 0x3ffe6e50  SAR     : 0x00000007  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace:0xfffffffe:0x3ffe6ee0 0x400f5de9:0x3ffe6f00 0x400f5e3a:0x3ffe6f20 0x400f5ffe:0x3ffe6f40 0x400f6025:0x3ffe6fa0 0x40095235:0x3ffe6fc0
0x400f5de9: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToSystemLayer(chip::DeviceLayer::ChipDeviceEvent const*) at /home/dominik/Projects/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.cpp:262

0x400f5e3a: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/dominik/Projects/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.cpp:222

0x400f5ffe: chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /home/dominik/Projects/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:364
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at /home/dominik/Projects/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:179

0x400f6025: chip::DeviceLayer::PlatformManager::RunEventLoop() at /home/dominik/Projects/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:296
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /home/dominik/Projects/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:204

0x40095235: vPortTaskWrapper at /home/dominik/tools/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: 69dd9177eb46ab53

Rebooting...
```

kpschoedel, 2021-09-09:
Current status: the crash is in `LayerImplLwIP::HandleEvent()` at
```
lReturn        = lEventDelegate->GetFunction()(aTarget, aEventType, aArgument);
```
as the pointer returned by `GetFunction()` is invalid.

PR #9569 (2021-09-09): Fix EventHandlerDelegate crash on LwIP platforms
#### Problem

3a505010973c #9366 caused a crash. In `AddEventHandlerDelegate()`,
`lDelegate` was accidentally a stack variable instead of a reference to
the caller-supplied storage.

This affects all platforms using LwIP.

#### Change overview

Add one small but important `&`.

Fixes #9563 _ESP32&ESP32C3 crash on system layer_

#### Testing

all-clusters-app on M5Stack
commit a0106ab9 (2021-09-09): Fix EventHandlerDelegate crash (#9569)

#### Problem

3a505010973c #9366 caused a crash. In `AddEventHandlerDelegate()`,
`lDelegate` was accidentally a stack variable instead of a reference to
the caller-supplied storage.

This affects all platforms using LwIP.

#### Change overview

Add one small but important `&`.

Fixes #9563 _ESP32&ESP32C3 crash on system layer_

#### Testing

all-clusters-app on M5Stack
closed 2021-09-09 by andy31415