project-chip/connectedhomeip #9586: Add API to handle and process IM events.

yunhanw-google, 2021-09-10:
#### Problem
Need API to handle and process iM events in controller side.

trombadore, 2021-11-02:
This is blocking Matter Test Harness.

For certain controller certification tests, a simulated node needs to be able to handle events like StartUp, ShutDown and Leave. @vivien-apple's simulated nodes for these controller tests leverage the matter stack to handle events. Therefore this bug needs to close so that the Matter stack supports events and thus the simulated nodes leveraged by the test harness can handle events as well.

mrjerryjohns, 2021-11-19:
Fixed in #11615

mrjerryjohns, 2021-11-19:
If there are further requirements on the controller side, please open a new issue.

closed 2021-11-19 by mrjerryjohns