project-chip/connectedhomeip #9598: Implement OTA Provider and Requestor device types

holbrookt, 2021-09-10:
#### Problem
Spec defines device types for OTA Provider and OTA Requestor.

These two clusters are very closely related, and these device types help simplify OTA interactions. Most devices that wish to receive OTA updates should be Provider clients and Requestor servers, and the opposite is true for devices wishing to serve OTA updates.

 #### Proposed Solution
Add device types for OTA Provider and OTA Requestor

selissia, 2022-01-14:
Comment from Trevor: 
found one small issue with the device type for OTA Requestor:
            <include cluster="OTA Software Update Requestor" client="false" server="true" clientLocked="true" serverLocked="true"></include>
            <include cluster="OTA Software Update Provider" client="false" server="true" clientLocked="true" serverLocked="true"></include>
should be a client of OTA Provider cluster not server

carol-apple, 2022-01-26:
Fixed by https://github.com/project-chip/connectedhomeip/pull/13718

closed 2022-01-26 by carol-apple