project-chip/connectedhomeip #9601: Ethernet Network Diagnostics cluster name change, missing attributes, and unsupported attributes response from TE5 SDK repo

jaehs6sam, 2021-09-10:
#### Problem
<
During TE5 test case verification using chip-device-ctrl command of SDK TE5 branch, the following issues were found while using linghting-app as DUT
ex) command line: chip-device-ctrl > zclread EthernetNetworkDiagnostics PHYRate 10 1 0

1. Cluster name needs to be changed from EthernetNetworkDiagnostics → DiagnosticsNetworkEthernet
2. The following attributes return with unsupported attribute message response:
    PacketRxCount, PacketTxCount, TxErrCount, CollisionCount, OverrunCount, ClusterRevision
3. The following attributes are missing:
    PHYRate, FullDuplex, CarrierDetect, TimeSinceReset

Specification References:
11.15.2
System Configuration
SDK TE5 branch with Raspi (Controller and controlee) lighting-app as DUT

[image]

 >

 #### Proposed Solution

yufengwangca, 2021-09-17:
@jaehs6sam 

1. Cluster name needs to be changed from EthernetNetworkDiagnostics → DiagnosticsNetworkEthernet
[Yufeng]: PR will be posted by end of today.

2. The following attributes return with unsupported attribute message response:
PacketRxCount, PacketTxCount, TxErrCount, CollisionCount, OverrunCount, ClusterRevision
[Yufeng]: Which branch do you test this feature? This is planed in TE6

3. The following attributes are missing:
PHYRate, FullDuplex, CarrierDetect, TimeSinceReset
They are all optional attributes, the ZAP UI right now does not allow me enable optional features. Synced with Chaitanya and we will focus mandatory attribute in TE6.

jaehs6sam, 2021-09-17:
1. OK.
2. It was verified with TE5 branch since it was originally intended for TE5.
3. OK. Only mandatory diagnostics attributes were to be tested.  No plan for optional attribute support?  Should we remove all those optional attribute test cases, then?

yufengwangca, 2021-09-17:
@jaehs6sam  Could you mark all those optional attribute test cases as optional?

jaehs6sam, 2021-09-17:
@yufengwangca They are all labeled as optional in test plan document.  And there is no mention of conformance in test verification spreadsheet document.

yufengwangca, 2021-09-20:
The optional attributes are not available in chip-tool, and I am not able to enable them via ZAP UI. This apply to all clusters. So we decided to focus on those mandatory attributes at TE6 and follow-up with those optional attributes at TE7.

Regarding the naming change, it is caused by the cluster name and its cluster identifier name are not aligned in the spec, filed CHIP-Specifications/connectedhomeip-spec#4155 for follow-up.

closed 2021-09-20 by yufengwangca