project-chip/connectedhomeip #9623: TE5-chip-device-ctrl Commissioning failed！GDBus.Error:org.bluez.Error.Failed: Software caused connection abort.

luyou-tuya, 2021-09-11:
#### Problem
@jmartinez-silabs
Can you help me find out what caused it, how do I need to solve this problem here?
I tried the link "https://github.com/project-chip/connectedhomeip/issues/8416" mentioned method But no use。

the chip-device-ctrl log:
(python_env) chengqiang@chengqiang-desktop:~/Desktop/connectedhomeip$ sudo out/python_env/bin/chip-device-ctrl
[1631354646.580023][20603:20603] CHIP:CTL: Generating NOC
[1631354646.580446][20603:20603] CHIP:CTL: Generating ICAC
[1631354646.580776][20603:20603] CHIP:CTL: Generating RCAC
[1631354646.603597][20603:20603] CHIP:DL: Avahi client registered
[1631354646.605404][20603:20603] CHIP:ZCL: Using ZAP configuration...
[1631354646.605509][20603:20603] CHIP:ZCL: deactivate report event
[1631354646.606354][20603:20603] CHIP:DIS: Verifying the received credentials
[1631354646.608306][20603:20603] CHIP:DIS: Added new fabric at index: 1, Initialized: 1
[1631354646.608370][20603:20603] CHIP:DIS: Assigned fabric ID: 0x0000000000000000, node ID: 0x000000000001B669
[1631354646.608427][20603:20603] CHIP:CTL: Joined the fabric at index 1
[1631354646.621630][20603:20630] CHIP:DL: Platform main loop started.
Chip Device Controller Shell

chip-device-ctrl > connect -ble 3840 73141520 1234
Device is assigned with nodeid = 1234
[1631354666.679329][20603:20621] CHIP:BLE: BLE removing known devices.
[1631354666.679415][20603:20621] CHIP:BLE: BLE initiating scan.
[1631354667.330287][20603:20621] CHIP:BLE: New device scanned: 90:FD:9F:19:B7:75
[1631354667.330407][20603:20621] CHIP:BLE: Device discriminator match. Attempting to connect.
[1631354667.338853][20603:20621] CHIP:BLE: Scan complete notification without an active scan.
[1631354669.232448][20603:20621] **CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.Failed: Software caused** connection abort
Failed to establish secure session to device: 50
[1631354696.661662][20603:20630] CHIP:SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
CHIP Error 0x00000032: Timeout

The device log ：
 chip-efr32-lighting-example starting
00> <efr32 > ==================================================
00> <efr32 > Init CHIP Stack
00> <detail> [-] Initializing key value store
00> <detail> [-] KeyValueStore init complete: active keys 4, deleted keys 1, sectors 4, logical sector size 2048 bytes
00> <info  > [DL] Setting device name to : "EFR32_LIGHT"
00> <efr32 > Initializing OpenThread stack
00> <info  > [DL] OpenThread started: OK
00> <info  > [DL] Setting OpenThread device type to ROUTER
00> <efr32 > Starting Platform Manager Event Loop
00> <efr32 > Starting OpenThread task
00> <efr32 > Starting App Task
00> <efr32 > Starting FreeRTOS scheduler
00> <info  > [DL] Bluetooth stack booted: v3.2.1-b216
00> <info  > [DL] RAIL version:, v2.11.1-b0
00> <detail> [DL] Thread task running
00> <detail> [DL] CHIP task running
00> <detail> [DL] Start BLE advertissement
00> <info  > [ZCL] Using ZAP configuration...
00> <info  > [DL] _OnPlatformEvent default:  event->Type = 32779
00> <detail> [DL] OpenThread Sta<detail> [DL] bleAdv Timeout : Start slow advertissment
00> <info  > [DL] Connect Event for handle : 1
00> <info  > [DL] evt_UNKNOWN id = 080600a0
00> <info  > [DL] Disconnect Event for handle : 1
00> <info  > [DL] BLE GATT connection closed (con 1, reason 4104)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEConnectionError
00> <detail> [DL] Start BLE advertissement
00> <efr32 > Turning light ON
00> <efr32 > Light ON
00> <detail> [DL] bleAdv Timeout : Start slow advertissment
00> <efr32 > Turning light OFF
00> <efr32 > Light OFF
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

jmartinez-silabs, 2021-09-13:
This is usually the ble bluez package failing on the raspi. 
I get his error form time to time but usually just doing `exit` in the controller and starting it again work for me.
You can also try the following if the error persist :

sudo btmgmt -i hci0 power off
sudo btmgmt -i hci0 bredr off
sudo btmgmt -i hci0 power on

then retry.
Also rebooting the PI that some times help.

luyou-tuya, 2021-09-15:
> This is usually the ble bluez package failing on the raspi.
> I get his error form time to time but usually just doing `exit` in the controller and starting it again work for me.
> You can also try the [...]

Thanks for you help！

closed 2021-09-15 by luyou-tuya