project-chip/connectedhomeip #9634: TE5-chip-device-ctrl Commissioning CASE failed！

luyou-tuya, 2021-09-13:
#### Problem
@jmartinez-silabs 
 I run python controller on raspberry pi:
Linux ubuntu-desktop 5.8.0-1032-raspi #35-Ubuntu SMP PREEMPT Wed Jul 14 10:51:21 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
device ：
efr32-lighting-example
I get an error when I execute the following command, and the log of the controller is as follows：
chip-device-ctrl > resolve -5643169321044569094 1234
[1631522258.435171][16320:16328] CHIP:DL: < MdnsImpl.cpp:687 >[HandleResolve]Avahi resolve found
[1631522258.435377][16320:16328] CHIP:DIS: < Resolver.h:47 >[LogNodeIdResolved]Node ID resolved for 0x00000000000004D2 to [2001:db8::d31b:c44d:ebb8:b182]:5540
Node address has been updated
[1631522258.435980][16320:16328] CHIP:CTL: < CHIPDeviceController.cpp:1055 >[OperationalDiscoveryComplete]OperationalDiscoveryComplete for device ID 1234
[1631522264.159690][16320:16328] CHIP:EM: < ReliableMessageMgr.cpp:153 >[ExecuteActions]Failed to Send CHIP MsgId:00000000 sendCount: 3 max retries: 3
[1631522268.441371][16320:16328] CHIP:SC: < CASESession.cpp:266 >[OnResponseTimeout]CASESession timed out while waiting for a response from the peer. Expected message typ                                     e was 49
[1631522268.441476][16320:16328] CHIP:CTL: < CHIPDeviceController.cpp:1747 >[OnDeviceConnectionFailureFn]Device connection failed. Error ../../src/protocols/secure_channe                                     l/CASESession.cpp:269: CHIP Error 0x00000032: Timeout
Failed to commission: 50

the device log：
 SRP Client was started, detected server: fdc3:5899:2e56:b952:27dc:e11c:687f:4615
00> <info  > [DL] _OnPlatformEvent default:  event->Type = 32779
00> <detail> [DL] OpenThread State Changed (Flags: 0x200012a4)
00> <detail> [DL]    Device Role: CHILD
00> <detail> [DL]    Partition Id: 0x73966019
00> <info  > [DL] _OnPlatformEvent default:  event->Type = 32779
00> <detail> [DL] OpenThread State Changed (Flags: 0x00000001)
00> <detail> [DL]    Thread Unicast Addresses:
00> <detail> [DL]         2001:DB8::D31B:C44D:EBB8:B182/64 valid preferred
00> <detail> [DL]         FD26:38E2:1723:141B:0:FF:FE00:6804/64 valid rloc
00> <detail> [DL]         FD26:38E2:1723:141B:3B14:B0D9:EFD3:E22/64 valid
00> <detail> [DL]         FE80::FC42:AD50:385:1B46/64 valid preferred
00> <detail> [DL] LwIP Thread interface addresses updated
00> <detail> [DL]    FE80::FC42:AD50:385:1B46 IPv6 link-local address, preferred)
00> <detail> [DL]    FD26:38E2:1723:141B:3B14:B0D9:EFD3:E22 Thread mesh-local address)
00> <detail> [DL]    2001:DB8::D31B:C44D:EBB8:B182 IPv6 global unicast address, preferred)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLENotifyConfirm
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 23
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server response
00> <info  > [DL] HandleTXcharCCCDWrite - Config Flags value : 0
00> <info  > [DL] CHIPoBLE unsubscribe received
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEUnsubscribe
00> <detail> [IN] Clearing BLE pending packets.
00> <info  > [BLE] Releasing end point's BLE connection back to application.
00> <info  > [DL] Disconnect Event for handle : 1
00> <info  > [DL] BLE GATT connection closed (con 1, reason 4115)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEConnectionError
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server response
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server response
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server response
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server response
00> <error > [DL] OnSrpClientNotification: Timed out waiting on server respons

 #### Proposed Solution
I think the thread failed because the network is blocked. I ping 2001:DB8::D31B:C44D:EBB8:B182 and there is no response. But I don’t know why the network is [blocked。]
the wpan0 ipaddr:
wpan0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1280
        inet6 fdc3:5899:2e56:b952:0:ff:fe00:fc00  prefixlen 64  scopeid 0x0<global>
        inet6 fdc3:5899:2e56:b952:27dc:e11c:687f:4615  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::9c01:eecf:7661:21c8  prefixlen 64  scopeid 0x20<link>
        inet6 fdc3:5899:2e56:b952:0:ff:fe00:fc11  prefixlen 64  scopeid 0x0<global>
        inet6 2001:db8::a840:8488:cebe:b136  prefixlen 64  scopeid 0x0<global>
        inet6 fdc3:5899:2e56:b952:0:ff:fe00:fc10  prefixlen 64  scopeid 0x0<global>
        inet6 fdc3:5899:2e56:b952:0:ff:fe00:6800  prefixlen 64  scopeid 0x0<global>
        inet6 fdc3:5899:2e56:b952:0:ff:fe00:fc38  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 1  bytes 56 (56.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 155  bytes 16196 (16.1 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

<suggested fix, suggested enhancement>

jmartinez-silabs, 2021-09-13:
Hi,

I am a bit suprised by that fabric ID `resolve **-5643169321044569094** 1234`. Are you running with the current master of the TE5 branch?

Anyhow, you seem to have a routing problem on your network if you can't ping your wpan0 address. CASE session is established through IP so if the efr32 device and the controller cannot reach each other over the network the CASE session will timeout like you are seeing.

Can you describe your setup? was is running your OTBR. is it on a different raspberry pi. Can your device running the controller ping the otbr device? Did you define any routing to `2001:db8::`

jmartinez-silabs, 2021-09-13:
So you are running with a rather recent commit hash and not TE5 branch because compressed fabric ID is not in TE5 branch and we use fabric ID 0 to resolve for TE5. On master recent commits yes you have to use the compressed fabric id and you can get it with `get-fabricid`

It is hard for me to determine your routing issues on your network and even more without knowing your setup.

When the device is commissionned to the Thread network, you should be able to ping from your controller raspi to the Device IPV6. 

For example. I have OTBR running on one raspi 4 and the controller on the second raspi4. Both are connected to the same ethernet switch. I configured the OTBR to use eth0

I personnaly use the global prefix `fd11:22::` when setting up my thread network with the otbr. Is `2001:db8` conciderate a global ULA prefix? I would have to validate

On the controller I make a route to my prefix 
`sudo ip -6 route add fd11:22::/64 dev eth0`

jmartinez-silabs, 2021-09-13:
In your device log i see that the SRP is failling to start after joinning the thread network.
`OnSrpClientNotification: Timed out waiting on server response`

What OTBR commit hash do you use? if SRP isn't working properly, resolve wont be successful

luyou-tuya, 2021-09-14:
> In your device log i see that the SRP is failling to start after joinning the thread network.
> `OnSrpClientNotification: Timed out waiting on server response`
> 
> What OTBR commit hash do you use? if SRP isn't working properly, resolve wont be successful

the python conntorl:commit 6db59aa066599f9e5967e900c809a85037a56e8b (HEAD -> test_event_5, origin/test_event_5)

luyou-tuya, 2021-09-14:
> In your device log i see that the SRP is failling to start after joinning the thread network.
> `OnSrpClientNotification: Timed out waiting on server response`
> 
> What OTBR commit hash do you use? if SRP isn't working properly, resolve wont be successful

The python controller use :commit 6db59aa066599f9e5967e900c809a85037a56e8b (HEAD -> test_event_5, origin/test_event_5)
the otbr commit: commit 17cdb3bad795db87bfcd2c272366ab5a4f270ffc (HEAD -> hrt)

closed 2021-09-15 by luyou-tuya