project-chip/connectedhomeip #9670: transitionTime of MoveToLeve doesn't take effect

lucyShi0116, 2021-09-14:
#### Problem Description:
1. Check DUT CurrentLevel is 0
2. Send the cmd to DUT successfully: zcl LevelControl MoveToLevel 12344321 1 0 level=254 transitionTime=65535 optionMask=0 optionOverride=0
3. Check the DUT CurrentLevel
 * expected behavior
 DUT dimming level goes from 0 to 254, cost 6553.5s 
 * actual behavior
 transitionTime of MoveToLeve doesn't take effect
 * steps to reproduce
 Refer to 'Problem Description'
 * system configuration
 * logs:
`chip-device-ctrl > zclread LevelControl CurrentLevel 12344321 1 0
[1631509027.056155][2526:2535] CHIP:DMG: SendReadRequest: Client[0] [ INIT]
[1631509027.056465][2526:2535] CHIP:IN: Secure message was encrypted: Msg ID 66
[1631509027.056540][2526:2535] CHIP:IN: Encrypted message 0xd84ab98 to 0x0000000000BC5C01 of type 2 and protocolId 5 on exchange 39057.
[1631509027.056684][2526:2535] CHIP:IN: Sending msg 0xd84ab98 to 0x0000000000BC5C01 at utc time: 8661878 msec
[1631509027.056740][2526:2535] CHIP:IN: Sending secure msg on generic transport
[1631509027.056937][2526:2535] CHIP:IN: Secure msg send status ../../src/inet/IPEndPointBasis.cpp:915: Success
[1631509027.057033][2526:2535] CHIP:DMG: Client[0] moving to [AwaitingResponse]
[1631509027.220014][2526:2535] CHIP:EM: Received message of type 0x05 with vendorId 0x0000 and protocolId 0x0005 on exchange 39057
[1631509027.220164][2526:2535] CHIP:EM: Rxd Ack; Removing MsgId:00000042 from Retrans Table
[1631509027.220225][2526:2535] CHIP:EM: Removed CHIP MsgId:00000042 from RetransTable
[1631509027.220340][2526:2535] CHIP:DMG: ReportData =
[1631509027.220406][2526:2535] CHIP:DMG: {
[1631509027.220455][2526:2535] CHIP:DMG:        AttributeDataList =
[1631509027.220512][2526:2535] CHIP:DMG:        [
[1631509027.220569][2526:2535] CHIP:DMG:                AttributeDataElement =
[1631509027.220641][2526:2535] CHIP:DMG:                {
[1631509027.220695][2526:2535] CHIP:DMG:                        AttributePath =
[1631509027.220812][2526:2535] CHIP:DMG:                        {
[1631509027.220884][2526:2535] CHIP:DMG:                                NodeId = 0xbc5c01,
[1631509027.220955][2526:2535] CHIP:DMG:                                EndpointId = 0x1,
[1631509027.221038][2526:2535] CHIP:DMG:                                ClusterId = 0x8,
[1631509027.221121][2526:2535] CHIP:DMG:                                FieldTag = 0x0,
[1631509027.221199][2526:2535] CHIP:DMG:                        }
[1631509027.221281][2526:2535] CHIP:DMG:
[1631509027.221367][2526:2535] CHIP:DMG:                        Data = 0,
[1631509027.221446][2526:2535] CHIP:DMG:                        DataElementVersion = 0x0,
[1631509027.221515][2526:2535] CHIP:DMG:                },
[1631509027.221592][2526:2535] CHIP:DMG:
[1631509027.221652][2526:2535] CHIP:DMG:        ],
[1631509027.221725][2526:2535] CHIP:DMG:
[1631509027.221776][2526:2535] CHIP:DMG: }
[1631509027.222493][2526:2535] CHIP:ZCL: ReadAttributesResponse:
[1631509027.222563][2526:2535] CHIP:ZCL:   ClusterId: 0x0000_0008
[1631509027.222621][2526:2535] CHIP:ZCL:   attributeId: 0x0000_0000
[1631509027.222669][2526:2535] CHIP:ZCL:   status: Success                (0x0000)
[1631509027.222715][2526:2535] CHIP:ZCL:   attribute TLV Type: 0x04
[1631509027.222769][2526:2535] CHIP:ZCL:   attributeValue: 0
[1631509027.222953][2526:2535] CHIP:DMG: Client[0] moving to [INIT]
[1631509027.223013][2526:2535] CHIP:DMG: Client[0] moving to [UNINIT]
AttributeReadResult(path=AttributePath(nodeId=12344321, endpointId=1, clusterId=8, attributeId=0), status=0, value=0)
[1631509027.223074][2526:2535] CHIP:EM: Sending Standalone Ack for MsgId:00000022
[1631509027.223283][2526:2535] CHIP:IN: Secure message was encrypted: Msg ID 67
[1631509027.223358][2526:2535] CHIP:IN: Encrypted message 0xffff97ffdaf8 to 0x0000000000BC5C01 of type 16 and protocolId 0 on exchange 39057.
[1631509027.223432][2526:2535] CHIP:IN: Sending msg 0xffff97ffdaf8 to 0x0000000000BC5C01 at utc time: 8662045 msec
[1631509027.223575][2526:2535] CHIP:IN: chip-device-ctrl > Sending secure msg on generic transport
[1631509027.223823][2526:2535] CHIP:IN: Secure msg send status ../../src/inet/IPEndPointBasis.cpp:915: Success
[1631509027.223903][2526:2535] CHIP:EM: Flushed pending ack for MsgId:00000022

chip-device-ctrl >
chip-device-ctrl > zcl LevelControl MoveToLevel 12344321 1 0 level=254 transitionTime=65535 optionMask=0 optionOverride=0
[1631509032.902667][2526:2535] CHIP:DMG: ICR moving to [Initialize]
[1631509032.902769][2526:2535] CHIP:DMG: ICR moving to [AddCommand]
[1631509032.902961][2526:2535] CHIP:IN: Secure message was encrypted: Msg ID 68
[1631509032.903016][2526:2535] CHIP:IN: Encrypted message 0xd84ab98 to 0x0000000000BC5C01 of type 8 and protocolId 5 on exchange 39058.
[1631509032.903072][2526:2535] CHIP:IN: Sending msg 0xd84ab98 to 0x0000000000BC5C01 at utc time: 8667724 msec
[1631509032.903116][2526:2535] CHIP:IN: Sending secure msg on generic transport
[1631509032.903269][2526:2535] CHIP:IN: Secure msg send status ../../src/inet/IPEndPointBasis.cpp:915: Success
[... 74 lines cut ...]
[1631509035.197984][2526:2535] CHIP:DMG:                        }
[1631509035.198052][2526:2535] CHIP:DMG:
[1631509035.198112][2526:2535] CHIP:DMG:                        Data = 254,
[1631509035.198190][2526:2535] CHIP:DMG:                        DataElementVersion = 0x0,
[1631509035.198245][2526:2535] CHIP:DMG:                },
[1631509035.198302][2526:2535] CHIP:DMG:
[1631509035.198346][2526:2535] CHIP:DMG:        ],
[1631509035.198397][2526:2535] CHIP:DMG:
[1631509035.198434][2526:2535] CHIP:DMG: }
[1631509035.198983][2526:2535] CHIP:ZCL: ReadAttributesResponse:
[1631509035.199037][2526:2535] CHIP:ZCL:   ClusterId: 0x0000_0008
[1631509035.199080][2526:2535] CHIP:ZCL:   attributeId: 0x0000_0000
[1631509035.199116][2526:2535] CHIP:ZCL:   status: Success                (0x0000)
[1631509035.199150][2526:2535] CHIP:ZCL:   attribute TLV Type: 0x04
[1631509035.199192][2526:2535] CHIP:ZCL:   attributeValue: 254
[1631509035.199333][2526:2535] CHIP:DMG: Client[0] moving to [INIT]
[1631509035.199379][2526:2535] CHIP:DMG: Client[0] moving to [UNINIT]
[1631509035.199427][2526:2535] CHIP:EM: Sending Standalone Ack for MsgId:00000024
AttributeReadResult(path=AttributePath(nodeId=12344321, endpointId=1, clusterId=8, attributeId=0), status=0, value=254)
[1631509035.199566][2526:2535] CHIP:IN: Secure message was encrypted: Msg ID 71
[1631509035.199620][2526:2535] CHIP:IN: Encrypted message 0xffff97ffdaf8 to 0x0000000000BC5C01 of type 16 and protocolId 0 on exchange 39059.
[1631509035.199674][2526:2535] CHIP:IN: Sending msg 0xffff97ffdaf8 to 0x0000000000BC5C01 at utc time: 8670021 msec
[1631509035.199713][2526:2535] CHIP:IN: Sending secure msg on generic transport
chip-device-ctrl > [1631509035.199893][2526:2535] CHIP:IN: Secure msg send status ../../src/inet/IPEndPointBasis.cpp:915: Success
[1631509035.199951][2526:2535] CHIP:EM: Flushed pending ack for MsgId:00000024`

cindycaowiz, 2021-11-01:
Test on te6, still failed , set 30sec, but was 45~50sec, for test log:
'''
3.2.2. [TC-CC-3.1] Hue MoveTo functionality with server as DUT

************step1

chip-device-ctrl > zclread ColorControl CurrentHue 12344321 1 0
[1634191083.950794][4649:4656] CHIP:DMG: SendReadRequest: Client[0] [ INIT]
[1634191083.951116][4649:4656] CHIP:IN: Secure message was encrypted: Msg ID 58
[1634191083.951189][4649:4656] CHIP:IN: Build encrypted message 0x7062258 to 0x0000000000BC5C01 of type 2 and protocolId 1 on exchange 11550r with MessageCounter 58.
[1634191083.951273][4649:4656] CHIP:IN: Sending encrypted msg 0x7062258 to 0x0000000000BC5C01 at utc time: 5977097 msec
[1634191083.951338][4649:4656] CHIP:IN: Sending msg on generic transport
[1634191083.951614][4649:4656] CHIP:DMG: Client[0] moving to [AwaitingInitialReport]
[1634191084.104240][4649:4656] CHIP:EM: Received message of type 0x05 with vendorId 0x0000 and protocolId 0x0001 on exchange 11550i
[1634191084.104390][4649:4656] CHIP:EM: Rxd Ack; Removing MessageCounter:0000003A from Retrans Table
[1634191084.104445][4649:4656] CHIP:EM: Removed CHIP MessageCounter:0000003A from RetransTable
[1634191084.104546][4649:4656] CHIP:DMG: ReportData =
[1634191084.104606][4649:4656] CHIP:DMG: {
[1634191084.104653][4649:4656] CHIP:DMG:        AttributeDataList =
[1634191084.104709][4649:4656] CHIP:DMG:        [
[1634191084.104764][4649:4656] CHIP:DMG:                AttributeDataElement =
[1634191084.104827][4649:4656] CHIP:DMG:                {
[1634191084.104883][4649:4656] CHIP:DMG:                        AttributePath =
[1634191084.104994][4649:4656] CHIP:DMG:                        {
[1634191084.105061][4649:4656] CHIP:DMG:                                NodeId = 0xbc5c01,
[1634191084.105146][4649:4656] CHIP:DMG:                                EndpointId = 0x1,
[1634191084.105225][4649:4656] CHIP:DMG:                                ClusterId = 0x300,
[1634191084.105310][4649:4656] CHIP:DMG:                                FieldTag = 0x0,
[1634191084.105382][4649:4656] CHIP:DMG:                        }
[1634191084.105471][4649:4656] CHIP:DMG:
[1634191084.105557][4649:4656] CHIP:DMG:                        Data = 0,
[1634191084.105632][4649:4656] CHIP:DMG:                        DataElementVersion = 0x0,
[1634191084.105693][4649:4656] CHIP:DMG:                },
[1634191084.105771][4649:4656] CHIP:DMG:
[1634191084.105833][4649:4656] CHIP:DMG:        ],
[1634191084.105909][4649:4656] CHIP:DMG:
[1634191084.105968][4649:4656] CHIP:DMG: }
[1634191084.106759][4649:4656] CHIP:ZCL: ReadAttributesResponse:
[1634191084.106838][4649:4656] CHIP:ZCL:   ClusterId: 0x0000_0300
[1634191084.106901][4649:4656] CHIP:ZCL:   attributeId: 0x0000_0000
[1634191084.106956][4649:4656] CHIP:ZCL:   status: Success                (0x0000)
[1634191084.107010][4649:4656] CHIP:ZCL:   attribute TLV Type: 0x04
[1634191084.107077][4649:4656] CHIP:ZCL:   attributeValue: 0
[1634191084.107338][4649:4656] CHIP:EM: Piggybacking Ack for MessageCounter:00000038 on exchange: 11550i
AttributeReadResult(path=AttributePath(nodeId=12344321, endpointId=1, clusterId=768, attributeId=0), status=0, value=0)
[1634191084.107529][4649:4656] CHIP:IN: Secure message was encrypted: Msg ID 59
[1634191084.107603][4649:4656] CHIP:IN: Build encrypted message 0x7062258 to 0x0000000000BC5C01 of type 1 and protocolId 1 on exchange 11550r with MessageCounter 59.
[1634191084.107690][4649:4656] CHIP:IN: Sending encrypted msg 0x7062258 to 0x0000000000BC5C01 at utc time: 5977253 msec
[1634191084.107754][4649:4656] CHIP:IN: Sending msg on generic transport
chip-device-ctrl > [1634191084.108071][4649:4656] CHIP:DMG: Client[0] moving to [UNINIT]
[1634191084.200575][4649:4656] CHIP:EM: Received message of type 0x10 with vendorId 0x0000 and protocolId 0x0000 on exchange 11550i
[1634191084.200736][4649:4656] CHIP:EM: Rxd Ack; Removing MessageCounter:0000003B from Retrans Table
[1634191084.200801][4649:4656] CHIP:EM: Removed CHIP MessageCounter:0000003B from RetransTable

************step2a

chip-device-ctrl > zcl ColorControl MoveToHue 12344321 1 0 hue=1 direction=0 transitionTime=300 optionsMask=1 optionsOverride=1
[1634191838.361651][4649:4656] CHIP:DMG: ICR moving to [Initialize]
[1634191838.361761][4649:4656] CHIP:DMG: ICR moving to [AddCommand]
[1634191838.361991][4649:4656] CHIP:IN: Secure message was encrypted: Msg ID 60
[1634191838.362054][4649:4656] CHIP:IN: Build encrypted message 0x7062258 to 0x0000000000BC5C01 of type 8 and protocolId 1 on exchange 11551r with MessageCounter 60.
[1634191838.362123][4649:4656] CHIP:IN: Sending encrypted msg 0x7062258 to 0x0000000000BC5C01 at utc time: 6731507 msec
[1634191838.362223][4649:4656] CHIP:IN: Sending msg on generic transport
[1634191838.362497][4649:4656] CHIP:DMG: ICR moving to [   Sending]
[1634191838.726122][4649:4656] CHIP:EM: Received message of type 0x09 with vendorId 0x0000 and protocolId 0x0001 on exchange 11551i
[1634191838.726321][4649:4656] CHIP:EM: Rxd Ack; Removing MessageCounter:0000003C from Retrans Table
[1634191838.726388][4649:4656] CHIP:EM: Removed CHIP MessageCounter:0000003C from RetransTable
[1634191838.726500][4649:4656] CHIP:DMG: InvokeCommand =
[1634191838.726570][4649:4656] CHIP:DMG: {
[1634191838.726626][4649:4656] CHIP:DMG:        CommandList =
[... 1019 lines cut ...]
[1634192559.976575][4649:4656] CHIP:DMG:
[1634192559.976648][4649:4656] CHIP:DMG:                        Data = 155,
[... cut ...]
[1634192559.978537][4649:4656] CHIP:IN: Sending encrypted msg 0x7062258 to 0x0000000000BC5C01 at utc time: 7453124 msec
[1634192559.978593][4649:4656] CHIP:IN: Sending msg on generic transport
chip-device-ctrl > [1634192559.978916][4649:4656] CHIP:DMG: Client[0] moving to [UNINIT]
[1634192560.156871][4649:4656] CHIP:EM: Received message of type 0x10 with vendorId 0x0000 and protocolId 0x0000 on exchange 11575i
[1634192560.157003][4649:4656] CHIP:EM: Rxd Ack; Removing MessageCounter:0000006D from Retrans Table
[1634192560.157051][4649:4656] CHIP:EM: Removed CHIP MessageCounter:0000006D from RetransTable
'''

woody-apple, 2022-02-07:
@lucyShi0116 Can you please re-verify this on a more recent matter?

closed 2022-02-07 by woody-apple