project-chip/connectedhomeip #9680: Not support to write OnOff and GlobalSceneControl of OnOff Cluster

lucyShi0116, 2021-09-14:
####  Problem Description
 Not support to write OnOff and GlobalSceneControl of OnOff Cluster
 * expected behavior
 OnOff and GlobalSceneControl can be wrote successfully
 * actual behavior
 Return UnknownAttribute
 * steps to reproduce
 Send cmd: zclwrite OnOff OnOff 12344321 1 0 FALSE, zclwrite OnOff GlobalSceneControl 12344321 1 0 FALSE
 * system configuration
 commit 6db59aa (HEAD -> test_event_5, origin/test_event_5)
 * Logs:

```
chip-device-ctrl > zclwrite OnOff OnOff 12344321 1 0 FALSE
An exception occurred during writing ZCL attribute:
UnknownAttribute: cluster: OnOff, attribute: OnOff
chip-device-ctrl > [1631512536.577584][2526:2532] CHIP:DL: Got IP address on interface: wlan0 IP: 192.168.37.110
[1631512536.577865][2526:2535] CHIP:DL: HandlePlatformSpecificBLEEvent 32770

chip-device-ctrl > zclwrite OnOff GlobalSceneControl 12344321 1 0 FALSE
An exception occurred during writing ZCL attribute:
UnknownAttribute: cluster: OnOff, attribute: GlobalSceneControl
```

turon, 2022-02-01:
May be outdated, but need to verify before closing.
OnOff cluster must be spec compliant and work properly for v1.
Removing tag `v1_triage_split_4`

woody-apple, 2022-02-07:
@lucyShi0116 Can you please re-verify this and re-open if needed?

closed 2022-02-07 by woody-apple