project-chip/connectedhomeip #9681: Command OffWithEffect/OnWithTimedOff /OnWithRecallGlobalScene can't work

lucyShi0116, 2021-09-14:
#### Problem Description:
 Send OnOff command to DUT, DUT can't recognize the cmd of OffWithEffect/OnWithTimedOff /OnWithRecallGlobalScene
 * expected behavior
 Dut can recognize the cmd and return success
 * actual behavior
 Dut can't recognize the cmd and return failed
 * steps to reproduce
 1. Send cmd: zcl OnOff OffWithEffect 12344321 1 0 effectId=0 effectVariant=0
 2. Send cmd: zcl OnOff OnWithTimedOff 12344321 1 0 onOffControl=1 onTime=300 offWaitTime=300
 3. Send cmd: zcl OnOff OnWithRecallGlobalScene 12344321 1 0
 * system configuration
 commit 6db59aa (HEAD -> test_event_5, origin/test_event_5)
 * Logs:
```

chip-device-ctrl > zcl OnOff OffWithEffect 12344321 1 0 effectId=0 effectVariant=0
[1631598369.746230][947694:947705] CHIP:DMG: ICR moving to [Initialize]
[1631598369.746348][947694:947705] CHIP:DMG: ICR moving to [AddCommand]
[1631598369.746584][947694:947705] CHIP:IN: Secure message was encrypted: Msg ID 36
[1631598369.746658][947694:947705] CHIP:IN: Encrypted message 0x2258db98 to 0x0000000000BC5C01 of type 8 and protocolId 5 on exchange 39042.
[1631598369.746729][947694:947705] CHIP:IN: Sending msg 0x2258db98 to 0x0000000000BC5C01 at utc time: 98004568 msec
[1631598369.746784][947694:947705] CHIP:IN: Sending secure msg on generic transport
[1631598369.746978][947694:947705] CHIP:IN: Secure msg send status ../../src/inet/IPEndPointBasis.cpp:915: Success
[1631598369.747069][947694:947705] CHIP:DMG: ICR moving to [   Sending]
[1631598369.946838][947694:947705] CHIP:EM: Received message of type 0x09 with vendorId 0x0000 and protocolId 0x0005 on exchange 39042
[1631598369.946972][947694:947705] CHIP:EM: Rxd Ack; Removing MsgId:00000024 from Retrans Table
[1631598369.947029][947694:947705] CHIP:EM: Removed CHIP MsgId:00000024 from RetransTable
[1631598369.947136][947694:947705] CHIP:DMG: InvokeCommand =
[1631598369.947200][947694:947705] CHIP:DMG: {
[1631598369.947251][947694:947705] CHIP:DMG:    CommandList =
[1631598369.947313][947694:947705] CHIP:DMG:    [
[1631598369.947370][947694:947705] CHIP:DMG:            CommandDataElement =
[1631598369.947436][947694:947705] CHIP:DMG:            {
[1631598369.947507][947694:947705] CHIP:DMG:                    CommandPath =
[1631598369.947595][947694:947705] CHIP:DMG:                    {
[1631598369.947684][947694:947705] CHIP:DMG:                            EndpointId = 0x1,
[1631598369.947781][947694:947705] CHIP:DMG:                            ClusterId = 0x6,
[1631598369.947859][947694:947705] CHIP:DMG:                            CommandId = 0x40,
[1631598369.947931][947694:947705] CHIP:DMG:                    },
[1631598369.948017][947694:947705] CHIP:DMG:
[1631598369.948084][947694:947705] CHIP:DMG:                    StatusElement =
[1631598369.948163][947694:947705] CHIP:DMG:                    {
[1631598369.948237][947694:947705] CHIP:DMG:                            GeneralCode = 0xd,
[1631598369.948323][947694:947705] CHIP:DMG:                            ProtocolId = 0x0,
[1631598369.948400][947694:947705] CHIP:DMG:                            protocolCode = 0x81,
[1631598369.948473][947694:947705] CHIP:DMG:                    },
[1631598369.948556][947694:947705] CHIP:DMG:
[1631598369.948620][947694:947705] CHIP:DMG:            },
[1631598369.948697][947694:947705] CHIP:DMG:
[1631598369.948752][947694:947705] CHIP:DMG:    ],
[1631598369.948822][947694:947705] CHIP:DMG:
[1631598369.948872][947694:947705] CHIP:DMG: }
SetCommandIndexStatus commandHandle=1 commandIndex=1
[1631598369.949479][947694:947705] CHIP:ZCL: DefaultResponse:
[1631598369.949536][947694:947705] CHIP:ZCL:   Transaction: 0xffff9d7814a0
[1631598369.949583][947694:947705] CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
[1631598369.949634][947694:947705] CHIP:ZCL: IMDefaultResponseCallback: Missing success callback
[1631598369.949680][947694:947705] CHIP:ZCL: IMDefaultResponseCallback: Missing failure callback
[1631598369.949848][947694:947705] CHIP:DMG: ICR moving to [Uninitiali]
[1631598369.949921][947694:947705] CHIP:EM: Sending Standalone Ack for MsgId:00000013
Received command status response:
[1631598369.950063][947694:947705] CHIP:IN: Secure message was encrypted: Msg ID 37
[1631598369.950182][947694:947705] CHIP:IN: Encrypted message 0xffff95ff9af8 to 0x0000000000BC5C01 of type 16 and protocolId 0 on exchange 39042.
[1631598369.950256][947694:947705] CHIP:IN: Sending msg 0xffff95ff9af8 to 0x0000000000BC5C01 at utc time: 98004772 msec
Container:
    ProtocolId = 0
    ProtocolCode = 129
    EndpointId = 1
    ClusterId = 6
    CommandId = 64
[... 360 lines cut ...]
[1631599265.673309][947694:947705] CHIP:DMG:
[1631599265.673360][947694:947705] CHIP:DMG: }
SetCommandIndexStatus commandHandle=1 commandIndex=1
[1631599265.674018][947694:947705] CHIP:ZCL: DefaultResponse:
[1631599265.674074][947694:947705] CHIP:ZCL:   Transaction: 0xffff9d7814a0
[1631599265.674187][947694:947705] CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
[1631599265.674244][947694:947705] CHIP:ZCL: IMDefaultResponseCallback: Missing success callback
[1631599265.674290][947694:947705] CHIP:ZCL: IMDefaultResponseCallback: Missing failure callback
[1631599265.674458][947694:947705] CHIP:DMG: ICR moving to [Uninitiali]
[1631599265.674531][947694:947705] CHIP:EM: Sending Standalone Ack for MsgId:0000001C
Received command status response:
[1631599265.674675][947694:947705] CHIP:IN: Secure message was encrypted: Msg ID 55
[1631599265.674743][947694:947705] CHIP:IN: Encrypted message 0xffff95ff9af8 to 0x0000000000BC5C01 of type 16 and protocolId 0 on exchange 39051.
[1631599265.674813][947694:947705] CHIP:IN: Sending msg 0xffff95ff9af8 to 0x0000000000BC5C01 at utc time: 98900496 msec
Container:
    ProtocolId = 0
    ProtocolCode = 129
    EndpointId = 1
    ClusterId = 6
    CommandId = 66
    CommandIndex = 1[1631599265.674867][947694:947705] CHIP:IN: Sending secure msg on generic transport

[1631599265.675233][947694:947705] CHIP:IN: Secure msg send status ../../src/inet/IPEndPointBasis.cpp:915: Success
[1631599265.675315][947694:947705] CHIP:EM: Flushed pending ack for MsgId:0000001C
```

tcarmelveilleux, 2022-01-31:
@lucyShi0116 Is this still happening? If so, what platform is it on?

woody-apple, 2022-02-07:
@lucyShi0116 Can you please re-open if this is still an issue.

closed 2022-02-07 by woody-apple