project-chip/connectedhomeip #9691: About onboarding process on ESP32

dhrishi, 2021-09-14:
#### Problem
Onboarding issues for ESP32 as mentioned by @jonsmirl in : https://github.com/project-chip/connectedhomeip/issues/9264#issuecomment-907730513 and https://github.com/project-chip/connectedhomeip/issues/9264#issuecomment-907808053

@sweetymhaiske Can you please take a look

jonsmirl, 2021-09-14:
There are two different issues

1) A generic one, the Python app does not store the generated keys permanently.  So every time you restart the Python app the newly generated keys don't match the ones stored in the device nodes.  This needs to be fixed in the Python app. The C-based chiptool stores the keys in a file and they are kept across sessions.

2) ESP specific one. When I was debugging this I experienced many failures in the onboarding process. In general the ESP code can not tolerate the failure of a step in the onboarding process -the only way to recover is to reboot.  If the onboard process fails, ESP should delete any partial data and revert back to BLE advertising mode.  Consider just the simple case of a typo in a command and wanting to redo the command with the corrected string. You should examine each step of the on-boarding process and figure out what it makes sense for the ESP to do to recover if the process does not complete. Don't leave the chip in a state where the only recovery is to reflash wiping the stored data.

sweetymhaiske, 2021-09-14:
I will check and update.

dhrishi, 2021-09-14:
> A generic one, the Python app does not store the generated keys permanently. So every time you restart the Python app the newly generated keys don't match the ones stored in the device nodes. This [...]

@erjiaqing Do you happen to know if this is being fixed already?

sweetymhaiske, 2021-09-15:
Hi @jonsmirl,
1. Related to the typo issue, it is just a minor change in the python controller. PR- #9724 
2. For the onboard process failure before the PASE session, this thing can be handle. I am working on it.
3. After the PASE session, if there is a failure in commissioning, even after certain attempts we failed to commission, then this concept of failSafe timer is already present in Matter, which is not completely implemented yet. Once it comes into Matter then we can reset the device state after the failSafe timer expires so that the next commissioning attempt proceeds as expected

hidaris, 2021-09-16:
> There are two different issues
>
> 1. A generic one, the Python app does not store the generated keys permanently.  So every time you restart the Python app the newly generated keys don't match the [...]

I also encountered this issue.

bzbarsky-apple, 2022-06-13:
The python app bits are fixed, I believe.

The esp-side bits should also be fixed, via the fail-safe handling.

Please reopen if this is still an issue, and provide steps to reproduce.

closed 2022-06-13 by bzbarsky-apple