project-chip/connectedhomeip #9721: Can BLE work as a node? is it support cluster?

KenneyQu, 2021-09-15:
#### Problem
According to the architecture pic below, BLE can support ipv6. But according to the demo, example, BLE is just work for config thread network.
I went through the matter spec 0.7 roughly, but did not find any answer. 

[image]

 #### Proposed Solution
Is there any introduction about how we support BLE device work as a node/endpoint.

jonsmirl, 2021-09-16:
CHIP runs over TCPIP, so anything that supports TCPIP can run CHIP. CHIP over BLE IPv6 can work, but I am not aware of any existing code implementing that.

Having said that, consider that you are going to need a BLE IPv6 border router in the system for this work. Another solution is to use normal BLE GATT and then build CHIP proxies for the BLE devices.  The proxy scheme is a good way to deal with battery powered devices not always being available.

KenneyQu, 2021-09-17:
Thanks @jonsmirl for your answer.
So do you know if Matter Alliance have some discussion on this, either BLE IPv6 or Matter Proxies should work?

For my understanding, there are lots of legacy BLE only devices on the market. So it is import that Matter to support communication with BLE only devices. So that I can control the device through the network, or to say from Cloud. 

Comments?

dhrishi, 2021-09-17:
>CHIP over BLE IPv6 can work, but I am not aware of any existing code implementing that.

BLE over IPv6 has been moved out. AFAIK, Matter specification 1.0 will run on Wi-Fi and Thread and will use BLE only for commissioning. See here: https://buildwithmatter.com/

jonsmirl, 2021-09-17:
Look at examples/bridge-app for hints on how to make proxies.

closed 2021-11-08 by KenneyQu