project-chip/connectedhomeip #9768: Global Attribute Cluster Revision needs to be fixed for Thermostat User Interface Configuration Cluster

raju-apple, 2021-09-16:
Global Attribute Cluster Revision needs to be fixed for Thermostat User Interface Configuration Cluster
Spec snippet : 

[image]

Value needs to be 2

bzbarsky-apple, 2022-05-04:
General question: how do we want to handle this?  Right now cluster revision defaults to 1 in the XML but then you can set the right thing in ZAP.  Should we be adding overrides of the default value to our cluster XML definitions?  Seems like it would be a good idea....

@vivien-apple @tecimovic

mrjerryjohns, 2022-05-04:
The cluster revision is a quality of the schema definition, not the product-configuration that is encapsulated in the ZAP file.

It should exist in the schema definition. I've recommended to Terrance (who's working on en-mass, automatically inferring a default value for Cluster Revision if one doesn't exist in the attribute store) to automatically stuff a value of 1 if one doesn't exist.

Which means we can have the rule that you either specify the cluster revision in the XML (which can be set to a specific value) or you don't, in which case, it would default to 1.

bzbarsky-apple, 2022-05-04:
That seems pretty sensible, yes.

tecimovic, 2022-05-04:
XML already has a per-cluster default value for a global attribute. Is this the case here?

bzbarsky-apple, 2022-05-04:
Yes, I know we can set defaults via `<globalAttribute>` in the cluster XML.  The question was whether that's what we want here.  Sounds like it is, and we should do this for all our clusters...

@tecimovic I assume after that we will still need to manually update `.zap` files that now have the wrong values?

tecimovic, 2022-05-04:
The values inside .zap files will take precedence, as they are treated as "user typed this in!" which overrides the defaults, wherever they come from.

You would have to either:
  1.) Manually update them.
  2.) Use the zap scripting to update them automatically via a script.

When I refer to zap scripting I refer to the "post-load" functionality.
When you run zap, you can specify a ".js" file via a --postImportScript command line arg. They point to a file that looks like this:
```
// Example file for the post load functionality.

async function postLoad(api, context) {
  let endpoints = await api.endpoints(context)
  let ep = endpoints[0]
  // Here we disable cluster with code 2
  await api.disableClientCluster(context, ep, 2)

  // Here we turn off attribute 4 on cluster 0, the manufacturer name
  await api.disableServerAttribute(context, ep, 0, 4)

  await api.enableOutgoingCommand(context, ep, 0x00, 0x00, api.client)
  await api.disableOutgoingCommand(context, ep, 0x00, 0x00, api.client)
}

exports.postLoad = postLoad
```
(Example takes from some of the unit tests in the test/resource/ dir...)

Essentially, we have this little API that you can use from the .js file, that allows you to muck around with the data right after the ZAP file is loaded into the tables.
So you can then use the:
   `zap convert --out FILE.ZAP --in FILE.ZAP --postImportScript SCRIPT.JS`
command, and that will convert the ZAP file with whatever the script tells it to do, plus any automatic upgrades that zap does to the ZAP file along the way anyway.

mrjerryjohns, 2022-05-04:
A 'default' value for a ClusterRevision doesn't make any sense in an XML definition - unlike other attributes where the value can change, this attribute should have a fixed value that is a function of what is defined in the XML. It should not be over-ridable in a ZAP file, or tweakable later.

mrjerryjohns, 2022-05-04:
I'm almost inclined to say that 'ClusterRevision' not be specified as an attribute in the XML definition but rather, as a separate bespoke tag/value pair to wean it off the association to the attribute, which is just a way to surface it in the Matter IM, and in this representation, couples in baggage and implicit assumptions about the behavior of attributes (which don't apply to this quality/metadata).

woody-apple, 2022-05-25:
SDK Spec Review: At this point we believe that the version is correct to be spec compliant.

woody-apple, 2022-05-25:
SDK Spec Review: Given no cert tests are failing here (cert blocker), we believe this is likely resolved.

closed 2022-05-25 by woody-apple