project-chip/connectedhomeip #9796: Reliable messaging crashes if a standalone ack is lost

bzbarsky-apple, 2021-09-17:
#### Problem
Consider the following sequence of events:
1. Initiator sends a reliable message to responder.
2. Responder takes a while to process the message, hits standalone ack timeout, sends standalone ack.
3. Standalone ack is lost.
4. Responder sends an application-level response.  No ack is piggybacked, because the standalone ack was already sent.
5. Initiator processes application-level response, sends another application-level reliable message on the same exchange.

This crashes the initiator in `ReliableMessageMgr::AddToRetransTable` because of this line:

> `VerifyOrDie(rc != nullptr && !rc->IsOccupied());`

In this case, `rc->IsOccupied()` is true: it's waiting for an ack to that first message it sent.

 #### Proposed Solution
We have a few options:

* Allow a given `ReliableMessageContext` to be waiting for more than one ack.
* Don't clear ack information completely when sending a standalone ack; piggyback that same ack on the next application-level message as well.  This will require fixing the behavior we have now where processing of a message that acks something we think does not need acking is completely aborted.
* Other ideas?

@turon @kghost @yufengwangca @pan-apple @msandstedt 

<details>
<summary>Unit test exercising this scenario, which crashes on ToT:</summary>

```
diff --git a/src/messaging/tests/TestReliableMessageProtocol.cpp b/src/messaging/tests/TestReliableMessageProtocol.cpp
index 34d6ef108..ba23cd414 100644
--- a/src/messaging/tests/TestReliableMessageProtocol.cpp
+++ b/src/messaging/tests/TestReliableMessageProtocol.cpp
@@ -1111,6 +1111,135 @@ void CheckMessageAfterClosed(nlTestSuite * inSuite, void * inContext)
     NL_TEST_ASSERT(inSuite, rm->TestGetCountRetransTable() == 0);
 }
 
+void CheckLostStandaloneAck(nlTestSuite * inSuite, void * inContext)
+{
+    /**
+     * This tests the following scenario:
+     * 1) A reliable message is sent from initiator to responder.
+     * 2) The responder sends a standalone ack, which is lost.
+     * 3) The responder sends an application-level response.
+     * 4) The initiator sends a reliable response to the app-level response.
+     */
+    TestContext & ctx = *reinterpret_cast<TestContext *>(inContext);
+
+    chip::System::PacketBufferHandle buffer = chip::MessagePacketBuffer::NewWithData(PAYLOAD, sizeof(PAYLOAD));
+    NL_TEST_ASSERT(inSuite, !buffer.IsNull());
+
+    CHIP_ERROR err = CHIP_NO_ERROR;
+
+    MockAppDelegate mockReceiver;
+    err = ctx.GetExchangeManager().RegisterUnsolicitedMessageHandlerForType(Echo::MsgType::EchoRequest, &mockReceiver);
+    NL_TEST_ASSERT(inSuite, err == CHIP_NO_ERROR);
+
+    mockReceiver.mTestSuite = inSuite;
+
+    MockAppDelegate mockSender;
+    ExchangeContext * exchange = ctx.NewExchangeToAlice(&mockSender);
+    NL_TEST_ASSERT(inSuite, exchange != nullptr);
+
+    mockSender.mTestSuite = inSuite;
+
+    ReliableMessageMgr * rm = ctx.GetExchangeManager().GetReliableMessageMgr();
+    NL_TEST_ASSERT(inSuite, rm != nullptr);
+
+    // Ensure the retransmit table is empty right now
+    NL_TEST_ASSERT(inSuite, rm->TestGetCountRetransTable() == 0);
+
+    // We send a message, the other side sends a standalone ack first (which is
+    // lost), then an application response, then we respond to that response.
+    // We need to keep both exchanges alive for that (so we can send the
+    // response from the receiver and so the initial sender exchange can send a
+    // response to that).
+    gLoopback.mSentMessageCount    = 0;
+    gLoopback.mNumMessagesToDrop   = 0;
+    gLoopback.mDroppedMessageCount = 0;
+    mockReceiver.mRetainExchange   = true;
+    mockSender.mRetainExchange     = true;
+
+    // And ensure the ack heading back our way is dropped.
+    mockReceiver.mDropAckResponse = true;
+
+    err = exchange->SendMessage(Echo::MsgType::EchoRequest, std::move(buffer), SendFlags(SendMessageFlags::kExpectResponse));
+    NL_TEST_ASSERT(inSuite, err == CHIP_NO_ERROR);
+
+    // Ensure the message was sent.
+    NL_TEST_ASSERT(inSuite, gLoopback.mSentMessageCount == 1);
+    NL_TEST_ASSERT(inSuite, gLoopback.mDroppedMessageCount == 0);
+
+    // And that it was received.
+    NL_TEST_ASSERT(inSuite, mockReceiver.IsOnMessageReceivedCalled);
+
+    // And that we have not gotten any app-level responses or acks so far.
+    NL_TEST_ASSERT(inSuite, !mockSender.IsOnMessageReceivedCalled);
+    NL_TEST_ASSERT(inSuite, rm->TestGetCountRetransTable() == 1);
+
+    ReliableMessageContext * receiverRc = mockReceiver.mExchange->GetReliableMessageContext();
+    // Ack should have been dropped.
+    NL_TEST_ASSERT(inSuite, !receiverRc->IsAckPending());
+
+    // Don't drop any more acks.
+    mockReceiver.mDropAckResponse = false;
+
+    // Now send a message from the other side.
+    buffer = chip::MessagePacketBuffer::NewWithData(PAYLOAD, sizeof(PAYLOAD));
+    NL_TEST_ASSERT(inSuite, !buffer.IsNull());
+
+    mockReceiver.mExchange->SendMessage(Echo::MsgType::EchoResponse, std::move(buffer),
+                                        SendFlags(SendMessageFlags::kExpectResponse));
+
+    // Ensure the response was sent.
+    NL_TEST_ASSERT(inSuite, gLoopback.mSentMessageCount == 2);
+    NL_TEST_ASSERT(inSuite, gLoopback.mDroppedMessageCount == 0);
+
+    // Ensure that we have received that response and it did not have a
+    // piggyback ack.
+    NL_TEST_ASSERT(inSuite, mockSender.IsOnMessageReceivedCalled);
+    NL_TEST_ASSERT(inSuite, !mockSender.mReceivedPiggybackAck);
+    // We now have our un-acked message still waiting to retransmit and the
+    // message we just received is waiting for an ack.
+    NL_TEST_ASSERT(inSuite, rm->TestGetCountRetransTable() == 2);
+
+    // Reset various state so we can measure things again.
+    mockReceiver.IsOnMessageReceivedCalled = false;
+    mockReceiver.mReceivedPiggybackAck     = false;
+    mockSender.IsOnMessageReceivedCalled   = false;
+    mockSender.mReceivedPiggybackAck       = false;
+
+    // Stop retaining the recipient exchange.
+    mockReceiver.mRetainExchange = false;
+
+    // Now send a new message to the other side.
+    buffer = chip::MessagePacketBuffer::NewWithData(PAYLOAD, sizeof(PAYLOAD));
+    NL_TEST_ASSERT(inSuite, !buffer.IsNull());
+
+    // Claim (falsely) that we expect a response, so exchange stays open and we
+    // can use it to clear out reliable message state later.
+    err = exchange->SendMessage(Echo::MsgType::EchoRequest, std::move(buffer),
+                                SendFlags(SendMessageFlags::kExpectResponse));
+    NL_TEST_ASSERT(inSuite, err == CHIP_NO_ERROR);
+
+    // Ensure the message and the standalone ack to it were sent.
+    NL_TEST_ASSERT(inSuite, gLoopback.mSentMessageCount == 4);
+    NL_TEST_ASSERT(inSuite, gLoopback.mDroppedMessageCount == 0);
+
+    // And that it was received.
+    NL_TEST_ASSERT(inSuite, mockReceiver.IsOnMessageReceivedCalled);
+    NL_TEST_ASSERT(inSuite, mockReceiver.mReceivedPiggybackAck);
+
+    // And that receiver has no ack pending.
+    NL_TEST_ASSERT(inSuite, !receiverRc->IsAckPending());
+
+    // And that we are still expecting an ack for the one message an ack was
+    // lost for.
+    NL_TEST_ASSERT(inSuite, rm->TestGetCountRetransTable() == 1);
+
+    // Clear out our one pending retransmission entry.
+    rm->ClearRetransTable(exchange);
+
+    // And we should have no pending work left now.
+    NL_TEST_ASSERT(inSuite, rm->TestGetCountRetransTable() == 0);
+}
+
 // Test Suite
 
 /**
@@ -1134,6 +1263,7 @@ const nlTest sTests[] =
     NL_TEST_DEF("Test ReliableMessageMgr::CheckSendStandaloneAckMessage", CheckSendStandaloneAckMessage),
     NL_TEST_DEF("Test command, response, default response, with receiver closing exchange after sending response", CheckMessageAfterClosed),
     NL_TEST_DEF("Test that unencrypted message is dropped if exchange requires encryption", CheckUnencryptedMessageReceiveFailure),
+    NL_TEST_DEF("Test that an application-level response-to-response after a lost standalone ack to the initial message works", CheckLostStandaloneAck),
 
     NL_TEST_SENTINEL()
 };
```

</details>

turon, 2021-09-17:
The way input technologies handled this is the 2nd option:

Even after sending a standalone Ack, which can get lost, and the actual response message must also piggyback the Ack.

So a responder on an exchange always marks the next tx message on an exchange A=1 when for an rx message with R=1, even if it will be a duplicate Ack to a standalone Ack already sent.

msandstedt, 2021-09-17:
The problem is that we have a conflation between the transport-layer / MRP ack and exchange tracking.  The exchange layer uses exchange IDs.  MRP uses message IDs.  So given that ACKs also carry message IDs, why are we trying to use them to update state in the exchange layer?  Once we dispatch from exchange layer to MRP layer, the exchange layer has done it's job for the outgoing message and shouldn't need to worry about it anymore.

kghost, 2021-09-17:
This is not quite easy to solve. If we just do "don't clear ack information completely when sending a standalone ack", a malicious peer can still trigger the crash by sending multiple messages without ack but trigger response. (This is a potential backdoor which enables end user run unauthenticated code on the device).

I can recall that Boris have a PR which defines whether the exchange is waiting for a message or is about to send message. Along with this information, we should drop ALL incoming messages when the exchange is about to send message, but this may require spec change.

msandstedt, 2021-09-17:
Interactions are stateful transactions with rules about what type of incoming and outgoing messages are required and expected, and in what sequence they are required and expected.

It obviously makes sense for IM to instruct the exchange layer that responses are expected, but no more information than this should be required.  When an exchange emits a message, there are only two possible things IM can instruct: 1) no response is expected -> close exchange; OR 2) responses are expected -> keep exchange open.

Beyond that, it should be possible for the exchange layer to dispatch all received traffic to IM and for IM to sort it out.  It's not apparent to me how acks enter into this at all.

turon, 2021-09-17:
@msandstedt MRP uses MessageCounter for tracking messages, but runs in the context of an exchange.

So for any given exchange, an MRP receiver tracks the last R=1 message and AckCounter from its exchange peer.  Once in acknowledgement state, any messages to the exchange peer will have A=1 and that AckCounter.  These duplicate Acks remain sticky and only cleared when a new message is from the exchange peer is received.  If that new message is R=1, then A=1 actually remains active but now with an updated AckCounter.

turon, 2021-09-17:
@kghost I'm not sure I understand your scenario, but we certainly must not let any sequence of messages to allow running unsigned code on a device.  I would assume that would only be possible via some buffer overflow which would be an implementation issue, not a protocol one.

Generally threat scenarios where "the attacker" is a commissioned Matter peer w/ ACL access to the target device are not in scope of the threat model as such an attacker was granted trusted access to the target device -- DoS attacks from a trusted yet malicious peer are impossible to prevent.

kghost, 2021-09-17:
How does the spec handles the scenario I have proposed:

1. Mike send a message w/o an ack and requires a response.
2. Mike send another message with different message counter w/o an ack and requires a response.

In out implementation, on the receiver side:
1. Alice received a packet with Counter=1 NoAck
  a. Alice handles this packet to IM layer
  b. Alice generates a response with AliceCounter=1 AliceAck=1
2. Alice received a packet with Counter=2 NoAck
  a. Alice handles this packet to IM layer
  b. Alice generates a response with AliceCounter=2 AliceAck=2

Mike haven't acked any packet from Alice, so Alice is forced to store multiple response packets in her send queue, even if we allow Alice queue multiple packets, Mike can easily exhaust the queue space by sending messages.

The root cause of this problem is there is no flow control between application layer and transport layer. The transport layer will accept all incoming messages and assume application layer can handle it and application layer successfully handled the packet or not.

Take TCP as example, when the receiver side is not able to handle incoming data, the application just don't call read on the socket, so that the message will pile up in the queue, and eventually block the sender. But in our implementation, on the contract, the exchange layer can not express that "my resources is exhausted, please don't accept more packets", We need a API on exchange context to express that, and we need a flow control protocol to notify the sender, do not send more messages.

msandstedt, 2021-09-17:
> @msandstedt MRP uses MessageCounter for tracking messages, but runs in the context of an exchange.
> 
> So for any given exchange, an MRP receiver tracks the last R=1 message and AckCounter from its exchange peer. Once in acknowledgement state, any messages to the exchange peer will have A=1 and that AckCounter. These duplicate Acks remain sticky and only cleared when a new message from the exchange peer is received. If that new message is R=1, then A=1 actually remains active but now with an updated AckCounter.

Is this maybe the source of the problem then?  TCP, for instance, is doing its own acks and doesn't bubble up any status except when the connection is lost.  Why have we pushed MRP into the exchange context?  Why is this treated differently?

pidarped, 2021-09-17:
@kghost, messages 1 and 2 should be on separate exchange contexts. Only one outstanding protocol message should be in flight from a sending node at a time for a given exchange. If A sends a message that has R=1 to B, and needs a response, then A needs to wait for the response to come or an exchange timeout before closing the exchange. For A to send a message with R=0, then A does not care if the message reached B but should still wait until exchange timeout for the response, if a response is expected.

kghost, 2021-09-17:
So to completely solve the problem, We need a clear define of the state of an exchange:

Active: The exchange is about to send a command or a response.
Ack: The exchange is waiting for the ack of the last message.
Wait: The exchange is waiting for a data from other peer.

1. In Active state, RMP will discard any further messages, 
  1a. Go to Ack state when a message that requires a ack is send
  1b. Go to Wait state when a message that doesn't require a ack is send
2. In Ack state, RMP will discard any further messages, unless the message is bundled with an expect ack
  2a. When the expect ack is received w/o data, go to Wait state.
  2b. When the expect ack is received with data, propagated the message to application layer, go to Active state.
3. In Wait state, RMP will accept one message.
  3a. Go to Active state once a message is received and propagated the message to application layer.

turon, 2021-09-17:
@kghost I agree defining the state machine and associated flow chart clearly would help us and yours is a good initial stab.

1. Only one pending outbound message per exchange.
2. Ack handling as discussed here.
3. This wait state handling (for R=0 outbound messages) is better handled by IM or the layer above Exchange.

@msandstedt The scoping of MRP state to a given Exchange provides a form of channelized flow control over a session such that multiple independent endpoint interactions can occur in parallel yet each has simple, linear rules to follow (per the Exchange state machine) keeping embedded resource overhead low.

msandstedt, 2021-09-17:
I must just be thinking about this differently.

How do acks give us exchange-scoped back pressure?  If you send out-of-sequence or unexpected messages in an exchange, won't this just immediately be identified as illegal in the interaction model, resulting in an aborted interaction and exchange?  I thought all interactions were strictly back-and-forth.  Are we saying that there are legal interactions that can include several messages in series from sender to receiver with only acks flowing in the other direction?  If so, I guess I can see how the acks are needed.

As to the general resource starvation issue, I also don't follow.  It is completely legal to attempt to start a very large number of concurrent exchanges with a peer over a given session context.  If responders want to impose fairness, they need to limit the number of allowed exchanges per session context.

pidarped, 2021-09-17:
@msandstedt, the simpler semantics of Acking only one active sent message in MRP is what made scoping the reliability semantics within an Exchange necessary. Since MRP does not support a sliding window of sent messages that can be Acked at once, like TCP, having it at the Messaging layer would complicate Ack handling and would not allow multiple exchanges to be active at the same time. However, currently, peers can have multiple exchanges active simultaneously(to a configured limit depending on resources) although each such exchange caters to only one outstanding message at a time.

bzbarsky-apple, 2021-09-18:
> We need a clear define of the state of an exchange:

Yes.

> In Ack state, RMP will discard any further messages, unless the message is bundled with an expect ack

This only works if we make the change from my second bullet, which is how @turon says it works in the input technology.

Otherwise if the ack is lost the following (ack-less, now) data response (call it data message 2) will be dropped and then that exchange is hosed: when we retransmit the original data message 1 (because it never got acked) the receiver will ack but not send another message 2, so we will never reach message 3 and time out on the IM level or something.

If we make the change where even if we send a standalone ack we still bundle acks with any following data messages, then this would work.

Apart from that, the state machine looks pretty reasonable.

I'm not really following the resource-starvation scenario.  Within a single exchange it can't happen because the exchange will get closed on the first unexpected/bogus data message, and across multiple exchanges nothing we do to address this bug would affect it.

bzbarsky-apple, 2021-09-18:
I'll do the second bullet thing.

bzbarsky-apple, 2021-09-22:
OK.  So the second bullet is not enough on its own, for a very simple reason: it relies on the _responder_ having a certain behavior to avoid a crash on the _initiator_.  A malicious responder could still trigger the crash.

I assume this is what @kghost was alluding to in https://github.com/project-chip/connectedhomeip/issues/9796#issuecomment-922025044 ?  This crash is not exploitable, since it's a `VerifyOrDie` failure but is a DoS, but otherwise that seems to be the gist of that comment.

It seems like the right solution should be a combination of:

1. Doing bullet 2 above.
2. Replacing the `VerifyOrDie` with a non-fatal error return, which presumably (indirectly) leads to the exchange being closed.

closed 2021-09-28 by bzbarsky-apple