project-chip/connectedhomeip #9813: After send ColorLoopSet command, EnhancedCurrentHue can't start loop

cindycaowiz, 2021-09-18:
Problem Description:

expected behavior

actual behavior

steps to reproduce

logs:
```
====11a
chip-device-ctrl > zcl ColorControl ColorLoopSet 12344321  1 0 updateFlags=2 action=0 direction=0 time=0 startHue=0 optionsMask=0 optionsOverride=0
[1631606853.142800][295582:295590] CHIP:DMG: ICR moving to [Initialize]
[1631606853.142925][295582:295590] CHIP:DMG: ICR moving to [AddCommand]
[1631606853.143157][295582:295590] CHIP:IN: Secure message was encrypted: Msg ID 602
[1631606853.143223][295582:295590] CHIP:IN: Encrypted message 0x38d33ab8 to 0x0000000000BC5C01 of type 8 and protocolId 5 on exchange 39325.
[1631606853.143297][295582:295590] CHIP:IN: Sending msg 0x38d33ab8 to 0x0000000000BC5C01 at utc time: 96656606 msec
[1631606853.143353][295582:295590] CHIP:IN: Sending secure msg on generic transport
[1631606853.143606][295582:295590] CHIP:IN: Secure msg send status ../../src/inet/IPEndPointBasis.cpp:915: Success
[1631606853.143706][295582:295590] CHIP:DMG: ICR moving to [   Sending]
[1631606853.408842][295582:295590] CHIP:EM: Received message of type 0x09 with vendorId 0x0000 and protocolId 0x0005 on exchange 39325
[1631606853.408990][295582:295590] CHIP:EM: Rxd Ack; Removing MsgId:0000025A from Retrans Table
[1631606853.409049][295582:295590] CHIP:EM: Removed CHIP MsgId:0000025A from RetransTable
[1631606853.409162][295582:295590] CHIP:DMG: InvokeCommand =
[1631606853.409226][295582:295590] CHIP:DMG: {
[1631606853.409270][295582:295590] CHIP:DMG:    CommandList =
[1631606853.409327][295582:295590] CHIP:DMG:    [
[1631606853.409383][295582:295590] CHIP:DMG:            CommandDataElement =
[1631606853.409445][295582:295590] CHIP:DMG:            {
[1631606853.409504][295582:295590] CHIP:DMG:                    CommandPath =
[1631606853.409618][295582:295590] CHIP:DMG:                    {
[1631606853.409732][295582:295590] CHIP:DMG:                            EndpointId = 0x1,
[1631606853.409797][295582:295590] CHIP:DMG:                            ClusterId = 0x300,
[1631606853.409878][295582:295590] CHIP:DMG:                            CommandId = 0x44,
[1631606853.409951][295582:295590] CHIP:DMG:                    },
[1631606853.410031][295582:295590] CHIP:DMG:
[1631606853.410093][295582:295590] CHIP:DMG:                    StatusElement =
[1631606853.410157][295582:295590] CHIP:DMG:                    {
[1631606853.410226][295582:295590] CHIP:DMG:                            GeneralCode = 0x0,
[1631606853.410301][295582:295590] CHIP:DMG:                            ProtocolId = 0x5,
[1631606853.410375][295582:295590] CHIP:DMG:                            protocolCode = 0x0,
[1631606853.410444][295582:295590] CHIP:DMG:                    },
[1631606853.410521][295582:295590] CHIP:DMG:
[1631606853.410586][295582:295590] CHIP:DMG:            },
[1631606853.410667][295582:295590] CHIP:DMG:
[1631606853.410724][295582:295590] CHIP:DMG:    ],
[1631606853.410798][295582:295590] CHIP:DMG:
[1631606853.410850][295582:295590] CHIP:DMG: }
SetCommandIndexStatus commandHandle=1 commandIndex=1
[1631606853.411478][295582:295590] CHIP:ZCL: DefaultResponse:
[1631606853.411535][295582:295590] CHIP:ZCL:   Transaction: 0xffffbcb0a4a0
[1631606853.411585][295582:295590] CHIP:ZCL:   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
[1631606853.411639][295582:295590] CHIP:ZCL: IMDefaultResponseCallback: Missing success callback
[1631606853.411688][295582:295590] CHIP:ZCL: IMDefaultResponseCallback: Missing failure callback
[1631606853.411856][295582:295590] CHIP:DMG: ICR moving to [Uninitiali]
[1631606853.411935][295582:295590] CHIP:EM: Sending Standalone Ack for MsgId:0000012E
Received command status response:
[1631606853.412086][295582:295590] CHIP:IN: Secure message was encrypted: Msg ID 603
[1631606853.412157][295582:295590] CHIP:IN: Encrypted message 0xffffb93f3af8 to 0x0000000000BC5C01 of type 16 and protocolId 0 on exchange 39325.
[1631606853.412228][295582:295590] CHIP:IN: Sending msg 0xffffb93f3af8 to 0x0000000000BC5C01 at utc time: 96656875 msec
Container:
    ProtocolId = 5
    ProtocolCode = 0
    EndpointId = 1
    ClusterId = 768
    CommandId = 68
    CommandIndex = 1[1631606853.412284][295582:295590] CHIP:IN: Sending secure msg on generic transport

[1631606853.412699][295582:295590] CHIP:IN: Secure msg send status ../../src/inet/IPEndPointBasis.cpp:915: Success
[1631606853.412782][295582:295590] CHIP:EM: Flushed pending ack for MsgId:0000012E
[... 988 lines cut ...]
[1631607838.575375][295582:295590] CHIP:DMG:
[1631607838.575454][295582:295590] CHIP:DMG:                    Data = 16384,
[1631607838.575527][295582:295590] CHIP:DMG:                    DataElementVersion = 0x0,
[1631607838.575602][295582:295590] CHIP:DMG:            },
[1631607838.575676][295582:295590] CHIP:DMG:
[1631607838.575731][295582:295590] CHIP:DMG:    ],
[1631607838.575798][295582:295590] CHIP:DMG:
[1631607838.575848][295582:295590] CHIP:DMG: }
[1631607838.576493][295582:295590] CHIP:ZCL: ReadAttributesResponse:
[1631607838.576606][295582:295590] CHIP:ZCL:   ClusterId: 0x0000_0300
[1631607838.576662][295582:295590] CHIP:ZCL:   attributeId: 0x0000_4000
[1631607838.576712][295582:295590] CHIP:ZCL:   status: Success                (0x0000)
[1631607838.576758][295582:295590] CHIP:ZCL:   attribute TLV Type: 0x04
[1631607838.576813][295582:295590] CHIP:ZCL:   attributeValue: 16384
[1631607838.576989][295582:295590] CHIP:DMG: Client[0] moving to [INIT]
[1631607838.577049][295582:295590] CHIP:DMG: Client[0] moving to [UNINIT]
AttributeReadResult(path=AttributePath(nodeId=12344321, endpointId=1, clusterId=768, attributeId=16384), status=0, value=16384)
[1631607838.577116][295582:295590] CHIP:EM: Sending Standalone Ack for MsgId:00000151
[1631607838.577319][295582:295590] CHIP:IN: Secure message was encrypted: Msg ID 673
[1631607838.577391][295582:295590] CHIP:IN: Encrypted message 0xffffb93f3af8 to 0x0000000000BC5C01 of type 16 and protocolId 0 on exchange 39360.
[1631607838.577463][295582:295590] CHIP:IN: Sending msg 0xffffb93f3af8 to 0x0000000000BC5C01 at utc time: 97642040 msec
[1631607838.577533][295582:295590] CHIP:IN: Sending secure msg on generic transport
chip-device-ctrl > [1631607838.577779][295582:295590] CHIP:IN: Secure msg send status ../../src/inet/IPEndPointBasis.cpp:915: Success
[1631607838.577907][295582:295590] CHIP:EM: Flushed pending ack for MsgId:00000151
```

woody-apple, 2022-02-07:
@cindycaowiz is this still happening?

woody-apple, 2022-02-08:
Please re-open if this is still happening.

closed 2022-02-08 by woody-apple