project-chip/connectedhomeip #9816: TE5 Mac Chip Tool Testing: Getting Internal Error response for [TC-TSTAT-2.2] test case

Garik098, 2021-09-18:
While testing Thermostat Cluster 3.2.2. [TC-TSTAT-2.2] Setpoint Test Cases with server as DUT test case, getting internal error response while doing setpoint-raise-lower

shyam@shyams-MacBook-Pro chip-tool % ./out/debug/chip-tool thermostat setpoint-raise-lower 2 -30 1  
[1631917953869] [14707:708132] CHIP: [CTL] Read local id 0x000000000001B669, remote id 0x8B6CC7E459241F2C
[1631917953871] [14707:708132] CHIP: [CTL] Generating NOC
[1631917953871] [14707:708132] CHIP: [CTL] Generating ICAC
[1631917953871] [14707:708132] CHIP: [DL] _Init
[1631917953871] [14707:708132] CHIP: [IN] UDP::Init bind&listen port=5541
[1631917953871] [14707:708132] CHIP: [IN] UDP::Init bind&listen port=5541
[1631917953871] [14707:708132] CHIP: [IN] TransportMgr initialized
[1631917953871] [14707:708132] CHIP: [DIS] Init fabric pairing table with server storage
[1631917953872] [14707:708132] CHIP: [ZCL] Using ZAP configuration...
[1631917953872] [14707:708132] CHIP: [ZCL] deactivate report event
[1631917953872] [14707:708132] CHIP: [DIS] Verifying the received credentials
[1631917953873] [14707:708132] CHIP: [DIS] Added new fabric at index: 1, Initialized: 1
[1631917953873] [14707:708132] CHIP: [DIS] Assigned fabric ID: 0x0000000000000000, node ID: 0x000000000001B669
[1631917953873] [14707:708132] CHIP: [CTL] Joined the fabric at index 1
[1631917953873] [14707:708135] CHIP: [SC] Sent SigmaR1 msg
[1631917953902] [14707:708136] CHIP: [EM] Received message of type 0x31 with vendorId 0x0000 and protocolId 0x0000 on exchange 44249
[1631917953902] [14707:708136] CHIP: [EM] Rxd Ack; Removing MsgId:00000000 from Retrans Table
[1631917953902] [14707:708136] CHIP: [EM] Removed CHIP MsgId:00000000 from RetransTable
[1631917953902] [14707:708136] CHIP: [SC] Received SigmaR2 msg
[1631917953902] [14707:708136] CHIP: [SC] Peer assigned session key ID 10
[1631917953903] [14707:708136] CHIP: [SC] Sending SigmaR3
[1631917953903] [14707:708136] CHIP: [EM] Piggybacking Ack for MsgId:00000000 with msg
[1631917953903] [14707:708136] CHIP: [SC] Sent SigmaR3 msg
[1631917953903] [14707:708136] CHIP: [IN] New secure session created for device 0x8B6CC7E459241F2C, key 10!!
[1631917953903] [14707:708136] CHIP: [TOO] Sending cluster (0x0201) command (0x00) on endpoint 1
[1631917953903] [14707:708136] CHIP: [DMG] ICR moving to [Initialize]
[1631917953903] [14707:708136] CHIP: [DMG] ICR moving to [AddCommand]
[1631917953903] [14707:708136] CHIP: [IN] Secure message was encrypted: Msg ID 0
[1631917953903] [14707:708136] CHIP: [IN] Encrypted message 0x7fec4a81f270 to 0x8B6CC7E459241F2C of type 8 and protocolId 5 on exchange 44250.
[1631917953903] [14707:708136] CHIP: [IN] Sending msg 0x7fec4a81f270 to 0x8B6CC7E459241F2C at utc time: 96177589 msec
[1631917953903] [14707:708136] CHIP: [IN] Sending secure msg on generic transport
[1631917953903] [14707:708136] CHIP: [IN] Secure msg send status ../../third_party/connectedhomeip/src/inet/IPEndPointBasis.cpp:915: Success
[1631917953903] [14707:708136] CHIP: [DMG] ICR moving to [   Sending]
[1631917953911] [14707:708136] CHIP: [EM] Received message of type 0x10 with vendorId 0x0000 and protocolId 0x0000 on exchange 44249
[1631917953911] [14707:708136] CHIP: [EM] Rxd Ack; Removing MsgId:00000000 from Retrans Table
[1631917953911] [14707:708136] CHIP: [EM] Removed CHIP MsgId:00000000 from RetransTable
[1631917953913] [14707:708136] CHIP: [EM] Received message of type 0x09 with vendorId 0x0000 and protocolId 0x0005 on exchange 44250
[1631917953913] [14707:708136] CHIP: [EM] Rxd Ack; Removing MsgId:00000000 from Retrans Table
[1631917953913] [14707:708136] CHIP: [EM] Removed CHIP MsgId:00000000 from RetransTable
[1631917953913] [14707:708136] CHIP: [DMG] InvokeCommand =
[1631917953913] [14707:708136] CHIP: [DMG] {
[1631917953913] [14707:708136] CHIP: [DMG] 	CommandList =
[1631917953913] [14707:708136] CHIP: [DMG] 	[
[1631917953913] [14707:708136] CHIP: [DMG] 		CommandDataElement =
[1631917953913] [14707:708136] CHIP: [DMG] 		{
[1631917953913] [14707:708136] CHIP: [DMG] 			CommandPath =
[1631917953913] [14707:708136] CHIP: [DMG] 			{
[1631917953913] [14707:708136] CHIP: [DMG] 				EndpointId = 0x1,
[1631917953913] [14707:708136] CHIP: [DMG] 				ClusterId = 0x201,
[1631917953913] [14707:708136] CHIP: [DMG] 				CommandId = 0x0,
[1631917953913] [14707:708136] CHIP: [DMG] 			},
[1631917953913] [14707:708136] CHIP: [DMG] 			
[1631917953913] [14707:708136] CHIP: [DMG] 			StatusElement =
[1631917953913] [14707:708136] CHIP: [DMG] 			{
[1631917953913] [14707:708136] CHIP: [DMG] 				GeneralCode = 0xd,
[1631917953913] [14707:708136] CHIP: [DMG] 				ProtocolId = 0x0,
[1631917953913] [14707:708136] CHIP: [DMG] 				protocolCode = 0x85,
[1631917953913] [14707:708136] CHIP: [DMG] 			},
[1631917953913] [14707:708136] CHIP: [DMG] 			
[1631917953913] [14707:708136] CHIP: [DMG] 		},
[1631917953913] [14707:708136] CHIP: [DMG] 		
[1631917953913] [14707:708136] CHIP: [DMG] 	],
[1631917953913] [14707:708136] CHIP: [DMG] 	
[1631917953913] [14707:708136] CHIP: [DMG] }
[1631917953913] [14707:708136] CHIP: [ZCL] DefaultResponse:
[1631917953913] [14707:708136] CHIP: [ZCL]   Transaction: 0x109435f28
[1631917953913] [14707:708136] CHIP: [ZCL]   status: EMBER_ZCL_STATUS_FAILURE (0x01)
[1631917953913] [14707:708136] CHIP: [TOO] Default Failure Response: 0x01
[1631917953913] [14707:708136] CHIP: [DMG] ICR moving to [Uninitiali]
[1631917953913] [14707:708136] CHIP: [EM] Sending Standalone Ack for MsgId:00000001
[1631917953913] [14707:708136] CHIP: [IN] Secure message was encrypted: Msg ID 1
[1631917953913] [14707:708132] CHIP: [TOO] Run command failure: ../../commands/clusters/Commands.h:43: CHIP Error 0x000000AC: Internal error
[1631917953913] [14707:708136] CHIP: [IN] Encrypted message 0x7000089c5518 to 0x8B6CC7E459241F2C of type 16 and protocolId 0 on exchange 44250.
[1631917953913] [14707:708136] CHIP: [IN] Sending msg 0x7000089c5518 to 0x8B6CC7E459241F2C at utc time: 96177600 msec
[1631917953913] [14707:708136] CHIP: [IN] Sending secure msg on generic transport
[1631917953913] [14707:708136] CHIP: [IN] Secure msg send status ../../third_party/connectedhomeip/src/inet/IPEndPointBasis.cpp:915: Success
[1631917953913] [14707:708136] CHIP: [EM] Flushed pending ack for MsgId:00000001
[1631917953913] [14707:708132] CHIP: [CTL] Shutting down the commissioner
[1631917953913] [14707:708132] CHIP: [CTL] Shutting down the controller
[1631917953914] [14707:708132] CHIP: [DL] Inet Layer shutdown
[1631917953914] [14707:708132] CHIP: [DL] BLE layer shutdown
[1631917953914] [14707:708132] CHIP: [DL] System Layer shutdown

chrisbeachresideo, 2021-10-19:
@Garik098 This looks more like an issue with the chip-tool than with the thermostat cluster, which a ZCL error.  
I have made updates to the thermostat server cluster implementation based on the spec PR4025 which clarified a bunch of functionality.  

Can you provide some information regarding the configuration of the DUT so I can configure a DUT in a like manner  and test against the latest chip-tool code.

Garik098, 2021-11-18:
I am using Raspberry pi as my accessory. I am going through the steps mentioned in TC-TSAT-2.2 where at one step it says to do the following,
Test Harness sends Server DUT a SetpointRaiseCommand with the mode field set to Heat (0x00), and the amount set to 0xE2 (-30 units = -3 degrees).

The chip tool command for this step is ./out/debug/chip-tool thermostat setpoint-raise-lower 0 30 1 1

The output I got in TE7 is different (Default Failure Response: 0xc3) from the above one even though it says that it is failure.

"[1637191582087] [2081:63134] CHIP: [CTL] Generating NOC
[1637191582088] [2081:63134] CHIP: [CTL] Generating ICAC
[1637191582088] [2081:63134] CHIP: [DL] AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT START ====
[1637191582088] [2081:63134] CHIP: [DL] AUDIT: * Validate buf1 and buf2 are <<<different every run/boot!>>>
[1637191582088] [2081:63134] CHIP: [DL] AUDIT: * Validate r1 and r2 are <<<different every run/boot!>>>
[1637191582088] [2081:63134] CHIP: [DL] AUDIT: * buf1: 43E6A0F1178486A2D6A9DB59EACC41D0
[1637191582088] [2081:63134] CHIP: [DL] AUDIT: * buf2: 1010D0F29C9974D6AE1B153454FE6D22
[1637191582088] [2081:63134] CHIP: [DL] AUDIT: * r1: 0x53BD7EEC r2: 0x0655B242
[1637191582088] [2081:63134] CHIP: [DL] AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT END ====
[1637191582088] [2081:63134] CHIP: [DL] _Init
[1637191582088] [2081:63134] CHIP: [IN] UDP::Init bind&listen port=5541
[1637191582089] [2081:63134] CHIP: [IN] UDP::Init bound to port=5541
[1637191582089] [2081:63134] CHIP: [IN] UDP::Init bind&listen port=5541
[1637191582089] [2081:63134] CHIP: [IN] UDP::Init bound to port=5541
[1637191582089] [2081:63134] CHIP: [IN] TransportMgr initialized
[1637191582089] [2081:63134] CHIP: [DIS] Init fabric pairing table with server storage
[1637191582089] [2081:63134] CHIP: [CTL] System State Initialized...
[1637191582089] [2081:63134] CHIP: [ZCL] Using ZAP configuration...
[1637191582090] [2081:63134] CHIP: [DIS] Verifying the received credentials
[1637191582090] [2081:63134] CHIP: [DIS] Added new fabric at index: 1, Initialized: 1
[1637191582090] [2081:63134] CHIP: [DIS] Assigned compressed fabric ID: 0x56791D9DD63C9A95, node ID: 0x000000000001B669
[1637191582090] [2081:63134] CHIP: [CTL] Joined the fabric at index 1. Compressed fabric ID is: 0x56791D9DD63C9A95
[1637191582090] [2081:63137] CHIP: [DL] _OnPlatformEvent
[1637191582090] [2081:63137] CHIP: [TOO] Sending command to node 0x1
[1637191582091] [2081:63137] CHIP: [IN] Prepared plaintext message 0x7fbde480d680 to 0x0000000000000000 of type 0x30 and protocolId (0, 0) on exchange 48315i with MessageCounter:3152448820.
[1637191582091] [2081:63137] CHIP: [IN] Sending plaintext msg 0x7fbde480d680 with MessageCounter:3152448820 to 0x0000000000000000 at monotonic time: 6489811 msec
[1637191582091] [2081:63137] CHIP: [SC] Sent Sigma1 msg
[1637191582098] [2081:63137] CHIP: [EM] Received message of type 0x31 with protocolId (0, 0) and MessageCounter:1737183474 on exchange 48315i
[1637191582098] [2081:63137] CHIP: [EM] Rxd Ack; Removing MessageCounter:3152448820 from Retrans Table on exchange 48315i
[1637191582098] [2081:63137] CHIP: [EM] Removed CHIP MessageCounter:3152448820 from RetransTable on exchange 48315i
[1637191582098] [2081:63137] CHIP: [SC] Received Sigma2 msg
[1637191582098] [2081:63137] CHIP: [SC] Peer assigned session session ID 50
[1637191582100] [2081:63137] CHIP: [SC] Sending Sigma3
[1637191582100] [2081:63137] CHIP: [EM] Piggybacking Ack for MessageCounter:1737183474 on exchange: 48315i
[1637191582100] [2081:63137] CHIP: [IN] Prepared plaintext message 0x7fbde480d680 to 0x0000000000000000 of type 0x32 and protocolId (0, 0) on exchange 48315i with MessageCounter:3152448821.
[1637191582100] [2081:63137] CHIP: [IN] Sending plaintext msg 0x7fbde480d680 with MessageCounter:3152448821 to 0x0000000000000000 at monotonic time: 6489820 msec
[1637191582100] [2081:63137] CHIP: [SC] Sent Sigma3 msg
[1637191582107] [2081:63139] CHIP: [EM] Received message of type 0x40 with protocolId (0, 0) and MessageCounter:1737183475 on exchange 48315i
[1637191582107] [2081:63139] CHIP: [EM] Rxd Ack; Removing MessageCounter:3152448821 from Retrans Table on exchange 48315i
[1637191582107] [2081:63139] CHIP: [EM] Removed CHIP MessageCounter:3152448821 from RetransTable on exchange 48315i
[1637191582107] [2081:63139] CHIP: [SC] Success status report received. Session was established
[1637191582107] [2081:63139] CHIP: [IN] New secure session created for device 0x0000000000000001, key 50!!
[1637191582107] [2081:63139] CHIP: [TOO] Sending cluster (0x00000201) command (0x00000000) on endpoint 1
[1637191582107] [2081:63139] CHIP: [DMG] ICR moving to [AddingComm]
[1637191582107] [2081:63139] CHIP: [DMG] ICR moving to [AddedComma]
[1637191582108] [2081:63139] CHIP: [IN] Prepared encrypted message 0x7fbde480d680 to 0x0000000000000001 of type 0x8 and protocolId (0, 1) on exchange 48316i with MessageCounter:0.
[1637191582108] [2081:63139] CHIP: [IN] Sending encrypted msg 0x7fbde480d680 with MessageCounter:0 to 0x0000000000000001 at monotonic time: 6489828 msec
[1637191582108] [2081:63139] CHIP: [DMG] ICR moving to [CommandSen]
[1637191582108] [2081:63139] CHIP: [EM] Sending Standalone Ack for MessageCounter:1737183475 on exchange 48315i
[1637191582108] [2081:63139] CHIP: [IN] Prepared plaintext message 0x700004277450 to 0x0000000000000000 of type 0x10 and protocolId (0, 0) on exchange 48315i with MessageCounter:3152448822.
[... cut ...]
[1637191582111] [2081:63134] CHIP: [-] ../../zzz_generated/chip-tool/zap-generated/cluster/Commands.h:45: CHIP Error 0x000000AC: Internal error at ../../examples/chip-tool/commands/common/CHIPCommand.cpp:84
[1637191582111] [2081:63134] CHIP: [TOO] Run command failure: ../../zzz_generated/chip-tool/zap-generated/cluster/Commands.h:45: CHIP Error 0x000000AC: Internal error
[1637191582150] [2081:63134] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1637191582150] [2081:63134] CHIP: [DL] Inet Layer shutdown
[1637191582150] [2081:63134] CHIP: [DL] BLE layer shutdown
[1637191582150] [2081:63134] CHIP: [DL] System Layer shutdown
"

chrisbeach59, 2021-11-22:
@Garik098 

- Are you using the thermostat-app on the RPI TE7 image OOB? 
- The OOB image has the Feature Map defined to be HEAT | COOL | AUTO.  
- There were a bunch of changes based on the cluster spec changes, which impose constraints on the DUT, especially when AUTO is set to true.
- The test plan was never updated after the changes were implemented. 
- Can you send me the exact sequence of commands sent to the DUT.  The spread sheet is not really complete and also has errors on it. 
- I need this information to determine if the behavior you are observing is as designed per the spec, or a bug.

closed 2021-12-15 by chrisbeach59