project-chip/connectedhomeip #9819: TE5 Mac Chip Tool testing: Getting Invalid Argument error for key code 1000 in TC-MC-3.1 Step 9

Garik098, 2021-09-19:
As per the expectation, for the Media Cluster TC-MC-3.1 step 9, the response should be value 1 and unsupported key, but with the Mac Chip Tool, getting an invalid argument error response. Below is the log,

Command: ./out/debug/chip-tool keypadinput send-key 1000 1
"[1631840870880] [5285:196515] CHIP: [CTL] Read local id 0x000000000001B669, remote id 0x6E9FA1F2CB243D07
[1631840870883] [5285:196515] CHIP: [CTL] Generating NOC
[1631840870883] [5285:196515] CHIP: [CTL] Generating ICAC
[1631840870883] [5285:196515] CHIP: [DL] _Init
[1631840870883] [5285:196515] CHIP: [IN] UDP::Init bind&listen port=5541
[1631840870883] [5285:196515] CHIP: [IN] UDP::Init bind&listen port=5541
[1631840870884] [5285:196515] CHIP: [IN] TransportMgr initialized
[1631840870884] [5285:196515] CHIP: [DIS] Init fabric pairing table with server storage
[1631840870884] [5285:196515] CHIP: [ZCL] Using ZAP configuration...
[1631840870884] [5285:196515] CHIP: [ZCL] deactivate report event
[1631840870884] [5285:196515] CHIP: [DIS] Verifying the received credentials
[1631840870885] [5285:196515] CHIP: [DIS] Added new fabric at index: 1, Initialized: 1
[1631840870885] [5285:196515] CHIP: [DIS] Assigned fabric ID: 0x0000000000000000, node ID: 0x000000000001B669
[1631840870885] [5285:196515] CHIP: [CTL] Joined the fabric at index 1
[1631840870885] [5285:196515] CHIP: [TOO] InitArgs: Invalid argument KeyCode: 1000
[1631840870885] [5285:196515] CHIP: [TOO] Run command failure: ../../commands/common/Commands.cpp:212: CHIP Error 0x0000002F: Invalid argument
[1631840870885] [5285:196515] CHIP: [CTL] Shutting down the commissioner
[1631840870885] [5285:196515] CHIP: [CTL] Shutting down the controller
[1631840870885] [5285:196515] CHIP: [DL] Inet Layer shutdown
[1631840870885] [5285:196515] CHIP: [DL] BLE layer shutdown
[1631840870885] [5285:196515] CHIP: [DL] System Layer shutdown
"

woody-apple, 2022-01-25:
Marking older bugs as closed, please re-verify with a more recent CHIP version.

closed 2022-01-25 by woody-apple

liamgonyea, 2022-02-03:
### Update 
Testing with commit ID: `b3ad0ff7253214d4d73f1871617338965961d263`

Test case `TC-MC-3.1`, Step 9, is passing with the above testing setup for chip-tool as the controller and chip-tv-app as the accessory.