project-chip/connectedhomeip #9826: Fix ByteSpan overload of `TLVReader::Get` to only permit reading byte strings

mrjerryjohns, 2021-09-19:
#### Problem

`TLVReader::Get(ByteSpan &v)` today permits reading both UTF8 strings as well as Byte strings. Given the presence of a dedicated `GetString` method, the former method should be fixed to only read Byte strings.

This would require further auditing of the code to fix callers that aren't compliant.

mrjerryjohns, 2021-09-19:
As part of this, we should also add a dedicated `TLVReader::Get(Span<const char> &v)` option as well for reading UTF-8 strings 'in-place'.

bzbarsky-apple, 2021-09-20:
We have such a dedicated method already on, the contiguous buffer reader.  On the non-contiguous-buffer reader, the method can't be safely implemented, because the string may span buffers.

The right thing to do is to move all consumers that want these in-place methods over to the contiguous buffer reader....