project-chip/connectedhomeip #9832: provisioning over BLE on raspi is broken as BLE advertisement disabled

cjandhyala, 2021-09-20:
With ToT (fb53b776694a5910a8e0f36fa810d75c02d877be) Provisioning over BLE on raspi is broken.  From the control log statement "Network already provisioned. Disabling BLE advertisement" , it looks like the BLE advertisement is disabled, hence the controller couldn't provision the device.  Steps to reproduce

1. Build controlee
a) scripts/examples/gn_build_example.sh examples/all-clusters-app/linux/ out/all-clusters-app
b) sudo rm /tmp/chip_*
c) Launch app with sudo ./out/all-clusters-app/chip-all-clusters-app --wifi
2. Build Controller 
a)  ./scripts/build_python.sh --clusters_for_ip_commissioning true --chip_detail_logging true --chip_mdns platform
b) launch controller  $ chip-device-ctrl -n 9999
c) connect -ble 3840 20202021 12344321

Here is controller log
~~~~~~~~~~~~~~~~~~

chip-device-ctrl > connect -ble 3840 20202021 12344321
Device is assigned with nodeid = 12344321
[1632163222.726891][163780:163788] CHIP:SC: Assigned local session key ID 1
[1632163222.727097][163780:163788] CHIP:IN: Build plaintext message 0xffff76f7b1c0 to 0x0000000000000000 of type 32 and protocolId 0 on exchange 31631 with MessageCounter 320294574.
[1632163222.727174][163780:163788] CHIP:IN: Sending plaintext msg 0xffff76f7b1c0 to 0x0000000000000000 at utc time: 869656582 msec
[1632163222.727219][163780:163788] CHIP:IN: Sending msg on generic transport
[1632163222.727265][163780:163788] CHIP:IN: Message appended to BLE send queue
[1632163222.727307][163780:163788] CHIP:SC: Sent PBKDF param request
[1632163222.727372][163780:163788] CHIP:CTL: SyncSetKeyValue on StartKeyID
[1632163222.727850][163780:163824] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1632163222.728089][163780:163784] CHIP:DL: TRACE: Bus acquired for name C-7fc4
[1632163222.738952][163780:163788] CHIP:DL: PlatformBlueZInit init success
[1632163222.745181][163780:163784] CHIP:BLE: BLE removing known devices.
[1632163222.746934][163780:163784] CHIP:BLE: BLE initiating scan.
[1632163223.177155][163780:163784] CHIP:BLE: Device 5F:A2:16:89:D2:E8 does not look like a CHIP device.
[1632163223.209088][163780:163784] CHIP:BLE: Device 53:64:48:90:DB:92 does not look like a CHIP device.
[1632163223.224314][163780:163784] CHIP:BLE: Device 69:47:EA:D2:58:EF does not look like a CHIP device.
[1632163223.230465][163780:163784] CHIP:BLE: Device 7A:7D:36:5E:5A:EF does not look like a CHIP device.
[1632163223.286877][163780:163784] CHIP:BLE: Device 4C:CB:C8:FA:57:F7 does not look like a CHIP device.
[1632163223.293974][163780:163784] CHIP:BLE: Device 54:6E:31:CA:1F:B3 does not look like a CHIP device.
[1632163223.295279][163780:163784] CHIP:BLE: Device 49:42:C3:E9:98:A9 does not look like a CHIP device.
[1632163223.301933][163780:163784] CHIP:BLE: Device 4E:8E:29:C7:A4:05 does not look like a CHIP device.
[1632163223.326934][163780:163784] CHIP:BLE: Device 7C:3B:21:6B:08:D7 does not look like a CHIP device.
[1632163223.388569][163780:163784] CHIP:BLE: Device 35:54:B9:7D:5A:DB does not look like a CHIP device.
[1632163223.529109][163780:163784] CHIP:BLE: Device 2F:D8:51:76:4F:D1 does not look like a CHIP device.
[1632163223.849086][163780:163784] CHIP:BLE: Device 54:6E:31:CA:1F:B3 does not look like a CHIP device.
[1632163224.388422][163780:163784] CHIP:BLE: Device E4:20:74:8E:20:C3 does not look like a CHIP device.
[1632163224.517046][163780:163784] CHIP:BLE: Device FF:8E:7E:77:29:34 does not look like a CHIP device.
[1632163232.750974][163780:163784] CHIP:IN: Clearing BLE pending packets.
[1632163232.751679][163780:163784] CHIP:IN: BleConnection Error: ../../src/platform/Linux/BLEManagerImpl.cpp:821: CHIP Error 0x00000032: Timeout
[1632163252.729005][163780:163788] CHIP:CTL: SyncSetKeyValue on StartKeyID
Failed to establish secure session to device: 50
[1632163252.729402][163780:163788] CHIP:SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
CHIP Error 0x00000032: Timeout
[1632163252.729482][163780:163788] CHIP:CTL: SyncSetKeyValue on StartKeyID
Failed to establish secure session to device: 50

Raspi Device Log
~~~~~~~~~~~~~~
$ sudo ./out/all-clusters-app/chip-all-clusters-app --wifi
[1632163075.935912][6080:6080] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-KzmwtY)
[1632163075.936115][6080:6080] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1632163075.936321][6080:6080] CHIP:DL: writing settings to file (/tmp/chip_config.ini-4Bq0fY)
[1632163075.936434][6080:6080] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1632163075.936577][6080:6080] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-36s6UW)
[1632163075.936682][6080:6080] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1632163075.936884][6080:6080] CHIP:DL: Device Configuration:
[1632163075.936913][6080:6080] CHIP:DL:   Serial Number: TEST_SN
[1632163075.936933][6080:6080] CHIP:DL:   Vendor Id: 9050 (0x235A)
[1632163075.936953][6080:6080] CHIP:DL:   Product Id: 65279 (0xFEFF)
[1632163075.936974][6080:6080] CHIP:DL:   Product Revision: 0
[1632163075.936995][6080:6080] CHIP:DL:   Setup Pin Code: 20202021
[1632163075.937016][6080:6080] CHIP:DL:   Setup Discriminator: 3840 (0xF00)
[1632163075.937038][6080:6080] CHIP:DL:   Manufacturing Date: (not set)
[1632163075.937056][6080:6080] CHIP:DL:   Device Type: 65535 (0xFFFF)
[1632163075.937109][6080:6080] CHIP:SVR: SetupQRCode: [MT:YNJV7VSC00KA0648G00]
[... 85 lines cut ...]
[1632163076.053891][6080:6080] CHIP:DIS: Scheduling Discovery timeout in secs=900
[1632163076.053924][6080:6080] CHIP:IN: CASE Server enabling CASE session setups
[1632163076.053949][6080:6080] CHIP:SVR: Server Listening...
[1632163076.054233][6080:6086] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1632163076.054351][6080:6083] CHIP:DL: TRACE: Bus acquired for name MATTER-3840
[1632163076.054570][6080:6083] CHIP:DL: CREATE service object at /chipoble/17c0/service
[1632163076.055321][6080:6083] CHIP:DL: Create characteristic object at /chipoble/17c0/service/c1
[1632163076.055856][6080:6083] CHIP:DL: Create characteristic object at /chipoble/17c0/service/c2
[1632163076.056202][6080:6083] CHIP:DL: CHIP BTP C1 /chipoble/17c0/service
[1632163076.056231][6080:6083] CHIP:DL: CHIP BTP C2 /chipoble/17c0/service
[1632163076.056250][6080:6083] CHIP:DL: CHIP_ENABLE_ADDITIONAL_DATA_ADVERTISING is TRUE
[1632163076.056271][6080:6083] CHIP:DL: Create characteristic object at /chipoble/17c0/service/c3
[1632163076.056732][6080:6083] CHIP:DL: CHIP BTP C3 /chipoble/17c0/service
[1632163076.063417][6080:6080] CHIP:DL: PlatformBlueZInit init success
[1632163076.069533][6080:6083] CHIP:DL: BluezPeripheralRegisterAppDone done
[1632163076.069601][6080:6080] CHIP:DL: HandlePlatformSpecificBLEEvent 16392
[1632163154.229765][6080:6080] CHIP:DIS: Replying to DNS-SD service listing request
[1632163154.229858][6080:6080] CHIP:DIS: Replying to DNS-SD service listing request
[1632163154.229910][6080:6080] CHIP:DIS: Replying to DNS-SD service listing request
[1632163154.229964][6080:6080] CHIP:DIS: Replying to DNS-SD service listing request
[1632163154.230017][6080:6080] CHIP:DIS: Replying to DNS-SD service listing request
[1632163154.230067][6080:6080] CHIP:DIS: Replying to DNS-SD service listing request
[1632163154.230173][6080:6080] CHIP:DIS: Replying to DNS-SD service listing request
[1632163154.230233][6080:6080] CHIP:DIS: Broadcasting mDns reply
[1632163154.465062][6080:6080] CHIP:DIS: Broadcasting mDns reply

cjandhyala, 2021-09-21:
Here is further debug info,
The API IsWiFiStationProvisioned() is returning true. When I run the  lighting app in debugger, it returns false. so I am not hitting that if condition , so its not printing the log msg "Network already provisioned. Disabling BLE advertisement"

if I disable the wlan0 interface (I have both eth0 and wlan0 interfaces) on my raspi accessory, I am able to provision the device (I see the device advertising with compressed fabricID and nodeID) but the resolve is failing with a timeout error.

chip-device-ctrl > resolve 15370486864828282265 12344321
[1632176829.562266][245794:245802] CHIP:DL: Avahi resolve found
[1632176829.562407][245794:245802] CHIP:DIS: Node ID resolved for 0x0000000000BC5C01 to [192.168.6.215]:5540
[1632176829.563321][245794:245802] CHIP:IN: Build plaintext message 0x24617928 to 0x0000000000000000 of type 48 and protocolId 0 on exchange 25673 with MessageCounter 879511743.
[1632176829.563407][245794:245802] CHIP:IN: Sending plaintext msg 0x24617928 to 0x0000000000000000 at utc time: 883263419 msec
[1632176829.563447][245794:245802] CHIP:IN: Sending msg on generic transport
[1632176829.563625][245794:245802] CHIP:SC: Sent SigmaR1 msg
[1632176829.563720][245794:245802] CHIP:CTL: SyncSetKeyValue on PairedDevicebc5c01
Node address has been updated
[1632176829.563841][245794:245802] CHIP:CTL: OperationalDiscoveryComplete for device ID 12344321
[1632176829.563876][245794:245802] CHIP:CTL: Operational certificates already provisioned for this device
[1632176829.563923][245794:245802] CHIP:CTL: SyncSetKeyValue on PairedDevicebc5c01
[1632176829.563957][245794:245802] CHIP:CTL: SyncSetKeyValue on StartKeyID
[1632176829.569448][245794:245802] CHIP:EM: Received message of type 0x31 with vendorId 0x0000 and protocolId 0x0000 on exchange 25673
[1632176829.569528][245794:245802] CHIP:EM: ec id: 25673, Delegate: 0x(nil)
[1632176829.569573][245794:245802] CHIP:EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:295: CHIP Error 0x0000002A: Invalid message type
[1632176834.509324][245794:245802] CHIP:IN: Received a duplicate message with MessageCounter: 781509671
[1632176834.509442][245794:245802] CHIP:EM: Received message of type 0x31 with vendorId 0x0000 and protocolId 0x0000 on exchange 25673
[1632176834.509534][245794:245802] CHIP:EM: ec id: 25673, Delegate: 0x(nil)
[1632176834.509605][245794:245802] CHIP:EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:295: CHIP Error 0x0000002A: Invalid message type
[1632176834.514979][245794:245802] CHIP:IN: Sending plaintext msg 0x24617928 to 0x0000000000000000 at utc time: 883268370 msec
[1632176834.515050][245794:245802] CHIP:IN: Sending msg on generic transport
[1632176834.515220][245794:245802] CHIP:EM: Retransmit MessageCounter:346C48BF Send Cnt 1
[1632176834.516387][245794:245802] CHIP:EM: Received message of type 0x10 with vendorId 0x0000 and protocolId 0x0000 on exchange 25673
[1632176834.516559][245794:245802] CHIP:EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:268: CHIP Error 0x00000070: Unsolicited msg with originator bit clear
[1632176834.762183][245794:245802] CHIP:IN: Received a duplicate message with MessageCounter: 781509671
[1632176834.762278][245794:245802] CHIP:EM: Received message of type 0x31 with vendorId 0x0000 and protocolId 0x0000 on exchange 25673
[1632176834.762353][245794:245802] CHIP:EM: ec id: 25673, Delegate: 0x(nil)
[1632176834.762408][245794:245802] CHIP:EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:295: CHIP Error 0x0000002A: Invalid message type
[1632176834.770635][245794:245802] CHIP:IN: Sending plaintext msg 0x24617928 to 0x0000000000000000 at utc time: 883268626 msec
[1632176834.770695][245794:245802] CHIP:IN: Sending msg on generic transport
[1632176834.770849][245794:245802] CHIP:EM: Retransmit MessageCounter:346C48BF Send Cnt 2
[1632176834.771930][245794:245802] CHIP:EM: Received message of type 0x10 with vendorId 0x0000 and protocolId 0x0000 on exchange 25673
[1632176834.772006][245794:245802] CHIP:EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:268: CHIP Error 0x00000070: Unsolicited msg with originator bit clear
[1632176835.017842][245794:245802] CHIP:IN: Received a duplicate message with MessageCounter: 781509671
[1632176835.017948][245794:245802] CHIP:EM: Received message of type 0x31 with vendorId 0x0000 and protocolId 0x0000 on exchange 25673
[1632176835.018092][245794:245802] CHIP:EM: ec id: 25673, Delegate: 0x(nil)
[1632176835.018150][245794:245802] CHIP:EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:295: CHIP Error 0x0000002A: Invalid message type
[1632176835.027469][245794:245802] CHIP:IN: Sending plaintext msg 0x24617928 to 0x0000000000000000 at utc time: 883268883 msec
[1632176835.027524][245794:245802] CHIP:IN: Sending msg on generic transport
[1632176835.027663][245794:245802] CHIP:EM: Retransmit MessageCounter:346C48BF Send Cnt 3
[1632176835.028781][245794:245802] CHIP:EM: Received message of type 0x10 with vendorId 0x0000 and protocolId 0x0000 on exchange 25673
[1632176835.028852][245794:245802] CHIP:EM: OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:268: CHIP Error 0x00000070: Unsolicited msg with originator bit clear
[1632176835.283296][245794:245802] CHIP:EM: Failed to Send CHIP MessageCounter:346C48BF sendCount: 3 max retries: 3
[1632176839.566805][245794:245802] CHIP:SC: CASESession timed out while waiting for a response from the peer. Expected message type was 49
[1632176839.566907][245794:245802] CHIP:CTL: Device connection failed. Error ../../src/protocols/secure_channel/CASESession.cpp:276: CHIP Error 0x00000032: Timeout
Failed to commission: 50

andy31415, 2021-09-21:
@cjandhyala could you do some additional steps to try to help finding an owner here:

 - can this be reproduced without RPI and using only mac or linux (this helps widen the developer pool who can debug this)?
 - could you bisect to see what change may have introduced this regression?

yunhanw-google, 2021-09-22:
Looks like the issue is to relate resolve after provisioning(mdns discovery failure)

yunhanw-google, 2021-09-22:
@cjandhyala for his your initial description regarding BLE adv not up after provisioning is completed, after you disable wifi interface, your ble advertisement comes back, as the log says ""Network already provisioned. Disabling BLE advertisement", it is expected behavior per apple's PR https://github.com/project-chip/connectedhomeip/pull/4597

PR #9933 (2021-09-24): Do not disable BLE if a fabric has not been provisioned on the device
#### Problem
* Fixes #9832 

#### Change overview
Server application was checking the network status to determine if the the device is already provisioned. This code was added long ago, and worked since fabric commissioning was not supported then.

This change uses fabric table to determine if the device is currently commissioned, instead of checking for network status.

#### Testing
Tested BLE commissioning process on m5stack (by configuring it with WiFi credentials through menuconfig) and chip-tool.
closed 2021-09-24 by vivien-apple