project-chip/connectedhomeip #9854: boot loop in esp32 when generating rotating device id

mspang, 2021-09-21:
#### Problem

The following crash stack is observed when booting for the first time on m5stack:

```
0x40090962: panic_abort at /home/spang/tools/esp-idf/components/esp_system/panic.c:341

0x40090f39: esp_system_abort at /home/spang/tools/esp-idf/components/esp_system/system_api.c:106

0x40097656: abort at /home/spang/tools/esp-idf/components/newlib/abort.c:46

0x4013bef2: esp_sha_read_digest_state at /home/spang/tools/esp-idf/components/mbedtls/port/esp32/sha.c:272

0x40139886: mbedtls_sha256_finish_ret at /home/spang/tools/esp-idf/components/mbedtls/port/esp32/esp_sha256.c:395

0x400ed940: chip::Crypto::Hash_SHA256_stream::Finish(chip::Span<unsigned char>&) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:267

0x400f2334: chip::AdditionalDataPayloadGenerator::generateRotatingDeviceIdAsBinary(unsigned short, char const*, unsigned int, chip::Span<unsigned char>&) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/setup_payload/AdditionalDataPayloadGenerator.cpp:101

0x400f2399: chip::AdditionalDataPayloadGenerator::generateRotatingDeviceIdAsHexString(unsigned short, char const*, unsigned int, char*, unsigned int, unsigned int&) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/setup_payload/AdditionalDataPayloadGenerator.cpp:117

0x400d8666: chip::app::MdnsServer::GenerateRotatingDeviceId(char*, unsigned int) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Mdns.cpp:489

0x400d883c: chip::app::MdnsServer::Advertise(bool, chip::Mdns::CommissioningMode) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Mdns.cpp:335

0x400d894d: chip::app::MdnsServer::AdvertiseCommissionableNode(chip::Mdns::CommissioningMode) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Mdns.cpp:400

0x400d8a61: chip::app::MdnsServer::StartServer(chip::Mdns::CommissioningMode) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Mdns.cpp:457

0x400d7ef8: chip::CommissioningWindowManager::OpenCommissioningWindow() at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/CommissioningWindowManager.cpp:172

0x400d81c6: chip::CommissioningWindowManager::OpenBasicCommissioningWindow(unsigned short, chip::CommissioningWindowAdvertisement) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/CommissioningWindowManager.cpp:196

0x400d90bd: chip::Server::Init(AppDelegate*, unsigned short, unsigned short) at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../third_party/connectedhomeip/src/app/server/Server.cpp:148

0x400d6c70: app_main at /ssd/src/connectedhomeip5/examples/all-clusters-app/esp32/build/../main/main.cpp:633

0x400d3e34: main_task at /home/spang/tools/esp-idf/components/esp32/cpu_start.c:609

0x40090f41: vPortTaskWrapper at /home/spang/tools/esp-idf/components/freertos/xtensa/port.c:143

```

Usually the board does recover after several crashes.

 #### Proposed Solution

Unknown, but disabling rotating device id avoids these symptoms.

@dhrishi 
@sweetymhaiske 
@gjc13

bzbarsky-apple, 2021-09-21:
@mspang What version of the esp-idf SDK are you using?

bzbarsky-apple, 2021-09-22:
Yeah, it would be nice if the build would at least notice the SDK version is an old one with a known problem and complain....