project-chip/connectedhomeip #9882: Writing a maximum allowed length value to any string attribute fails

bluebin14, 2021-09-22:
#### Problem
Writing a maximum allowed length value to any string attribute fails with EMBER_ZCL_STATUS_INVALID_VALUE (0x87). Examples:
* writing 2 character string to Basic/Location attribute
* writing 32 character string to Basic/UserLabel attribute

chip-device-ctrl > zclwrite Basic UserLabel 12344321 0 0 12345678901234567890123456789012
...
[1632301717.783335][2312:2320] CHIP:ZCL: WriteResponse:
[1632301717.783403][2312:2320] CHIP:ZCL:   status: EMBER_ZCL_STATUS_INVALID_VALUE (0x87)

 #### Proposed Solution
Fix codegen since it is a general issue.

vivien-apple, 2021-09-22:
That is strange. I tried with `chip-tool` with the attached command and it returns a `EMBER_ZCL_STATUS_SUCCESS`:
```
./out/debug/standalone/chip-tool basic write user-label 12345678901234567890123456789012 0
```

Can you try running it with `chip-tool` just in case ?

bluebin14, 2021-09-22:
It fails with chip-tool + Raspberry + Thread device setup.

~/connectedhomeip/src/controller/python$ out/host/chip-tool basic write user-label 12345678901234567890123456789012 0
[1632312086.615529][27645:27645] CHIP:CTL: Read local id 0x000000000001B669, remote id 0x439A72CD4F6363AD
[1632312086.617942][27645:27645] CHIP:CTL: Generating NOC
[1632312086.618289][27645:27645] CHIP:CTL: Generating ICAC
[1632312086.623408][27645:27645] CHIP:IN: UDP::Init bind&listen port=5541
[1632312086.623543][27645:27645] CHIP:IN: UDP::Init bind&listen port=5541
[1632312086.623632][27645:27645] CHIP:IN: TransportMgr initialized
[1632312086.623657][27645:27645] CHIP:DIS: Init fabric pairing table with server storage
[1632312086.627300][27645:27645] CHIP:DL: Avahi client registered
[1632312086.628293][27645:27645] CHIP:ZCL: Using ZAP configuration...
[1632312086.628380][27645:27645] CHIP:ZCL: deactivate report event
[1632312086.629105][27645:27645] CHIP:DIS: Verifying the received credentials
[1632312086.630916][27645:27645] CHIP:DIS: Added new fabric at index: 1, Initialized: 1
[1632312086.631058][27645:27645] CHIP:DIS: Assigned fabric ID: 0x0000000000000000, node ID: 0x000000000001B669
[1632312086.631093][27645:27645] CHIP:CTL: Joined the fabric at index 1
[1632312086.631390][27645:27652] CHIP:DL: CHIP task running
[1632312086.632210][27645:27652] CHIP:SC: Sent SigmaR1 msg
[1632312087.170865][27645:27652] CHIP:EM: Received message of type 0x31 with vendorId 0x0000 and protocolId 0x0000 on exchange 39017
[1632312087.170989][27645:27652] CHIP:EM: Rxd Ack; Removing MsgId:00000000 from Retrans Table
[1632312087.171043][27645:27652] CHIP:EM: Removed CHIP MsgId:00000000 from RetransTable
[1632312087.171120][27645:27652] CHIP:SC: Received SigmaR2 msg
[1632312087.171190][27645:27652] CHIP:SC: Peer assigned session key ID 2
[1632312087.177781][27645:27652] CHIP:SC: Sending SigmaR3
[1632312087.178266][27645:27652] CHIP:EM: Piggybacking Ack for MsgId:00000000 with msg
[1632312087.178481][27645:27652] CHIP:SC: Sent SigmaR3 msg
[1632312087.178529][27645:27652] CHIP:IN: New secure session created for device 0x439A72CD4F6363AD, key 2!!
[1632312087.178619][27645:27652] CHIP:TOO: Sending cluster (0x0028) command (0x01) on endpoint 0
[1632312087.178796][27645:27652] CHIP:DMG: WriteClient moving to [Initialize]
[1632312087.178854][27645:27652] CHIP:DMG: WriteClient moving to [AddAttribu]
[1632312087.178994][27645:27652] CHIP:IN: Secure message was encrypted: Msg ID 0
[1632312087.179040][27645:27652] CHIP:IN: Encrypted message 0xaaaaf451cf80 to 0x439A72CD4F6363AD of type 6 and protocolId 5 on exchange 39018.
[1632312087.179082][27645:27652] CHIP:IN: Sending msg 0xaaaaf451cf80 to 0x439A72CD4F6363AD at utc time: 10970132 msec
[1632312087.179115][27645:27652] CHIP:IN: Sending secure msg on generic transport
[1632312087.179237][27645:27652] CHIP:IN: Secure msg send status ../../../../../src/inet/IPEndPointBasis.cpp:915: Success
[1632312087.179283][27645:27652] CHIP:DMG: WriteClient moving to [AwaitingRe]
[1632312087.531856][27645:27652] CHIP:EM: Received message of type 0x10 with vendorId 0x0000 and protocolId 0x0000 on exchange 39017
[1632312087.531938][27645:27652] CHIP:EM: Rxd Ack; Removing MsgId:00000000 from Retrans Table
[1632312087.531974][27645:27652] CHIP:EM: Removed CHIP MsgId:00000000 from RetransTable
[1632312087.642649][27645:27652] CHIP:EM: Received message of type 0x07 with vendorId 0x0000 and protocolId 0x0005 on exchange 39018
[1632312087.642736][27645:27652] CHIP:EM: Rxd Ack; Removing MsgId:00000000 from Retrans Table
[1632312087.642772][27645:27652] CHIP:EM: Removed CHIP MsgId:00000000 from RetransTable
[1632312087.642836][27645:27652] CHIP:DMG: WriteResponse =
[1632312087.642875][27645:27652] CHIP:DMG: {
[1632312087.642908][27645:27652] CHIP:DMG: 	AttributeStatusList =
[1632312087.642948][27645:27652] CHIP:DMG: 	[
[1632312087.642984][27645:27652] CHIP:DMG: 		AttributeStatusElement =
[1632312087.643032][27645:27652] CHIP:DMG: 		{
[1632312087.643075][27645:27652] CHIP:DMG: 			AttributePath =
[1632312087.643124][27645:27652] CHIP:DMG: 			{
[1632312087.643174][27645:27652] CHIP:DMG: 				FieldTag = 0x5,
[1632312087.643230][27645:27652] CHIP:DMG: 				NodeId = 0x439a72cd4f6363ad,
[1632312087.643282][27645:27652] CHIP:DMG: 				ClusterId = 0x28,
[1632312087.643332][27645:27652] CHIP:DMG: 				EndpointId = 0x0,
[1632312087.643380][27645:27652] CHIP:DMG: 			}
[1632312087.643439][27645:27652] CHIP:DMG: 				
[1632312087.643485][27645:27652] CHIP:DMG: 			StatusElement =
[1632312087.643536][27645:27652] CHIP:DMG: 			{
[1632312087.643585][27645:27652] CHIP:DMG: 				GeneralCode = 0x1,
[1632312087.643635][27645:27652] CHIP:DMG: 				ProtocolId = 0x0,
[1632312087.643710][27645:27652] CHIP:DMG: 				protocolCode = 0x87,
[1632312087.643882][27645:27652] CHIP:DMG: 			},
[1632312087.643968][27645:27652] CHIP:DMG: 			
[1632312087.644030][27645:27652] CHIP:DMG: 		},
[1632312087.644105][27645:27652] CHIP:DMG: 		
[1632312087.644157][27645:27652] CHIP:DMG: 	],
[1632312087.644225][27645:27652] CHIP:DMG: 	
[1632312087.644273][27645:27652] CHIP:DMG: }
[1632312087.644381][27645:27652] CHIP:ZCL: WriteResponse:
[1632312087.644429][27645:27652] CHIP:ZCL:   status: EMBER_ZCL_STATUS_INVALID_VALUE (0x87)
[1632312087.644483][27645:27652] CHIP:TOO: Default Failure Response: 0x87
[1632312087.644579][27645:27652] CHIP:DMG: WriteClient moving to [Uninitiali]
[1632312087.644648][27645:27652] CHIP:EM: Sending Standalone Ack for MsgId:00000001
[... cut ...]
[1632312087.645690][27645:27645] CHIP:CTL: Shutting down the controller
[1632312087.647222][27645:27645] CHIP:DL: Inet Layer shutdown
[1632312087.647378][27645:27650] CHIP:IN: Async DNS worker thread woke up.
[1632312087.647464][27645:27650] CHIP:IN: Async DNS worker thread exiting.
[1632312087.647497][27645:27651] CHIP:IN: Async DNS worker thread woke up.
[1632312087.647598][27645:27651] CHIP:IN: Async DNS worker thread exiting.
[1632312087.647869][27645:27645] CHIP:DL: BLE layer shutdown
[1632312087.647932][27645:27645] CHIP:DL: System Layer shutdown

bluebin14, 2021-09-22:
Device log when problem occurs, with data print:

14:01:27.234  11020.863        6  0 Decrypted msgId 0 keyId 2:
                                    05 06 6A 98 05 00 15 36 01 15 37 00 24 03 05 27 
14:01:27.254  11020.883       20  0 00 AD 63 63 4F CD 72 9A 43 24 02 28 24 01 00 18 
                                    30 02 20 31 32 33 34 35 36 37 38 39 30 31 32 33 
14:01:27.255  11020.884        1  0 34 35 36 37 38 39 30 31 32 33 34 35 36 37 38 39 
                                    30 31 32 24 01 00 18 18 18 
14:01:27.264  11020.893        9  0 I: 251046 [EM]Received message of type 0x06 with vendorId 0x0000 and protocolId 0x0005 on exchange 39018
14:01:27.268  11020.897        4  0 D: 251055 [EM]ec id: 39018, Delegate: 0x0x20005138
14:01:27.272  11020.901        4  0 D: 251060 [-]Receive Write request
14:01:27.276  11020.905        4  0 D: 251063 [DMG]IM RH moving to [Initialized]
14:01:27.288  11020.917       12  0 D: 251068 [ZCL]Data to write exceedes the attribute size claimed.
                                    D: 251074 [DMG]IM RH moving to [AddAttributeStatusCode]
14:01:27.305  11020.934       17  0 Before encryption msgId 1 keyId 3:
                                    06 07 6A 98 05 00 00 00 00 00 15 36 00 15 37 00 
                                    24 03 05 27 00 AD 63 63 4F CD 72 9A 43 24 02 28 
                                    24 01 00 18 36 01 04 01 04 00 04 87 18 18 18 18

bluebin14, 2021-09-22:
Fixed.

closed 2021-09-22 by bluebin14