project-chip/connectedhomeip #9887: chip-tool (controller) and lighting-app device (k32w) not working

saumildmehta, 2021-09-22:
I have been trying to run chip-tool example running on Raspberry pi (chip-tool, otbr-agent) + K32W dongle (RCP)
And lighting-app k32w example running on K32W dongle.

I am trying to commission device over BLE and using THREAD as operational network.
Somehow **commissioning is failing** with latest code. lighting-app device is still showing **DETATCHED** state even it got all operational dataset.
Analysing sniffer packets, **Device is not advertising on network and not responding to mDNS query from coordinator**.
I have attached logs of both the examples.
End device lighting app is created with gn args "k32w_sdk_root=\"${K32W061_SDK_ROOT}\" chip_with_OM15082=0 chip_with_ot_cli=1 is_debug=false chip_crypto=\"mbedtls\" chip_with_se05x=0"

Is that something I did wrong? Any help will be appreciated
I have seen same behaviour when I was having code of 2 months older. In that atleast device was getting **CHILD** role and **not responding to mDNS queries from controller.*
[chip-lighting-device-k32w.log](https://github.com/project-chip/connectedhomeip/files/7212176/chip-lighting-device-k32w.log)
[chip-tool.log](https://github.com/project-chip/connectedhomeip/files/7212178/chip-tool.log)
*

bzbarsky-apple, 2022-06-27:
@saumildmehta Is this still an issue?

woody-apple, 2022-07-22:
Issue Review: Closing, given this is an older issue. Please re-file if this still an issue.

closed 2022-07-22 by woody-apple