project-chip/connectedhomeip #9888: M5Stack crashes sometimes when starting the mDNS server

sagar-apple, 2021-09-22:
#### Problem

```
E (32370) chip[DL]: Failed to post event to CHIP Platform event queue
I (32370) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
E (32370) chip[CSL]: Failed to queue CHIP System Layer event (type 1005): Error 0x01000000
E (32390) chip[DIS]: Have operational credentials
I (32400) app-devicecallbacks: Current free heap: 28720

E (32400) chip[DL]: Long dispatch time: 330 ms, for event type 32782
E (32410) chip[DL]: Failed to post event to CHIP Platform event queue
V (32410) chip[DIS]: Mdns StartServer mode=0
E (32420) chip[CSL]: Failed to queue CHIP System Layer event (type 1005): Error 0x01000000
E (32430) chip[DL]: Failed to post event to CHIP Platform event queue
E (32440) chip[CSL]: Failed to queue CHIP System Layer lambda event: Error 0x01000000
V (32450) chip[CSL]: Object::DeferredRelease 0x3ffd0f60->PostEvent failed err(1000000)

abort() was called at PC 0x400f4ebc on core 0
0x400f4ebc: chip::System::Object::DeferredRelease(chip::System::LayerLwIP*, chip::System::Object::ReleaseDeferralErrorTactic) at system/SystemObject.cpp:100 (discriminator 1)

Backtrace:0x4009186a:0x3ffe6c50 0x40091f95:0x3ffe6c70 0x4009843a:0x3ffe6c90 0x400f4ebc:0x3ffe6d00 0x40106cda:0x3ffe6d40 0x401046ff:0x3ffe6d60 0x400fc193:0x3ffe6d80 0x400f0b76:0x3ffe6da0 0x400dc6f8:0x3ffe6e60 0x400e647e:0x3ffe6ec0 0x4019eecf:0x3ffe6ee0 0x4019fdb5:0x3ffe6f00 0x400f855d:0x3ffe6f20 0x400f878a:0x3ffe6f40 0x400f87b1:0x3ffe6fa0 0x40095191:0x3ffe6fc0
0x4009186a: panic_abort at /Users/sagar_dhawan/Work/misc/esp-idf/components/esp_system/panic.c:367
0x40091f95: esp_system_abort at /Users/sagar_dhawan/Work/misc/esp-idf/components/esp_system/system_api.c:112
0x4009843a: abort at /Users/sagar_dhawan/Work/misc/esp-idf/components/newlib/abort.c:46
0x400f4ebc: chip::System::Object::DeferredRelease(chip::System::LayerLwIP*, chip::System::Object::ReleaseDeferralErrorTactic) at system/SystemObject.cpp:100 (discriminator 1)
0x40106cda: chip::Inet::EndPointBasis::DeferredFree(chip::System::Object::ReleaseDeferralErrorTactic) at inet/EndPointBasis.cpp:51
0x401046ff: chip::Inet::UDPEndPoint::Free() at inet/UDPEndPoint.cpp:435
0x400fc193: mdns::Minimal::ServerBase::Shutdown() at lib/mdns/minimal/Server.cpp:118
 (inlined by) mdns::Minimal::ServerBase::Shutdown() at lib/mdns/minimal/Server.cpp:135
0x400f0b76: chip::Mdns::(anonymous namespace)::AdvertiserMinMdns::Start(chip::Inet::InetLayer*, unsigned short) at lib/mdns/Advertiser_ImplMinimalMdns.cpp:277
0x400dc6f8: chip::app::MdnsServer::StartServer(chip::Mdns::CommissioningMode) at /Users/sagar_dhawan/Work/connectedhomeip/examples/all-clusters-app/esp32/build/../third_party/app/server/Mdns.cpp:416
0x400e647e: DeviceCallbacks::DeviceEventCallback(chip::DeviceLayer::ChipDeviceEvent const*, int) at /Users/sagar_dhawan/Work/connectedhomeip/examples/all-clusters-app/esp32/build/../main/DeviceCallbacks.cpp:84
0x4019eecf: chip::DeviceManager::CHIPDeviceManager::CommonDeviceEventHandler(chip::DeviceLayer::ChipDeviceEvent const*, int) at /Users/sagar_dhawan/Work/connectedhomeip/examples/all-clusters-app/esp32/build/../main/CHIPDeviceManager.cpp:47
0x4019fdb5: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToApplication(chip::DeviceLayer::ChipDeviceEvent const*) at include/platform/internal/GenericPlatformManagerImpl.cpp:305 (discriminator 2)
0x400f855d: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at include/platform/internal/GenericPlatformManagerImpl.cpp:252
0x400f878a: chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at include/platform/PlatformManager.h:383
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:183
0x400f87b1: chip::DeviceLayer::PlatformManager::RunEventLoop() at include/platform/PlatformManager.h:308
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:208
0x40095191: vPortTaskWrapper at /Users/sagar_dhawan/Work/misc/esp-idf/components/freertos/port/xtensa/port.c:168
```

dhrishi, 2021-10-14:
Hi @sagar-apple Did you encounter this issue with the latest codebase as well?

holbrookt, 2022-02-01:
Removing 1.0 as this hasn't seen any activity since October. @sagar-apple please re-label if this is still an issue.

closed 2022-02-01 by sagar-apple