project-chip/connectedhomeip #9903: Pump Configuration and Control Cluster - optional attributes in Pump Dynamic Information Attributes not implemented

raju-apple, 2021-09-22:
Spec Snapshot : 
[image]

In the above Pump Dynamic Information Attributes table , only the mandatory attributes seem to be implemented by the SDK , all the optional attributes need to be implemented as well

tlykkeberg-grundfos, 2022-04-21:
I'm not quite sure here, what is meant by "implemented by the SDK"? Is it because the attributes are available in the all-clusters-app .zap file or some other .zap file?

naveenkommareddi, 2022-05-17:
We should move this to 1.x since optional attributes are out of scope for 1.0

closed 2023-06-10 by stale[bot]