project-chip/connectedhomeip #9908: Status code handling in IM is a bit broken

bzbarsky-apple, 2021-09-23:
#### Problem
`AddStatusCode` on CommandClient/Handler, and `WriteHandler::AddAttributeStatusCode` both take a protocol id but then take the IM-type status enum for the actual "protocol status". And we do have some consumers that are passing in some other protocol id, but still an IM status code to go with it....

Also, none of this stuff matches what the IM and encoding spec seem to have right now, which is just encoding a TLV struct with an IM status and maybe a cluster status....

 #### Proposed Solution
Adjust code to follow spec wire format, fix APIs so they can't be misused.

mrjerryjohns, 2021-11-19:
This has been fixed:

```
    CHIP_ERROR AddStatus(const ConcreteCommandPath & aCommandPath, const Protocols::InteractionModel::Status aStatus) override;

    CHIP_ERROR AddClusterSpecificSuccess(const ConcreteCommandPath & aCommandPath, ClusterStatus aClusterStatus) override;

    CHIP_ERROR AddClusterSpecificFailure(const ConcreteCommandPath & aCommandPath, ClusterStatus aClusterStatus) override;
```

bzbarsky-apple, 2021-11-20:
But most of those APIs are not actually implemented.... certainly not on WriteHandler.