project-chip/connectedhomeip #9917: chip-tool pairing ble-thread fail

JasonHu1, 2021-09-23:
@jepenven-silabs 
**Problem**

I use chip-tool and lighting-app with silabs BRD4161A
git commitid follow:
lighting-app :741ab0a032c290f4653593ceac06048c357ebae5
chip-tool:6db59aa066599f9e5967e900c809a85037a56e8b
ot-br-posix:17cdb3bad795db87bfcd2c272366ab5a4f270ffc

**Run the following CLI：**
./chip-tool pairing ble-thread operationalDataset=hex:0e080000000000010000000300000f35060004001fffe00208a8baf4311d2c1acf0708fd06217852196229051000112233445566778899aabbccddeeff030f4f70656e5468726561642d34343234010212340410978d42ebeab4875a137cf69b3df690f90c0402a0fff8 12344321 73141520 3840

**However, the following error message appears:**

[1632395145.634725][3716:3721] CHIP:ZCL: DefaultResponse:
[1632395145.634767][3716:3721] CHIP:ZCL:   Transaction: 0xaaaac07e4210
[1632395145.634864][3716:3721] CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
[1632395145.634914][3716:3721] CHIP:TOO: Default Failure Response: 0x01
[1632395145.635001][3716:3721] CHIP:DMG: ICR moving to [Uninitiali]
[1632395145.635027][3716:3716] CHIP:TOO: Run command failure: ../../commands/pairing/PairingCommand.cpp:332: CHIP Error 0x000000AC: Internal error

[chip-tool_pairing_ble-thread_fail.txt](https://github.com/project-chip/connectedhomeip/files/7217343/chip-tool_pairing_ble-thread_fail.txt)

bluebin14, 2021-09-23:
Remove "operationalDataset=" from command line

JasonHu1, 2021-09-26:
> Remove "operationalDataset=" from command line

thanks for help me
follow your instructions,i can see this log:

[1632620614.021923][1627:1632] CHIP:DIS: **mDNS broadcast success**

Does this mean that the commissioning process runs to step13 Operational discovery over operational network?

the after error occur is
[1632620726.574230][1627:1627] CHIP:TOO: No response from device
[1632620726.574339][1627:1627] CHIP:TOO: Run command failure: ../../commands/common/Command.h:217: CHIP Error 0x000000AC: In                   ternal error

Please see the attachment for the completed log

[chip-tool_pairing_ble-thread_fail_1.txt](https://github.com/project-chip/connectedhomeip/files/7230593/chip-tool_pairing_ble-thread_fail_1.txt)

closed 2021-11-01 by JasonHu1