project-chip/connectedhomeip #9948: python controller connect ble fail

wqx6, 2021-09-24:
#### Problem
After #9757 merged, the python controller can not establish secure session to device with PASESession time out. 

The python controller logs:
```
$ ./out/python_env/bin/chip-device-ctrl -n 8888                                                                                                            
[1632468527.438996][3991016:3991016] CHIP:CTL: SyncSetKeyValue on ExampleOpCredsCAKey                                                                      
[1632468527.439047][3991016:3991016] CHIP:CTL: SyncSetKeyValue on ExampleOpCredsICAKey                                  
[1632468527.439087][3991016:3991016] CHIP:CTL: Generating NOC                                                                                              
[1632468527.439148][3991016:3991016] CHIP:CTL: Generating ICAC                                                                                             
[1632468527.439185][3991016:3991016] CHIP:CTL: Generating RCAC                                                                                             
[1632468527.439221][3991016:3991016] CHIP:CTL: SyncSetKeyValue on ExampleCARootCert0                                                                       
[1632468527.441248][3991016:3991016] CHIP:IN: UDP::Init bind&listen port=0                                                                                 
[1632468527.441262][3991016:3991016] CHIP:IN: UDP::Init bound to port=59592                                                                                
[1632468527.441265][3991016:3991016] CHIP:IN: UDP::Init bind&listen port=0                                                                                 
[1632468527.441272][3991016:3991016] CHIP:IN: UDP::Init bound to port=56370                                                                                
[1632468527.441274][3991016:3991016] CHIP:IN: TransportMgr initialized                                                                                     
[1632468527.441276][3991016:3991016] CHIP:DIS: Init fabric pairing table with server storage                                                               
[1632468527.441318][3991016:3991016] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-QBW6p7)                                                     
[1632468527.441402][3991016:3991016] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)                                                            
[1632468527.441409][3991016:3991016] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 27000 (0x6978)  
[1632468527.442673][3991016:3991016] CHIP:DL: Avahi client registered                                                                                      
[1632468527.442962][3991016:3991016] CHIP:ZCL: Using ZAP configuration...                                                                                  
[1632468527.442979][3991016:3991016] CHIP:ZCL: deactivate report event                                                                                     
[1632468527.443103][3991016:3991016] CHIP:DIS: Verifying the received credentials            
[1632468527.443442][3991016:3991016] CHIP:DIS: Added new fabric at index: 1, Initialized: 1                                                                
[1632468527.443446][3991016:3991016] CHIP:DIS: Assigned compressed fabric ID: 0xF2FEF6B23796A41A, node ID: 0x00000000000022B8
[1632468527.443449][3991016:3991016] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0xF2FEF6B23796A41A                                   
[1632468527.443464][3991016:3991016] CHIP:IN: UDP::Init bind&listen port=5550                                                                              
[1632468527.443474][3991016:3991016] CHIP:IN: UDP::Init bound to port=5550                                                                                 
[1632468527.443477][3991016:3991016] CHIP:IN: UDP::Init bind&listen port=5550                                                                              
[1632468527.443484][3991016:3991016] CHIP:IN: UDP::Init bound to port=5550                                                                                 
[1632468527.443487][3991016:3991016] CHIP:IN: TransportMgr initialized                                                                                     
[1632468527.443490][3991016:3991016] CHIP:DIS: Mdns StartServer mode=0                                                                                     
[1632468527.443719][3991016:3991016] CHIP:DIS: Failed to find a valid admin pairing. Node ID unknown
[1632468527.443722][3991016:3991016] CHIP:DIS: Start dns-sd server - no current nodeId                                                                     
[1632468527.443854][3991016:3991016] CHIP:DIS: Using wifi MAC for hostname                                                                                 
[1632468527.443881][3991016:3991016] CHIP:DIS: Advertise commission parameter vendorID=9050 productID=65279 discriminator=3840/00           
[1632468527.443886][3991016:3991016] CHIP:DL: Cannot get device role with Thread api client:                                                               
[1632468527.443978][3991016:3991016] CHIP:DL: Cannot set hostname on this system, continue anyway...                                                       
[1632468527.443988][3991016:3991016] CHIP:DL: PublishService F41ADB0109DBAF23._matterc._udp5540.-1                                                         
[1632468527.444798][3991016:3991016] CHIP:DIS: Scheduling Discovery timeout in secs=900                                                                    
[1632468527.444894][3991016:3991016] CHIP:DIS: Using wifi MAC for hostname                                                                                 
[1632468527.444905][3991016:3991016] CHIP:DIS: Advertise commission parameter vendorID=9050 productID=65279 discriminator=3840/00                          
[1632468527.444910][3991016:3991016] CHIP:DL: Cannot get device role with Thread api client:                                                               
[1632468527.445009][3991016:3991016] CHIP:DL: Cannot set hostname on this system, continue anyway...                                                       
[1632468527.445018][3991016:3991016] CHIP:DL: PublishService F41ADB0109DBAF23._matterd._udp5550.-1                                                         
[1632468527.445211][3991016:3991016] CHIP:DL: Avahi publish service failed: ../../src/platform/Linux/MdnsImpl.cpp:474: CHIP Error 0x000000AC: Internal error
[1632468527.445216][3991016:3991016] CHIP:DIS: Failed to advertise commissioner: ../../src/platform/Linux/MdnsImpl.cpp:474: CHIP Error 0x000000AC: Internal error
[1632468527.447095][3991016:3991024] CHIP:DL: Platform main loop started.                                                                                  
Chip Device Controller Shell                                                                                                                               
                                                                                                                                                           
chip-device-ctrl > [1632468528.429452][3991016:3991024] CHIP:DL: Avahi group established                  
                                                                                                                                                           
chip-device-ctrl > connect -ble 3840 20202021 12345                                                                                                        
Device is assigned with nodeid = 12345                                                                                                                     
[1632468542.215927][3991016:3991024] CHIP:SC: Assigned local session key ID 1                                                                              
[1632468542.216034][3991016:3991024] CHIP:IN: Prepared plaintext message 0x7fdf37ffd880 to 0x0000000000000000 of type 0x20 and protocolId (0, 0) on exchange 12098i with MessageCounter:2477711099.
[1632468542.216075][3991016:3991024] CHIP:IN: Sending plaintext msg 0x7fdf37ffd880 with MessageCounter:2477711099 to 0x0000000000000000 at monotonic time: 1918997643 msec
[1632468542.216093][3991016:3991024] CHIP:IN: Message appended to BLE send queue                     
[... cut ...]
[1632468570.766776][3991016:3991024] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1632468572.216882][3991016:3991024] CHIP:CTL: SyncSetKeyValue on StartKeyID
Failed to establish secure session to device: 50
[1632468572.217169][3991016:3991024] CHIP:SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
[1632468572.217207][3991016:3991024] CHIP:CTL: SyncSetKeyValue on StartKeyID
CHIP Error 0x00000032: Timeout
Failed to establish secure session to device: 50
```

wqx6, 2021-09-24:
@sharadb-amazon

sharadb-amazon, 2021-09-24:
Checking.

sharadb-amazon, 2021-10-06:
Should be fixed when https://github.com/project-chip/connectedhomeip/pull/10248 is merged.

andy31415, 2021-10-06:
BLE connect works now for me, however issuing a subsequent command fails with "Not connected":

```
....
[1633549299.582059][4129306:4129322] CHIP:CTL: Operational credentials provisioned on device 0x2484c88
Secure Session to Device Established
Device temporary node id (**this does not match spec**): 964669
chip-device-ctrl > zcl NetworkCommissioning AddThreadNetwork 1 0 0 operationalDataset=hex:0e080000000000010000000300001335060004001fffe00208ebe7b16aef5344fc0708fd0e6216d438396605109bdb719f736680abf025c123eab48a0e030f4f70656e5468726561642d3062346401020b4d04102e103f378a4635592d82000ed44d687c0c0402a0fff8 breadcrumb=0 timeoutMs=1000
An exception occurred during process ZCL command:
CHIP Error 0x00000048: Not connected
chip-device-ctrl > 
```

closed 2021-10-06 by andy31415