project-chip/connectedhomeip #9963: commissionable node advertisement should be disabled after device is provisioned

cjandhyala, 2021-09-24:
commissionable node mDNS advertisement should be disabled after device is provisioned. The current SDK implementation (ToT 327a47125fc0710510a1df6402d61e0730320251) is still showing device in commissionable mode (CM=1 subtype) even after the device commissioned.

chrisdecenzo, 2021-10-07:
when commissioning is done using clusters (currently a compile-time option, which @cecille is working to enable by default, then commissioning mode advertising works as designed.

The problem with some older builds is that the commissioning complete cluster command was not getting sent.

So this issue should get resolved soon.

mrjerryjohns, 2022-01-31:
Should be resolved given commissioning complete is now sent.

closed 2022-01-31 by mrjerryjohns