project-chip/connectedhomeip #9975: Hang in TestInetEndPoint on Darwin

bzbarsky-apple, 2021-09-25:
#### Problem
I have intermittently seen hangs in TestInetEndPoint, and I finally had the time to attach a debugger to one.

We're in `TestResolveHostAddress` looping forever in this loop:
```
        while (!callbackHandlerCalled)
        {
            ServiceNetwork(kSleepTimeMilliseconds);
        }
```
on lines 146-149.  That's the wait for results after calling:
```
    err = gInet.ResolveHostAddress(testHostName1, 1, testDestAddr, HandleDNSResolveComplete, &callbackHandlerCalled);
```
where `testHostName1` is `"www.google.com"`.

 #### Proposed Solution
What I don't really know is how we got into this state where the callback is just not getting called....

@kpschoedel @sagar-apple @woody-apple @carol-apple @andy31415

tcarmelveilleux, 2022-01-31:
@bzbarsky-apple Is this still happening? If not please close

bzbarsky-apple, 2022-01-31:
And that's because we removed the DNS resolver and its test in https://github.com/project-chip/connectedhomeip/pull/10725

For the record, that test was probably failing on Darwin because we compile Darwin with locking disabled but the DNS resolver used a helper thread and locking-based synchronization, so most likely ended up with data races and then hung.