project-chip/connectedhomeip #9998: Attempting to send an error response partway through encoding a data command response crashes

bzbarsky-apple, 2021-09-27:
#### Problem
Steps to reproduce:

1. `git checkout 9be31d7a0f1f267507e750d83e7e10018642add2`
2. `./gn_build.sh` (or whatever way you want to build `chip-tool`
3. `scripts/examples/gn_build_example.sh examples/ota-provider-app/linux/ out/apps chip_config_network_layer_ble=false`
4.  In a separate terminal, `out/apps/chip-ota-provider-app`
5. `out/debug/standalone/chip-tool pairing onnetwork 0 20202021 3840 ::1 5540`

The ota provider app crashes, like so:
```
* thread #3, queue = 'com.zigbee.chip.framework.controller.workqueue', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x0000000100013ee6 chip-ota-provider-app`emberAfSendDefaultResponseWithCallback(cmd=0x00000001000e8df0, status=EMBER_ZCL_STATUS_FAILURE, callback=0x0000000000000000)(chip::MessageSendDestination const&, EmberApsFrame*, unsigned short, unsigned char*, unsigned char)) at util.cpp:820:10
   817 	
   818 	    // If the Disable Default Response sub-field is set, Default Response commands
   819 	    // are only sent if there was an error.
-> 820 	    if ((cmd->buffer[0] & ZCL_DISABLE_DEFAULT_RESPONSE_MASK) && status == EMBER_ZCL_STATUS_SUCCESS)
   821 	    {
   822 	        return EMBER_SUCCESS;
   823 	    }
```
with a null `cmd->buffer`.

<details>
<summary>The stack: </summary>

```
  * frame #0: 0x0000000100013ee6 chip-ota-provider-app`emberAfSendDefaultResponseWithCallback(cmd=0x00000001000e8df0, status=EMBER_ZCL_STATUS_FAILURE, callback=0x0000000000000000)(chip::MessageSendDestination const&, EmberApsFrame*, unsigned short, unsigned char*, unsigned char)) at util.cpp:820:10
    frame #1: 0x0000000100013e76 chip-ota-provider-app`emberAfSendImmediateDefaultResponseWithCallback(status=EMBER_ZCL_STATUS_FAILURE, callback=0x0000000000000000)(chip::MessageSendDestination const&, EmberApsFrame*, unsigned short, unsigned char*, unsigned char)) at util.cpp:793:12
    frame #2: 0x0000000100014029 chip-ota-provider-app`emberAfSendImmediateDefaultResponse(status=EMBER_ZCL_STATUS_FAILURE) at util.cpp:798:12
    frame #3: 0x0000000100016b3f chip-ota-provider-app`emberAfOperationalCredentialsClusterCertificateChainRequestCallback(endpoint=0, commandObj=0x00000001000e5e38, certificateType='\x02') at operational-credentials-server.cpp:529:9
    frame #4: 0x000000010001d603 chip-ota-provider-app`chip::app::clusters::OperationalCredentials::DispatchServerCommand(apCommandObj=0x00000001000e5e38, aCommandId=2, aEndpointId=0, aDataTlv=0x00007000084af480) at IMClusterCommandHandler.cpp:584:21
    frame #5: 0x000000010001e355 chip-ota-provider-app`chip::app::DispatchSingleClusterCommand(aClusterId=62, aCommandId=2, aEndPointId=0, aReader=0x00007000084af480, apCommandObj=0x00000001000e5e38) at IMClusterCommandHandler.cpp:809:9
    frame #6: 0x000000010002ae1b chip-ota-provider-app`chip::app::CommandHandler::ProcessCommandDataElement(this=0x00000001000e5e38, aCommandElement=0x00007000084af610) at CommandHandler.cpp:106:9
    frame #7: 0x0000000100029705 chip-ota-provider-app`chip::app::Command::ProcessCommandMessage(this=0x00000001000e5e38, payload=0x00007000084b0678, aCommandRoleId=HandlerId) at Command.cpp:112:15
    frame #8: 0x000000010002a587 chip-ota-provider-app`chip::app::CommandHandler::OnInvokeCommandRequest(this=0x00000001000e5e38, ec=0x00000001000e4928, payloadHeader=0x00007000084b0128, payload=0x00007000084b0678) at CommandHandler.cpp:48:11
    frame #9: 0x000000010002e65c chip-ota-provider-app`chip::app::InteractionModelEngine::OnInvokeCommandRequest(this=0x00000001000e5e20, apExchangeContext=0x00000001000e4928, aPayloadHeader=0x00007000084b0128, aPayload=0x00007000084b0678) at InteractionModelEngine.cpp:215:48
    frame #10: 0x000000010002f09d chip-ota-provider-app`chip::app::InteractionModelEngine::OnMessageReceived(this=0x00000001000e5e20, apExchangeContext=0x00000001000e4928, aPayloadHeader=0x00007000084b0128, aPayload=0x00007000084b0678) at InteractionModelEngine.cpp:324:16
    frame #11: 0x000000010007bfc7 chip-ota-provider-app`chip::Messaging::ExchangeContext::HandleMessage(this=0x00000001000e4928, messageCounter=0, payloadHeader=0x00007000084b0128, peerAddress=0x00007000084b03d0, msgFlags=(mValue = 0), msgBuf=0x00007000084b0678) at ExchangeContext.cpp:461:27
    frame #12: 0x000000010007e67a chip-ota-provider-app`chip::Messaging::ExchangeManager::OnMessageReceived(this=0x00000001000e4750, packetHeader=0x00007000084b0200, payloadHeader=0x00007000084b0128, session=SessionHandle @ 0x00007000084aff98, source=0x00007000084b03d0, isDuplicate=No, msgBuf=0x00007000084b0678) at ExchangeMgr.cpp:303:30
    frame #13: 0x000000010008f7b8 chip-ota-provider-app`chip::SessionManager::SecureMessageDispatch(this=0x00000001000e3780, packetHeader=0x00007000084b0200, peerAddress=0x00007000084b03d0, msg=0x00007000084b0678) at SessionManager.cpp:488:14
    frame #14: 0x000000010008eed8 chip-ota-provider-app`chip::SessionManager::OnMessageReceived(this=0x00000001000e3780, peerAddress=0x00007000084b03d0, msg=0x00007000084b0678) at SessionManager.cpp:331:9
    frame #15: 0x0000000100093c03 chip-ota-provider-app`chip::TransportMgrBase::HandleMessageReceived(this=0x00000001000e3718, peerAddress=0x00007000084b03d0, msg=0x00007000084b0678) at TransportMgrBase.cpp:66:26
    frame #16: 0x00000001000b250d chip-ota-provider-app`chip::Transport::Base::HandleMessageReceived(this=0x00000001000e3740, source=0x00007000084b03d0, buffer=0x00007000084b0678) at Base.h:92:20
    frame #17: 0x00000001000b21fe chip-ota-provider-app`chip::Transport::UDP::OnUdpReceive(endPoint=0x0000000101005000, buffer=0x00007000084b0678, pktInfo=0x00007000084b0680) at UDP.cpp:120:10
    frame #18: 0x0000000100063c98 chip-ota-provider-app`chip::Inet::IPEndPointBasis::HandlePendingIO(this=0x0000000101005000, aPort=5540) at IPEndPointBasis.cpp:1130:9
    frame #19: 0x0000000100065314 chip-ota-provider-app`chip::Inet::UDPEndPoint::HandlePendingIO(this=0x0000000101005000, events=(mValue = '\x01')) at UDPEndPoint.cpp:938:26
    frame #20: 0x00000001000652a9 chip-ota-provider-app`invocation function for block in chip::Inet::UDPEndPoint::Bind(.block_descriptor=0x000000010130c110) at UDPEndPoint.cpp:258:19
    frame #21: 0x00007fff67372658 libdispatch.dylib`_dispatch_client_callout + 8
    frame #22: 0x00007fff67374818 libdispatch.dylib`_dispatch_continuation_pop + 414
    frame #23: 0x00007fff673844be libdispatch.dylib`_dispatch_source_invoke + 2084
    frame #24: 0x00007fff67377af6 libdispatch.dylib`_dispatch_lane_serial_drain + 263
    frame #25: 0x00007fff673785d6 libdispatch.dylib`_dispatch_lane_invoke + 363
    frame #26: 0x00007fff67381c09 libdispatch.dylib`_dispatch_workloop_worker_thread + 596
    frame #27: 0x00007fff675cca3d libsystem_pthread.dylib`_pthread_wqthread + 290
    frame #28: 0x00007fff675cbb77 libsystem_pthread.dylib`start_wqthread + 15

```

</details>

What happens here is we land inside `emberAfOperationalCredentialsClusterCertificateChainRequestCallback` and start writing our response.  We do:
```
   497 	    SuccessOrExit(err = commandObj->PrepareCommand(cmdParams));
```
then we try to call `GetProductAttestationIntermediateCert` on `dacProvider`, and that fails because it's a `chip::Credentials::UnimplementedDACProvider`.  So we `SuccessOrExit(err)`, land in the `exit:` block, and do `emberAfSendImmediateDefaultResponse(EMBER_ZCL_STATUS_FAILURE)`.

That lands us eventually in `emberAfSendDefaultResponseWithCallback`, we call `chip::app::Compatibility::IMEmberAfSendDefaultResponseWithCallback` to handle things, and that returns false because `currentCommandObject->AddStatusCode` fails due to the TLV writer not being in a good state for whatever `AddStatusCode` is trying to do, because we had already started writing command response bits into it.

We then press on to the ember default response handling bits and crash.

 #### Proposed Solution
Several things need to happen:

1. Fix `IMEmberAfSendDefaultResponseWithCallback` to only return false if `currentCommandObject == nullptr`.  That should fix the crash, at least, but the actual command response bits will still be broken.
2. Decide whether the CommandHandler API will suport this use case of starting a response, then hitting an error and trying  to send a failure status instead.
    * If supported, fix the CommandHandler API accordingly.  Add a test that exercises this via TestCluster + yaml.
    * If not supported, document clearly that no fallible operations other than TLV bits are allowed to happen between the `PrepareCommand` call and the `FinishCommand` call.  Fix the opcreds cluster to follow the API constraint.

@carol-apple @mrjerryjohns @yunhanw-google @erjiaqing

bzbarsky-apple, 2021-09-28:
Chances are addressing #8893 might help here, if the "send a response" bits become a little more atomic.

mrjerryjohns, 2021-09-28:
I think 2) makes the most sense in the short-term.

#8893 and moving to cluster objects will certainly make the encoding of responses 'API atomic', preventing the possibility of this particular situation from happening.

What should happen though if as part of encoding the cluster object, the TLV operation fails, for a reason other than bugs (e.g trying to write out too big a list)? I presume in that case, we should just terminate the interaction as opposed to attempting to encode a status element?

mrjerryjohns, 2021-11-19:
This is fixed now given we use cluster objects.

bzbarsky-apple, 2021-11-20:
It's not, because you can still end up trying to send a status response in the middle of encoding a cluster object if you implement its Encode and it does interesting work (like groups cluster does).

We need to really sort out the "who will send the status response" bits and then make sure everyone is doing the right thing.

bzbarsky-apple, 2022-06-13:
I think https://github.com/project-chip/connectedhomeip/issues/8893 fixed this sufficiently.  In particular, `AddResponseData` can never leave us in a bad state now, because it rolls back on failure.

closed 2022-06-13 by bzbarsky-apple