project-chip/connectedhomeip #10040: [ota-requestor-app] Use DefaultOTAProvider attribute write as trigger to begin OTA

holbrookt, 2021-09-29:
#### Problem
Currently, `ota-provider-app` begins the OTA process immediately upon startup. This is only feasible right now because `ota-requestor-app` is not commissioned, and instead "steals" credentials from `chip-tool` persisted data.

To mirror a more likely production scenario, `ota-requestor-app` should wait to be commissioned. For testing purposes, we will still want to be able to control when `ota-requestor-app` begins the OTA process. For starters, we can use the `DefaultOTAProvider` attribute write as the trigger to begin. This also provides a method for giving `ota-requestor-app` the location of the OTA Provider to connect to.

 #### Proposed Solution
* add OTA Requestor cluster to `ota-requestor-app` and include `DefaultOTAProvider` attribute
* listen for `DefaultOTAProvider` attribute writes, connect to OTA Provider and send `QueryImage` when this happens

tcarmelveilleux, 2021-09-29:
This is problematic since DefaultOTAProvider could be written before CommissioningComplete command has executed and it's possible it's not yet guaranteed/feasible for the Node to go check for OTA at that time. This methodology would not scale to real products.

Perhaps the AnnounceOtaProvider command could be used instead.

holbrookt, 2021-09-29:
Thanks @tcarmelveilleux, I had thought that the attribute write always took place after commissioning. Agree that AnnounceOTAProvider would be a better trigger. I'll edit this Issue to reflect that

carol-apple, 2021-10-29:
Closing as this issue is fixed by https://github.com/project-chip/connectedhomeip/pull/11010

closed 2021-10-29 by carol-apple