project-chip/connectedhomeip #10046: Cannot complete commissioning flow when running apps on macOS

erjiaqing, 2021-09-29:
#### Problem

CHIP Tool cannot finish commissioning flow on macOS:

https://gist.github.com/erjiaqing/158f8546b7d5563f95aaf96df48183aa

erjiaqing, 2021-09-29:
@bzbarsky-apple @pan-apple PTAL, thanks.

erjiaqing, 2021-09-29:
After running under debugger, seems this is caused by failure of setting up BLE advertisement on macOS, maybe we should disable it by default?

bzbarsky-apple, 2021-09-29:
So this is for the case when the client app:

1) Was not compiled with `chip_config_network_layer_ble=false`.
2) Is attempting to be commissioned via on-network commissioning.

right?

Ideally our client example apps would support being commissioned via either the IP network _or_ BLE in their "linux" configuration (which is what's being used here).

erjiaqing, 2021-09-29:
> So this is for the case when the client app:
> 
> 1. Was not compiled with `chip_config_network_layer_ble=false`.
> 2. Is attempting to be commissioned via on-network commissioning.
> 
> right?

Right

> 
> Ideally our client example apps would support being commissioned via either the IP network _or_ BLE in their "linux" configuration (which is what's being used here).

However, when it failed to setup BLEAdvertisement, it will close commissioning window. I think either a failure asseration or continue with IP commissioning is acceptable.

bzbarsky-apple, 2021-10-19:
The thing is, if it's compiled with `chip_config_network_layer_ble=true` then for me it succeeds in setting up the BLE advertisement, at first glance.... Which is why all the "compile and run locally" bits use `chip_config_network_layer_ble=false`.

woody-apple, 2022-01-25:
This seems to be working now

closed 2022-01-25 by woody-apple