project-chip/connectedhomeip #10074: Clarify the message context life cycle management

yunhanw-google, 2021-09-29:
@bzbarsky-apple If I understand correctly, `ProcessReportData` will trigger the transmission of a StatusResponse, which is the final message in this exchange, resulting in the closure of the exchange automatically.

So this setting of the `mpExchangeCtx` to nullptr is purely for cleanliness, and serves no functional value right?

_Originally posted by @mrjerryjohns in https://github.com/project-chip/connectedhomeip/pull/10061#discussion_r718804563_

yunhanw-google, 2021-09-29:
From @bzbarsky-apple  

The basic rules in the tree right now are as follows (see docs for ExchangeContext::SendMessage, ExchangeDelegate::OnResponseTimeout, ExchangeDelegate::OnMessageReceived
On exchange timeout notification, need to call WillSendMessage or SendMessage or null out the pointer.  In practice, null out the pointer.
OnMessageReceived, same thing.  In practice, if you are sending a response either send it or tell the exchange you plan to send it, otherwise null out the pointer.
When doing a SendMessage, if it succeeds and you are not expecting a response null out the pointer.
In all other cases, you are holding a ref to the exchange and should call Close if you want to close it.
So for example, for subscriptions when you get the last message, the one you don't plan to respond to, just null out the pointer and the exchange will know it's the last message and close itself.
On the other hand, if you are waiting for a response and you decide you want to kill off the transaction, you have to do the exchange closing yourself.

yunhanw-google, 2021-09-29:
this setting of the mpExchangeCtx to nullptr is purely for cleanliness, and serves no functional value right??
@bzbarsky-apple Except in cases where we have other code that tries to close it if non-null, yes

bzbarsky-apple, 2021-09-29:
The important part is that we're inside `OnMessageReceived` here, right?  So yes, given that the exchange will close itself unless we send a message expecting a response or claim that we will...

What I don't understand is why the following `ShutdownInternal` is conditional on `err` being success.  if `ProcessReportData` fails, what shuts us down?

closed 2023-04-02 by stale[bot]