project-chip/connectedhomeip #10089: Multiple Controllers: Remove UDC Service from CHIPDeviceController and make it a part of the CHIPDeviceControllerFactory

sagar-apple, 2021-09-29:
#### Problem
When creating more than 1 controller at a time, we end up trying to start multiple UDC Services all bound to the same port. 
This is not going to work. 

 #### Proposed Solution
- Create a single UDC service that's always running inside the DeviceControllerFactory(only because SystemState is owned by the Factory).
- Commissioner applications must provide a `UserConfirmationProvider` to the `CHIPDeviceControllerFactory` on init. 
- Commissioner Applications implement their own business logic to figure out what specific fabric to commission the client node to using the Factory APIs to setup the Commissioner.

The main thing stopping me from pulling UDC commissioning service out from Controller right now is the mDNS singleton usage in CHIPDeviceController.

msandstedt, 2021-12-31:
https://github.com/project-chip/connectedhomeip/pull/13294 introduces an approach where the commissioner is decoupled from the controller, and commissioner objects can be created on an as-needed basis for each commissionee.

I would think this would greatly simplify something like UDC in general because then the UDC service could instantiate its own commissioner instances with no reliance on other long-running singleton-like objects.

msandstedt, 2022-01-06:
Absent use of #13294, I think there may also be an architectural problem with the originally proposed solution.  In many ecosystem architectures, it may be the signing infrastructure that selects fabric.  This would make it difficult to select the fabric-scoped commissioner instance up front when UDC requests commissioning.

The architecture in #13294 very cleanly supports an implementation where commissioner instances don't have fabric scope.  That would solve this problem.